zephyrproject-rtos/zephyr #55450: Broken link to tc_util.h in ztest.rst

mbrunnen, 2023-03-06:
This line has a broken link `subsys/testsuite/include/tc_util.h`, which should be `subsys/testsuite/include/zephyr/tc_util.h`

```
See the file :zephyr_file:`subsys/testsuite/include/tc_util.h` to see which macros and/or defines can be overridden.
```

https://github.com/zephyrproject-rtos/zephyr/blob/29cc07efbd71b779d584a30fb495b4a3de6e0e91/doc/develop/test/ztest.rst?plain=1#LL539C44-L539C44

henrikbrixandersen, 2023-03-07:
Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

fabiobaltieri, 2023-03-07:
Hi, Zephyr is an open-source project, and as such we welcome and encourage the community to submit patches directly. Would you be able to submit a pull request with the fix? This is the [contribution guideline documentation](https://docs.zephyrproject.org/latest/contribute/guidelines.html#contribution-guidelines).

mbrunnen, 2023-05-05:
Created a PR

closed 2023-05-05 by mbrunnen