zephyrproject-rtos/zephyr #55460: External Source Code: Integrate Coremark as a module

carlescufi, 2023-03-06:
## Origin

Coremark, by the Embedded Microprocessor Benchmark Consortium:
https://github.com/eembc/coremark

## Purpose

Coremark is an industry standard for CPU benchmarking.

## Mode of integration

Would be integrated as a module, upstream already contains the `zephyr/` folder so I just suggest forking [the coremark repo](https://github.com/eembc/coremark) into the Zephyr GH org.

## Maintainership

@volodymyr-bondarchuk
@carlescufi 

## Pull Request

TBD

## Description

The Coremark library is very useful to those wanted to evaluate the performance and energy efficiency of a particular SoC. It allows users to compare different silicon vendors using the same standardized benchmark.

Goals: Benchmarking of MCUs without the hassle of porting the benchmarking software, using Zephyr's portability.

Coremark is the industry standard, it's maintained by a non-profit consortium and is small in code size and complexity.

## Dependencies

None

## Revision

d5fad6bd094899101a4e5fd53af7298160ced6ab

## License

There is an Acceptable User Agreement, but it states that "Licensor offers benchmarking software (“Software”) pursuant to an open source license"
```
Apache-2.0
```

volodymyr-bondarchuk, 2023-03-06:
PR #55461 with Coremark bring up created.

ruuddw, 2023-03-08:
> Similar attempt, but as a standalone application: https://github.com/claziss/coremark-pro-zephyr/tree/main

Note that that's for a subset of CoreMark-PRO, that's not the same as the base CoreMark.

galak, 2023-03-08:
I don't think this belongs in Zephyr.  Coremark glue code for Zephyr we should try and upstream to Coremark.  As Coremark is primarily benchmarking the CPU hardware and not RTOS I don't think Zephyr should be including.

carlescufi, 2023-03-15:
Removed the TSC label until we decide how to proceed with this internally at Nordic.

closed 2023-05-19 by carlescufi