zephyrproject-rtos/zephyr #55475: Bluetooth: settings: Possible Bluetooth settings leftovers

MarekPieta, 2023-03-06:
**Describe the bug**
Zephyr Bluetooth settings stores multiple records related to the same bond (e.g. keys, CCC, SC, CF).
In some edge cases, Zephyr Bluetooth settings may contain unused leftovers which would pollute settings area forever without notifying the application (the application may be completely unaware of the problem).
There seem to be no documented way to remove such settings leftovers in application once they appear.

Apart from this, the CCC, SC and CF leftovers may be improperly reused e.g. if the same peer bonds again (the peer could "inherit" the leftover settings data; the settings data are linked to peer relying only on peer's Bluetooth address and used Bluetooth local identity).
The leftover data also occupies space in the RAM arrays (e.g. `sc_cfg` and `struct _bt_gatt_ccc` configurations) after settings are loaded. This could make device unable to store new, valid data for a connected peer (number of records could exceed predefined limits because of the leftovers).

**To Reproduce**

The issue replicates for example if  `bt_unpair` is interrupted by a spurious power down - keys may already be removed, but CCC, SC and CF data could remain.

Steps to reproduce the behavior:
1. Use a sample that enables Bluetooth settings.
    For example, you can replicate the issue with `samples/bluetooth/peripheral_lbs` sample from  https://github.com/nrfconnect/sdk-nrf running on `nrf52840dk_nrf52840`.
2. Add the `sys_reboot` call during `bt_unpair` (right before `bt_gatt_clear` is called in `unpair` function, so that keys are removed and GATT settings data persists).
3. Add `bt_unpair` call to the button press handler - append the following code to the `button_changed` function to trigger bond erase using button 2 from DK:
```
        if (has_changed & button_state & DK_BTN2_MSK) {
                int err = bt_unpair(BT_ID_DEFAULT, NULL);

                if (err) {
                        printk("Unpair operation failed\n");
                } else {
                        printk("Successfully unpaired\n");
                }
        }
```
4. You can use the following code to print the settings leftover data in the sample:
```
static int log_cb(const char *key, size_t len, settings_read_cb read_cb, void *cb_arg, void *param)
{
        printk("k: %s len: %zu\n", key, len);
        /* Sleep is added to prevent flooding logs. */
        k_msleep(100);

        return 0;
}

static int settings_log(void)
{
         return settings_load_subtree_direct(NULL, log_cb, NULL);
}
```
5. Run the sample, connect and bond with a Bluetooth Central and then trigger `bt_unpair` (pressing Button 2 on the DK).
    See that settings contains leftover data after the reboot.

**Expected behavior**
Application needs a way to make sure that Bluetooth settings data is consistent and contains no leftovers:

1. We could introduce an API to let application trigger the Bluetooth settings cleanup or retrieve information about all of the the leftover records (for example get all of the related settings keys).
   Application could manually delete the leftovers.
2. The leftovers may be automatically removed while settings are loaded (for example settings `set` callback could remember the leftovers and we may remove it in the settings `commit` callback).
3. We could introduce a mechanism ensuring that set of related settings records is always modified together.
   For example, we may store an additional settings record informing that the unpair operation for given bond is started and delete it as soon as operation is complete.
   If the operation is interrupted by e.g. a power down, it could be properly finalized on subsequent boot (e.g. when settings are loaded).

I would suggest to follow the 1st proposed approach.
If the leftovers are caused only by some edge cases that are not likely to happen, an application API to trigger settings data cleanup may be a reasonable approach I think (if an application does not use it, there would be no additional memory overhead too).

**Impact**
Leftovers in the settings may lead to issues where there is no space to store new data (both in the settings partition and in the RAM arrays).
"Inheriting" settings data from previously bonded peers may lead to various unexpected behaviors (e.g. receiving spurious indications/notifications).

**Environment:**
Used main (`f68bcdf5b3ea479268593459aa4770059058780e`) branch from https://github.com/nrfconnect/sdk-nrf repository.

MarekPieta, 2023-03-06:
Similar issue related to storing Bluetooth data as multiple entries could occur for `save_id` function. There could be an edge case leading to storing only the new ID, but leaving the old IRK.

jori-nordic, 2023-03-09:
> We could introduce a mechanism ensuring that set of related settings records is always modified together.
For example, we may store an additional settings record informing that the unpair operation for given bond is started and delete it as soon as operation is complete.
If the operation is interrupted by e.g. a power down, it could be properly finalized on subsequent boot (e.g. when settings are loaded).

This could be done by the settings API, ie a `settings_transaction_begin` and `_end` so we can update groups of settings at the same time. If the power is cut, then the settings backend is free to garbage-collect the incomplete data.

Laczen, 2023-03-09:
Please remind that the settings subsystem is not responsible for how a subsystem is using it. If the bluetooth subsystem is leaving garbage in the settings backend it is the responsability of the bluetooth subsystem to clean this up. If the written data is incomplete (some part written, other not) this is the responsability of the subsystem (there is no way for the settings subsystem to figure this out).

Introducing a `settings_transaction_begin` an `_end` is not a good idea as the settings backend can be on a "circular buffer" that would move the `settings_transaction_begin` to the front of the buffer while leaving the `_end` at the end of the buffer. If data is grouped they should be stored together.

de-nordic, 2023-03-13:
> Please remind that the settings subsystem is not responsible for how a subsystem is using it. If the bluetooth subsystem is leaving garbage in the settings backend it is the responsability of the [...]

Agree. There has been also problem reported that after update to app that no longer uses some settings, these are left behind taking storage place. Settings does not know what is useful and what is not, and what is used by which subsystem so it is hard to have one cleanup solution that will not break something up. Subsystems should cleanup after themselves. Updated application should have mechanisms to discard keys, which have been used by previous versions, and are no longer used.

> Introducing a `settings_transaction_begin` an `_end` is not a good idea as the settings backend can be on a "circular buffer" that would move the `settings_transaction_begin` to the front of the [...]

Agree.

> This could be done by the settings API, ie a `settings_transaction_begin` and `_end` so we can update groups of settings at the same time. If the power is cut, then the settings backend is free to garbage-collect the incomplete data.

Maybe use settings for this, for example use `some/path/to/settings/sosb=<num>` for start of settings block, then you write all settings as `some/path/to/settings/<num>/some_keys` and complete it with `some/path/to/settings/<num>/eosb=num`

Then at read you search for `some/path/to/settings/sosb` to get <update_num> and then for `some/path/to/settings/<update_num>/eosb.
If you do not get the `eosb` then you know that data block is incomplete and need to try with lower <update_num> to recover.
If you get the `eosb` then you have all data and can now discard all blocks with lower update number.

alwa-nordic, 2023-03-13:
> Maybe use settings for this, for example use `some/path/to/settings/sosb=<num>` for start of settings block, then you write all settings as `some/path/to/settings/<num>/some_keys` and complete it [...]

I like this. It's does not mesh well with the `struct settings_handler`-API, but we can go over to using `settings_load_subtree_direct`?

Laczen, 2023-03-13:
> I like this. It's does not mesh well with the `struct settings_handler`-API, but we can go over to using `settings_load_subtree_direct`?

In order to do something like this one needs to know what data needs to be grouped, if this is already known it is much simpler to store them together. Introducing such a storage will increase the space needed, as most of the space is needed for the "name" and this is duplicated

Laczen, 2023-03-14:
> I realized: we want to have a data-journaling file-system. We could have a `journal/<transaction_id>/data/some/path` that we would write to to form a transaction. Then we write `journal/transaction_id/commit`. Then we copy over data from the `journal/<transaction_id>/data/` to `/some/path`. On boot fsck, we check each `journal/<transaction_id>/commit`. If exists, we know it's a interrupted copy, and we do the copy again from the start. If it does not, we delete the transaction as failed/incomplete. Once the copy is done, we delete the transaction.

Exactly why do you want (need) a data-journaling file-system ? If you really need it, probably settings is not the best solution for your problem.

alwa-nordic, 2023-03-14:
I'm not fully sure we need data-journaling. I agree we should not jump for the big gun if not needed :P. It was just my idea of a simplistic[^1] way to have transactions like @jori-nordic mentioned. And, I'm not at all insisting that the settings subsystem needs to implement this. It could be our own layer / library. It's better if the settings subsystem is simple.

[^1]: It may be more complicated if we want a read lock for "transactional" reads as well.

alwa-nordic, 2023-03-14:
@jori-nordic Shall we make a manifest of what the Bluetooth subsystem stores in settings currently, and look at what really needs to be atomic?

de-nordic, 2023-03-14:
> 
> EDIT: We would also need a `journal/<transaction_id>/deletes/`.

What will this be used for?

alwa-nordic, 2023-03-15:
> > EDIT: We would also need a `journal/<transaction_id>/deletes/`.
> 
> What will this be used for?

E.g. Deleting an identity.

theob-pro, 2023-06-19:
https://github.com/zephyrproject-rtos/zephyr/pull/59192 was just a first step to fix this issue. There is still work to do.

Sylphe88, 2024-04-18:
I've been running into such issues where no more space is left (for SC data in my case):
E: unable to save SC: no cfg left

It can get quite hard to deal with and to recover from, because it basically prevents further pairing from new devices. In addition to the possible reset during bt_unpair cause, I believe that MAC randomization may also cause side effects with the BT_SETTINGS subsystem, as I tend to get these errors more often when using latest Android or iOS central devices.

de-nordic, 2024-04-18:
> I've been running into such issues where no more space is left (for SC data in my case): E: unable to save SC: no cfg left
>
> It can get quite hard to deal with and to recover from, because it [...]

This is a problem with software leaving keys they no longer use or know about, for various reasons.
Settings can not remove anything, because it does now know who the data belongs to, it just reads and stores.

When software moves away from some keys, the version that does that (and probably two next versions) should search for the keys they no longer use and remove them.

closed 2024-09-14 by github-actions[bot]