zephyrproject-rtos/zephyr #55479: Bluetooth: gatt: Improper check if Bluetooth peer is bonded

MarekPieta, 2023-03-06:
**Describe the bug**
The GATT implementation relies on the `bt_addr_le_is_bonded` function to check if a connected peer is bonded.
The function relies only on `struct bt_keys` presence for a peer with a given Bluetooth address and therefore may lead to false positives if used before connection security is reestablished (someone may spoof address of a bonded peer and try to modify its subscriptions if they do not require encryption nor authentication - e.g. Service Changed subscription may be altered).
Sending Service Changed indication in the `sc_restore` on reconnection should also be delayed until peer reestablishes security (otherwise other device could "steal" the Service Changed indication prepared for a bonded peer by spoofing address of the bonded peer).

**To Reproduce**
The issue could be reproduced with any Bluetooth sample that acts as Bluetooth Peripheral, uses GATT and bonds.

Connect and bond with the sample using a Bluetooth Central (e.g. an Android phone).
Disconnect the Central and remove bonds on Bluetooth Central side.
Reconnect and see that the Central is still marked as bonded on Peripheral side right after the Bluetooth connection is established (e.g. print result of `bt_addr_le_is_bonded` call in `bt_gatt_connected` function).
You can also modify the Service Changed subscription without reestablishing security and modified state is stored in the settings.

**Expected behavior**
The Bluetooth Peripheral device should wait for security reestablishment before peer is actually considered bonded to prevent situations where Bluetooth address of a bonded peer is spoofed.

**Impact**
The issue could lead to modifying subscriptions of a bonded peer (e.g. for Service Changed characteristic) by other device that spoofs bonded peer's Bluetooth address.

alwa-nordic, 2023-03-17:
The easiest fix for this is to have an even stricter version of `config BT_GATT_AUTO_SEC_REQ`. The stricter version should re-establish security for most ATT read/write operations. We could have a new `BT_GATT_PERM_READ_BONDED_NOT_ENCRYPTED` for attributes that want to opt out of this protection.

kruithofa, 2023-05-25:
A fix is being worked on, but discovers some other issues. @theob-pro to create an issue for these other issues.

theob-pro, 2023-10-12:
> The Bluetooth Peripheral device should wait for security reestablishment before peer is actually considered bonded to prevent situations where Bluetooth address of a bonded peer is spoofed.

It seems the specification does not actually agree with this interpretation. Could you show where you got this requirement from?

Failing that, maybe we should bring this up to the Bluetooth SIG for clarification.

MarekPieta, 2023-10-16:
Hi @theob-pro,
I have not found an exact confirmation of this requirement in the specification. Still, If the peripheral does not wait for the security reestablishment, it may e.g. send Service Changed indication to a device that spoofs address of a bonded peer. Missing this indication may lead to quite serious problems on GATT Client side.

alwa-nordic, 2023-10-16:
@MarekPieta I think this needs to go trough SIG. Can you open an erratum ticket there?

The problem is that the insecure behavior follows from the wording of the spec. Making it secure necessitates behavioral changes that have complicated ripple effects. So I think to go forward on this, all the Bluetooth members have to be on-board for a well-defined solution.

theob-pro, 2023-10-18:
Closing as not a bug. See previous comments.

closed 2023-10-18 by theob-pro

MarekPieta, 2023-10-20:
> @MarekPieta I think this needs to go trough SIG. Can you open an erratum ticket there?
>
> The problem is that the insecure behavior follows from the wording of the spec. Making it secure necessitates [...]

If the insecure behavior follows the spec, then I think we may accept it as is. Thanks for the clarification.