zephyrproject-rtos/zephyr #55480: Bluetooth: Specify unit of RSSI in API

Thalley, 2023-03-06:
**Is your enhancement proposal related to a problem? Please describe.**
In `bluetooth.h` there are multiple uses of `rssi`, but the unit is not described. 

The unit of `struct bt_le_scan_recv_info` and `bt_le_scan_cb_t` is in dBm, but is not specified. 

**Describe the solution you'd like**
Specifiy the unit of RSSI in all uses, and if they are not already the same unit, they should be made into the same unit. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

hasheddan, 2023-03-30:
@Thalley I'd be happy to tackle this if still relevant!

Thalley, 2023-03-30:
> @Thalley I'd be happy to tackle this if still relevant!

You are absolutely more than welcome :D

Threephoneoff, 2026-02-21:
Hello @Thalley ,
As far as I can see this issue is still relevant. Can I take it?

Threephoneoff, 2026-03-12:
> Hello @Thalley If this issue is still open and relevant should I work on it as a first-time contributor I would be happy to help 

Sorry for my delayed response! Yes, please go ahead and take this issue!