zephyrproject-rtos/zephyr #55490: Getting compile error in C++ project for bad flags -Wno-pointer-sign, -Werror=implicit-int

yperess, 2023-03-06:
**Describe the bug**
Unable to build C++ projects since `check_set_compiler_property` always passes the check.

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout https://github.com/yperess/zds2023/
2. Follow instructions in README for setup
3. Build the app `west build -p -b tdk_robokit1 app/`
4. See error

**Expected behavior**
App should build

**Impact**
This is a stopper for 2 ZDS sample projects we're putting together and will soon be integrated in Chromium's EC

**Logs and console output**
```
[20/209] Building CXX object modules/pigweed/pw_hdlc/CMakeFiles/pw_hdlc.pw_rpc.dir/rpc_packets.cc.obj
FAILED: modules/pigweed/pw_hdlc/CMakeFiles/pw_hdlc.pw_rpc.dir/rpc_packets.cc.obj 
/home/peress/bin/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++ -DKERNEL -DPB_MAX_REQUIRED_FIELDS=64 -D_ANSI_SOURCE -D_FORTIFY_SOURCE=1 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/peress/workspace/zds2023/pigweed/pw_hdlc/public -I/home/peress/workspace/zds2023/zephyr/include -I/home/peress/workspace/zds2023/build/zephyr/include/generated -I/home/peress/workspace/zds2023/zephyr/soc/arm/atmel_sam/same70 -I/home/peress/workspace/zds2023/zephyr/lib/libc/newlib/include -I/home/peress/workspace/zds2023/zephyr/soc/arm/atmel_sam/common/. -I/home/peress/workspace/zds2023/modules/hal/cmsis/CMSIS/Core/Include -I/home/peress/workspace/zds2023/modules/hal/atmel/include -I/home/peress/workspace/zds2023/modules/hal/atmel/asf/sam/include/same70b -I/home/peress/workspace/zds2023/modules/lib/nanopb -I/home/peress/workspace/zds2023/pigweed/pw_assert_zephyr/public -I/home/peress/workspace/zds2023/pigweed/pw_assert_zephyr/public_overrides -I/home/peress/workspace/zds2023/pigweed/pw_preprocessor/public -I/home/peress/workspace/zds2023/pigweed/pw_bytes/public -I/home/peress/workspace/zds2023/pigweed/pw_containers/public -I/home/peress/workspace/zds2023/pigweed/pw_polyfill/public -I/home/peress/workspace/zds2023/pigweed/pw_span/public -I/home/peress/workspace/zds2023/pigweed/pw_assert/public -I/home/peress/workspace/zds2023/pigweed/pw_polyfill/standard_library_public -I/home/peress/workspace/zds2023/pigweed/pw_status/public -I/home/peress/workspace/zds2023/pigweed/pw_checksum/public -I/home/peress/workspace/zds2023/pigweed/pw_chrono_zephyr/public -I/home/peress/workspace/zds2023/pigweed/pw_chrono_zephyr/public_overrides -I/home/peress/workspace/zds2023/pigweed/pw_function/public -I/home/peress/workspace/zds2023/pigweed/third_party/fuchsia/repo/sdk/lib/fit/include -I/home/peress/workspace/zds2023/pigweed/third_party/fuchsia/repo/sdk/lib/stdcompat/include -I/home/peress/workspace/zds2023/pigweed/pw_chrono/public -I/home/peress/workspace/zds2023/pigweed/pw_interrupt_zephyr/public -I/home/peress/workspace/zds2023/pigweed/pw_interrupt_zephyr/public_overrides -I/home/peress/workspace/zds2023/pigweed/pw_interrupt/public -I/home/peress/workspace/zds2023/pigweed/pw_log_zephyr/public -I/home/peress/workspace/zds2023/pigweed/pw_log_zephyr/public_overrides -I/home/peress/workspace/zds2023/pigweed/pw_log/public -I/home/peress/workspace/zds2023/pigweed/pw_result/public -I/home/peress/workspace/zds2023/pigweed/pw_router/public -I/home/peress/workspace/zds2023/pigweed/pw_rpc/public -I/home/peress/workspace/zds2023/build/modules/pigweed/pw_rpc/pw_rpc.protos/pwpb -I/home/peress/workspace/zds2023/pigweed/pw_protobuf/public -I/home/peress/workspace/zds2023/pigweed/pw_stream/public -I/home/peress/workspace/zds2023/pigweed/pw_varint/public -I/home/peress/workspace/zds2023/pigweed/pw_string/public -I/home/peress/workspace/zds2023/pigweed/pw_sync/public -I/home/peress/workspace/zds2023/pigweed/pw_toolchain/public -I/home/peress/workspace/zds2023/pigweed/pw_sync_zephyr/public -I/home/peress/workspace/zds2023/pigweed/pw_sync_zephyr/public_overrides -I/home/peress/workspace/zds2023/pigweed/pw_thread/public -I/home/peress/workspace/zds2023/pigweed/pw_thread_zephyr/public -I/home/peress/workspace/zds2023/pigweed/pw_thread_zephyr/sleep_public_overrides -I/home/peress/workspace/zds2023/pigweed/pw_tokenizer/public -I/home/peress/workspace/zds2023/pigweed/pw_sys_io/public -Wall -Wextra -Wimplicit-fallthrough -Wno-type-limits -Wpointer-arith -Werror -Wno-error=cpp -Wno-error=deprecated-declarations -Wnon-virtual-dtor -g -fdiagnostics-color=always -fno-common -fno-exceptions -ffunction-sections -fdata-sections -fno-rtti -std=c++17 -Wno-register -fno-strict-aliasing -Os -fcheck-new -imacros /home/peress/workspace/zds2023/build/zephyr/include/generated/autoconf.h -gdwarf-4 -mcpu=cortex-m7 -mthumb -mabi=aapcs --sysroot=/home/peress/bin/zephyr-sdk-0.15.1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/peress/workspace/zds2023/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/peress/workspace/zds2023/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/peress/workspace/zds2023/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/peress/workspace/zds2023=WEST_TOPDIR -MD -MT modules/pigweed/pw_hdlc/CMakeFiles/pw_hdlc.pw_rpc.dir/rpc_packets.cc.obj -MF modules/pigweed/pw_hdlc/CMakeFiles/pw_hdlc.pw_rpc.dir/rpc_packets.cc.obj.d -o modules/pigweed/pw_hdlc/CMakeFiles/pw_hdlc.pw_rpc.dir/rpc_packets.cc.obj -c /home/peress/workspace/zds2023/pigweed/pw_hdlc/rpc_packets.cc
cc1plus: error: command-line option '-Wno-pointer-sign' is valid for C/ObjC but not for C++ [-Werror]
cc1plus: error: '-Werror=' argument '-Werror=implicit-int' is not valid for C++ [-Werror]
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: `main`

** Additional context
https://github.com/zephyrproject-rtos/zephyr/pull/55418 fixes the issue for me

yperess, 2023-03-06:
@tejlmand, I created the separate bug like you asked.

tejlmand, 2023-03-07:
thanks.
Was able to reproduce and investigate, and a fix has been pushed.

PR #55516 (2023-03-07): cmake: prefix local version of return variable
Fixes: #55490
Follow-up: #53124

Prefix local version of the return variable before calling `zephyr_check_compiler_flag_hardcoded()`.

This ensures that there will never be any naming collision between named return argument and the variable name used in later functions when PARENT_SCOPE is used.

The issue #55490 provided description of situation where the double de-referencing was not working correctly.
commit 599886a9 (2023-03-07): cmake: prefix local version of return variable

Fixes: #55490
Follow-up: #53124

Prefix local version of the return variable before calling
`zephyr_check_compiler_flag_hardcoded()`.

This ensures that there will never be any naming collision between named
return argument and the variable name used in later functions when
PARENT_SCOPE is used.

The issue #55490 provided description of situation where the double
de-referencing was not working correctly.
PR #55527 (2023-03-09): [Backport v3.2-branch] cmake: prefix local version of return variable
Backport 599886a9d3eaae98804c0bbc8fdeea9cb7bb3167 from #55516

Fixes #55490
commit 438da4bc (2023-03-09): cmake: prefix local version of return variable

Fixes: #55490
Follow-up: #53124

Prefix local version of the return variable before calling
`zephyr_check_compiler_flag_hardcoded()`.

This ensures that there will never be any naming collision between named
return argument and the variable name used in later functions when
PARENT_SCOPE is used.

The issue #55490 provided description of situation where the double
de-referencing was not working correctly.

(cherry picked from commit 599886a9d3eaae98804c0bbc8fdeea9cb7bb3167)
closed 2023-03-07 by carlescufi