zephyrproject-rtos/zephyr #55521: tests: drivers: fuel_gauge: sbs_gauge: drivers.sbs_gauge_new_api.emulated hangs

nordic-piks, 2023-03-07:
**Describe the bug**
The tests/drivers/fuel_gauge/sbs_gauge/drivers.sbs_gauge_new_api.emulated hangs and fails on timeout

Observed for:
- nrf5340dk_nrf5340_cpuapp_ns

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf5340dk_nrf5340_cpuapp_ns connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/drivers/fuel_gauge/sbs_gauge/ -p nrf5340dk_nrf5340_cpuapp_ns --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-03-05T00:58:53.303Z] [0mERROR   - I: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0mI: Registering 1 emulator(s) for i2c@100[0m
[2023-03-05T00:58:53.303Z] [0mI: Register emulator 'smartbattery@b' at I2C addr 0b[0m
[2023-03-05T00:58:53.303Z] [0m[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.2
 - Commit SHA or Version used:  zephyr-v3.3.0-740-gbc5fbc929bb8

fabiobaltieri, 2023-03-07:
cc @aaronemassey

aaronemassey, 2023-03-07:
I should have a PR for this today. The tests and experimental API haven't been fully validated on hardware tests yet. Only under emulation.

I'll make a PR that excludes hardware boards.

Apologies for the inconvenience!

aaronemassey, 2023-03-07:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/55564

PR #55564 (2023-03-08): test: Filter HW test from prototyped SBS Gauge
Issue #55521 shows hardware tests hanging when run against the Fuel Gauge's SBS gauge that's still being prototyped against emulated boards.

Prevent accidental running of HW tests on the emulated SBS tests until HW tests have been contributed upstream by filtering for qemu and native posix boards.

Fixes #55521
closed 2023-03-08 by carlescufi