zephyrproject-rtos/zephyr #55535: ST kit b_u585i_iot02a I2C clock frequency error

JiiZR, 2023-03-07:
**Describe the bug**
I added HTS221 register reading to blinky. Measured clock speed was 110 KHz and this is a little bit too much. I checked clocking and PLL is used (msi-pll-mode). I also checked dividers and everything seemed OK to my eyes.

Is there a bug at clock configuration or maybe a clock calibration missing ?

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b b_u585i_iot02a_ns samples\basic\blinky
2. west flash
3. boot and measure STMod+pin 7, I2C2_SCL
4. note that HTS221 is OK

**Expected behavior**
Freq should be max 100 KHz

**Impact**
Freq over I2C specification

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain (e.g Zephyr SDK, ...)
 - latest codes from github
[blink.zip](https://github.com/zephyrproject-rtos/zephyr/files/10909330/blink.zip)

erwango, 2023-05-17:
@b_u585i_iot02a Do you see the same issue with b_u585i_iot02a (ie non NS) ?

JiiZR, 2023-05-17:
I'll check b_u585i_iot02a (not today).

JiiZR, 2023-05-22:
Measured clock speed was about 110 KHz. I'am curious, was there any expectation that freq will be different ?

erwango, 2023-05-22:
> Measured clock speed was about 110 KHz. I'am curious, was there any expectation that freq will be different ?

Thanks. No I wasn't expcting a different value, but since you mentionned the _ns target, it was worth checking.

FRASTM, 2023-06-26:
When running the tests/drivers/i2c/i2c_target_api/  on the nucleo_u575zi_q, with  `I2C_BITRATE_STANDARD` which corresponds to the SCL @ 100KHz, the freq is measured at 100KHz

FRASTM, 2023-06-28:
I propose a new timing calculation in PR https://github.com/zephyrproject-rtos/zephyr/pull/59815
when running your sample, I can measure 400kHz for FAST bitrate and 100kHz for standard bitrate.

JiiZR, 2023-06-29:
Freq is now 100 kHz, great. I had to modify file, compiler problems. See zip file and search // text.
[i2c_ll_stm32_v2.zip](https://github.com/zephyrproject-rtos/zephyr/files/11905137/i2c_ll_stm32_v2.zip)

jhedberg, 2023-08-29:
@FRASTM any progress on this? It keeps getting the Stale label. If it's still valid, please remove the Stale label.

FRASTM, 2023-08-31:
> @FRASTM any progress on this? It keeps getting the Stale label. If it's still valid, please remove the Stale label.

see the PR https://github.com/zephyrproject-rtos/zephyr/pull/59815

erwango, 2023-09-22:
@JiiZR Did you try to use [`timings`](https://github.com/zephyrproject-rtos/zephyr/blame/8951bab7e741f3750532d4af68497b9af06a453c/dts/bindings/i2c/st%2Cstm32-i2c-v2.yaml#L23) in order to inject CubeMX computed timing configuration in the I2C config?

erwango, 2023-09-25:
Proposal is to set `timings` property as required.
Then provision a default `timings` table that would cover some common I2C freq/bitrate configuration.
Then driver should return an error if the requested runtime configuration is not covered by `timings` table.

erwango, 2023-09-25:
@JiiZR any update ?

JiiZR, 2023-09-26:
I did not use timings property, it is unknown property to me.

erwango, 2023-09-26:
> I did not use timings property, it is unknown property to me.

Sorry but this type of comment is not really helpful.

closed 2023-12-13 by FRASTM