zephyrproject-rtos/zephyr #55576: Sample build is failing for nrf9160 due to platform_crypto_keys

sgryphon, 2023-03-07:
**Describe the bug**
Build is failing for NRF9160 samples. Including https_client and the net/mqtt sample (with thingy91 overlay).

The error message, which started at at 267da434 (07 Feb 2023 23:38, joakim.andersson@nordicsemi.no) is:

```
CMake Error at /home/sly/ncs/v2.3.0/nrf/modules/tfm/tfm/boards/CMakeLists.txt:76 (target_sources):
  Cannot specify sources for target "platform_crypto_keys" which is not built
  by this project.
```

Building in linux (Ubuntu 22.04), targeting a Thingy:91.

A clear and concise description of what the bug is.
(Note that [Github Discussions](https://github.com/zephyrproject-rtos/zephyr/discussions) are available to first verify that the issue
is a genuine Zephyr bug and not a consequence of Zephyr services misuse.)

This is a regression. I was working on some changes to nrf9160 code and samples, which were working recently, but failed on the current head / when rebasing my work on the head.

It was a recent issue, so relatively quick to manually bisect (didn't bother with a tool, as there were only a few) and find where the error started.

**To Reproduce**
Steps to reproduce the behavior:
1. Get latest version of code (or checkout 267da434)
2. `cd samples/nrf9160/https_client`
3. `west build -p -c -b thingy91_nrf9160_ns`
4. See error

(Also fails with `west build -p -c -b nrf9160dk_nrf9160_ns`)

**Expected behavior**
Build should work. It was previously working at 2229fe5f at 02 Feb 2023 01:23, and earlier.

**Impact**
Samples for nrf9160 won't work.

Changes to nrf9160 code and examples can't be rebased on the main branch for development. 

**Logs and console output**
Build error:

```
CMake Error at /home/sly/ncs/v2.3.0/nrf/modules/tfm/tfm/boards/CMakeLists.txt:76 (target_sources):
  Cannot specify sources for target "platform_crypto_keys" which is not built
  by this project.
```

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 22.04
 - Toolchain: Nordic SDK nrf-sdk v 2.3.99 (i.e. 2.3.0 head)
 - Commit SHA or Version used: 267da434

**Additional context**
There were a few different errors at the commits before this, between the last working commit 2229fe5f at 02 Feb 2023 01:23 and the current error message.

Commit 691209ee at 02 Feb 2023 01:26 has the error:

```
/home/sly/ncs/v2.3.0/nrf/modules/tfm/tfm/boards/src/tfm_platform_system.c:22:10: fatal error: handle_attr.h: No such file or directory
   22 | #include "handle_attr.h"
```      |          ^~~~~~~~~~~~~~~

Error at the next commit 267da434 (until 267da434):

```
9: error: 'CRYPTO_STACK_SIZE' undeclared here (not in a function); did you mean 'S_PSP_STACK_SIZE'?
   75 |         .stack_size                 = CRYPTO_STACK_SIZE,
      |                                       ^~~~~~~~~~~~~~~~~
      |                                       S_PSP_STACK_SIZE
[35/172] Building C object secure_fw/spm/CMakeFiles/tfm_spm.dir/ffm/psa_api.o
ninja: build stopped: subcommand failed.
```

sgryphon, 2023-03-08:
Actually, I'm not even sure if this is the right project. The documentation is a bit confusing, a the nrf-sdk points to this project (Zephyr) for details about contributing.

nordicjm, 2023-03-08:
This is not the right place, please post this on https://devzone.nordicsemi.com/ unless you can reproduce the issue entirely in zephyr without using nRF connect SDK.

joerchan, 2023-03-08:
@sgryphon As said above, this is not the right project to report this issue. sdk-nrf is using devzone for support.
However your problem is most likely fixed by a 'west update'.

closed 2023-03-08 by joerchan