zephyrproject-rtos/zephyr #55590: [CI][test/drivers/flash] NXP flash driver is not called on RT10xx seriese

hakehuang, 2023-03-08:
**Describe the bug**

flash test report flash driver is not initized.

Please also mention any information which could help others to understand
the problem you're facing:
NXP mimxrt1060_evk/mimxrt1015_evk and all RT10xx series platform impacted.

the flash driver init is not called.

**To Reproduce**
Steps to reproduce the behavior:
cd tests/driver/flash
west build -b mimxrt1060_evk
west flash --runner=pyocd

```
*** Booting Zephyr OS build zephyr-v3.3.0-853-g419dda3ed72d ***
Running TESTSUITE flash_driver
===================================================================

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/src/main.c:59: flash_driver_setup: device_is_ready(flash_dev) is false

START - test_read_unaligned_address
 FAIL - test_read_unaligned_address in 0.001 seconds
===================================================================
TESTSUITE flash_driver failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -   0.00% [flash_driver]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.001 seconds
 - FAIL - [flash_driver.test_read_unaligned_address] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

```

**Expected behavior**
Test PASS

**Impact**
flash functions

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v3.3.0-853-g419dda3ed72d

**Additional context**

bisect found the issues  is introduced by below commit

```

8c6137d7a794f84ca963d79ed9529f43ff61eee5 is the first bad commit
commit 8c6137d7a794f84ca963d79ed9529f43ff61eee5
Author: Fabio Baltieri <fabiobaltieri@google.com>
Date:   Tue Feb 21 19:18:37 2023 +0000

    build: sort and compact drivers and subsys CMakeLists and Kconfig

    There are just list of child files to include, right now there's a hint
    of that trying to be alphabetical but then entries have been added with
    various logic, so one has to figure where stuff has to be added.

    Rewrite these to:
    - keep the list alphabetical, for CMakeLists unconditional first,
      conditional after so one just has to select and sort
    - drop the spaces in the Kconfig
    - use a single space between  target and directory
    - rename from "Sub Systems" to "Subsystems" (single word)

    Signed-off-by: Fabio Baltieri <fabiobaltieri@google.com>

 drivers/CMakeLists.txt |  88 +++++++++++------------
 drivers/Kconfig        | 184 +++++++++++++++----------------------------------
 subsys/CMakeLists.txt  |  47 ++++++-------
 subsys/Kconfig         |  61 ++++------------
 4 files changed, 137 insertions(+), 243 deletions(-)
```

hakehuang, 2023-03-08:
@dleach02 , @mmahadevan108 , this issue fails all flash fucntions in RT10xx series

decsny, 2023-03-08:
Same issue as #55258 , fixed in #55504 yesterday

closed 2023-03-08 by decsny

hakehuang, 2023-03-09:
@decsny in this case below driver will have the same problem
memc_mcux_flexspi_aps6408l.c
memc_mcux_flexspi_s27ks0641.c