zephyrproject-rtos/zephyr #55592: Fix the temperature ADC_STM32_Die_Temp_Read value getting incorrect when reading internal temperature from ADC. for nucleo_f429zi in Zephyr 3.2.0

Niranjan-eaton, 2023-03-08:
**Describe the bug**
nucleo_f429zi: incorrect ADC_STM32_Die_Temp_Read temperature value.
Zephyr 3.2.0

nucleo_f429zi.overlay
 /{
	zephyr,user {
		/* adjust channel number according to pinmux in board.dts */
		
		io-channels = <&adc1 18>;

	};
};
 - ...

**To Reproduce**
Steps to reproduce the behavior:
To Build: west build -b nucleo_f429zi ./ --pristine
To Flash: west flash
4. See error
*** Booting Zephyr OS build v3.2.0-rc1-18-g687816ba42c9 ***
STM32 Die temperature sensor test
Current temperature: -173.3 �°C
Current temperature: -126.2 �°C
Current temperature: -128.2 �°C
Current temperature: -128.8 �°C
Current temperature: -128.8 �°C
Current temperature: -126.9 �°C
Current temperature: -126.2 �°C
Current temperature: -126.5 �°C
Current temperature: -126.9 �°C
**Expected behavior**
positive value

**Impact**
High

**Logs and console output**
*** Booting Zephyr OS build v3.2.0-rc1-18-g687816ba42c9 ***
STM32 Die temperature sensor test
Current temperature: -173.3 �°C
Current temperature: -126.2 �°C
Current temperature: -128.2 �°C
Current temperature: -128.8 �°C
Current temperature: -128.8 �°C
Current temperature: -126.9 �°C
Current temperature: -126.2 �°C
Current temperature: -126.5 �°C
Current temperature: -126.9 �°C

**Environment (please complete the following information):**
 - OS: zephyr-v3.2.0
 - Toolchain zephyr-sdk-0.15.2
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2023-03-09:
Do you have the samples/sensor/stm32_temp_sensor/boards/nucleo_f429zi.overlay
For stm32f42x, the temp sensor is connected to ADC1_IN18 channel
```
/ {
	stm-temp {
		compatible = "st,stm32-temp";
		io-channels = <&adc1 18>;
		status = "okay";
	};
};
```
 Then is the console :                                       
`Current temperature: 19.0 �°C    `

FRASTM, 2023-03-15:
running the  west build -p auto -b nucleo_f429zi samples/sensor/stm32_temp_sensor
on the 02eac5cbc2252675d08ccf54f3eee2b132269953
gives a correct value, without any change : the nucleo_f429zi.dts enables the adc1 and die_temp nodes.
`Current temperature: 29.1 �°C`

closed 2023-03-15 by FRASTM