zephyrproject-rtos/zephyr #55595: ci: [test.device][mimxrt1060_evk] [usb.device.usb_disable] disable usb fail

hakehuang, 2023-03-08:
**Describe the bug**
ssertion failed at WEST_TOPDIR/zephyr/tests/subsys/usb/device/src/main.c:96: device_usb_test_usb_disable: (usb_disable() not equal to TC_PASS)

Please also mention any information which could help others to understand
the problem you're facing:

in NXP RT10xx series usb.device.usb_disable keeps fails.

**To Reproduce**
Steps to reproduce the behavior:
cd tests/subsys/usb/device
west build -b mimxrt1060_evk
west flash --runner=pyocd

**Expected behavior**
test pass

**Impact**
use device disable function

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v3.3.0-853-g419dda3ed72d ***
Running TESTSUITE device_usb
===================================================================
D: lock usb_enable_lock mutex
D: Attached
D: set cb, ep: 0x81
D: unlock usb_enable_lock mutex
START - test_usb_dc_api
 PASS - test_usb_dc_api in 0.001 seconds
===================================================================
START - test_usb_dc_api_invalid
E: Wrong endpoint index/address
E: Wrong endpoint index/address
E: Wrong endpoint index/address
E: Wrong endpoint index/address
E: Wrong endpoint index/address
E: Wrong endpoint index/address
E: Wrong endpoint index/address
E: Wrong endpoint index/address
E: Wrong endpoint index/address
E: Wrong endpoint index/address/direction
E: Wrong endpoint index/address/direction
E: Wrong endpoint index/address/direction
E: Wrong endpoint index/address
 PASS - test_usb_dc_api_invalid in 0.040 seconds
===================================================================
START - test_usb_dc_api_read_write
E: Wrong endpoint index/address/direction
E: Wrong endpoint index/address
 PASS - test_usb_dc_api_read_write in 0.007 seconds
===================================================================
START - test_usb_deconfig
 PASS - test_usb_deconfig in 0.001 seconds
===================================================================
START - test_usb_disable
W: Device not attached
E: Failed to disable ep 0x00
E: Failed to disable endpoint 0x00

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/usb/device/src/main.c:96: device_usb_test_usb_disable: (usb_disable() not equal to TC_PASS)
usb_disable() failed
 FAIL - test_usb_disable in 0.023 seconds
===================================================================
TESTSUITE device_usb failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  80.00% [device_usb]: pass = 4, fail = 1, skip = 0, total = 5 duration = 0.072 seconds
 - PASS - [device_usb.test_usb_dc_api] duration = 0.001 seconds
 - PASS - [device_usb.test_usb_dc_api_invalid] duration = 0.040 seconds
 - PASS - [device_usb.test_usb_dc_api_read_write] duration = 0.007 seconds
 - PASS - [device_usb.test_usb_deconfig] duration = 0.001 seconds
 - FAIL - [device_usb.test_usb_disable] duration = 0.023 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v3.3.0-853-g419dda3ed72d

**Additional context**
This cases fails since created

hakehuang, 2023-03-08:
@dleach02 , @mmahadevan108 , this is a long existing issue

dleach02, 2023-03-10:
> @dleach02 , @mmahadevan108 , this is a long existing issue

What changed that caused this to be reported now?

hakehuang, 2023-03-30:
> What changed that caused this to be reported now?

because I skip usb failure before, and after Derek is asking for checking USB, this issue occures to me.

decsny, 2023-06-06:
I found the issue also exists on the RT1050

decsny, 2023-06-06:
The issue is that earlier in the test the `usb_dc_reset` function sets the `dev_state.dev_struct.controllerHandle` to `NULL`, and the ehci0 SDK driver, unlike the LPC3511IP, checks if the controller handle argument is NULL and errors out if so. for some reason this is not an issue for the lpc3511ip  usb driver and so the test passes for it but not the ehci usb platforms. I do not know enough about the USB to know why these things are the case.

PR #59034 (2023-06-09): drivers: usb_mcux: Fix disable and reset functions
1. Fix the reset function to reset correctly.
2. Ensure the controller handle is initialized before calling the SDK functions.

Fixes #55595
closed 2023-06-09 by nashif