zephyrproject-rtos/zephyr #55598: Crash during entry into System OFF from an i2c device driver shutdown actions

oryjkov, 2023-03-08:
**Describe the bug**
I'm trying to enter System OFF mode using the power manager, i.e.
```
    struct pm_state_info off_state {
        PM_STATE_SOFT_OFF, 0, 0
    };

    pm_state_force(0u, &off_state);
    k_msleep(1000);
```

Instead of getting into system off, the device reboots. After attaching a debugger it looks like an assert is getting triggered during sensor Power manager shutdown actions. (These are happening from the idle thread which the assert is not happy about).
[zephyr_crash_stack.txt](https://github.com/zephyrproject-rtos/zephyr/files/10920995/zephyr_crash_stack.txt)

Please also mention any information which could help others to understand the problem you're facing:
 - Using zephyr v3.3.0
 - Using a Xiao BLE board with an I2C sensor (BMP388) attached.
 - CONFIG_ASSERT is enabled and CONFIG_PM_DEVICE 
 - Problem does not happen when the sensor is not configured in the device tree.

**To Reproduce**
Judging by the stack trace, it should be possible to reproduce this with any board that has an I2C device that implements power management and issues some i2c transactions during shutdown. In my case it is happening regularly and if I disable the sensor System Off happens without an issue.

**Expected behavior**
The i2c device should go to low power mode and system off should be entered.

**Impact**
Unable to perform System Off which is critical for my application.

**Logs and console output**
I think the stack trace should be sufficient, but I can provide more info if needed.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain  Zephyr workspace application
 - Commit SHA or Version used https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v3.3.0

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

anangl, 2023-05-09:
This is not a problem related to the nRF platform but to the device power management in general. The crash occurs because the sensor driver needs to send a [control value](https://github.com/zephyrproject-rtos/zephyr/blob/92f0b5490cce90ef845c09cfed90b581cabf1ec5/drivers/sensor/bmp388/bmp388.c#L442) to the sensor over the I2C bus to put it to sleep. This involves waiting on a semaphore for the transaction to complete. And that is not allowed in the idle thread from which `pm_system_suspend()` and in consequence `pm_suspend_devices()` is called. Similar problem can occur for any driver that needs to perform some communication etc. (and wait for its completion) in order to suspend the handled device. And I don't see an easy solution for this. It seems that suspending of devices needs to be reworked a bit.
@nashif  @ceolin @gmarull Could you comment on that, advise something?

carlescufi, 2023-05-10:
Thanks @gmarull. Closing this issue, but ideally we'd improve the documentation so that the use case is well-documented and other users do not fall into this issue.

closed 2023-05-10 by carlescufi