zephyrproject-rtos/zephyr #55605: Intel CAVS 15/25: tests/subsys/input/api/ crashing board

majunkier, 2023-03-08:
**Describe the bug**
After calling tests/subsys/input/api/ all other tests failing. For run other tests need to restart board.

**To Reproduce**
```
-W --hardware-map /home/ztest/cavs/cavs.map
--device-testing
-T tests/subsys/input/api/
-x=CONFIG_BOOT_DELAY=500
-v --no-skipped-report
```

**Expected behavior**
Tests pass.

**Logs and console output**
```
INFO    - 1/2 intel_adsp_cavs25         tests/subsys/input/api/subsys.input.api.synchronous  ERROR Device issue (Flash error?) (device 0.587s)
INFO    - 2/2 intel_adsp_cavs25         tests/subsys/input/api/subsys.input.api.thread      ERROR Device issue (Flash error?) (device 0.589s)
```

**Environment (please complete the following information):**
OS: Linux
Toolchain: Zephyr SDK
Latest main (419dda3)

fabiobaltieri, 2023-03-14:
Hey tried to run this on a board:

```
$ west build -p -b nrf52dk_nrf52832 tests/subsys/input/api
$ west flash
```

seems to run fine. The test should only require the iterable sections stuff to work, is that platform doing something odd with the linker?

fabiobaltieri, 2023-03-19:
Could you try to see if this fixes the problem by any chance? https://github.com/zephyrproject-rtos/zephyr/pull/55967

nashif, 2023-04-11:
@majunkier  is this still an issue?

majunkier, 2023-04-27:
tests/subsys/input/api/subsys.input.api.synchronous - it's working.

tests/subsys/input/api/input.api.thread - building is fine, when we execute :

`START - test_sequence_thread
report: 0
report: 1
input_cb_unfiltered: 0
report: 2
input_cb_filtered: 0
report: 3
...
report: 15
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/input/api/src/main.c:75: input_api_test_sequence_thread: (ret not equal to -ENOMSG)
ret: 0
FAIL - test_sequence_thread in 0.002 seconds
)`

Toolchain: Zephyr 0.16.0 
OS: Ubuntu 20.04.6 LTS

AmberHibberd, 2023-05-18:
Hi @majunkier any update on this?

nashif, 2023-05-18:
more readable output of the failure:

```
*** Booting Zephyr OS build zephyr-v3.3.0-4093-g5781478fd5c4 ***
Running TESTSUITE input_api
===================================================================
START - test_sequence_thread
report: 0
report: 1
input_cb_unfiltered: 0
report: 2
input_cb_filtered: 0
report: 3
report: 4
report: 5
report: 6
report: 7
report: 8
report: 9
report: 10
report: 11
report: 12
report: 13
report: 14
report: 15
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/input/api/src/main.c:75: input_api_test_sequence_thread: (ret not equal to -ENOMSG)
ret: 0
FAIL - test_sequence_thread in 0.002 seconds
===================================================================
TESTSUITE input_api failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [input_api]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.002 seconds
- FAIL - [input_api.test_sequence_thread] duration = 0.002 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 4453d3d0b465075ce77101694eb64bf0
PROJECT EXECUTION FAILED
```

so, it it is not crashing, is it? This is just a test failing.

nashif, 2023-05-18:
test passes if we just use one core (i.e. CONFIG_SMP=n)

closed 2023-08-13 by github-actions[bot]