zephyrproject-rtos/zephyr #55606: esp32 blinky does not build after fresh install

nbxmike, 2023-03-08:
This appears to be very similar to the error reported in

https://github.com/zephyrproject-rtos/zephyr/issues/30315

After a fresh install on Ubuntu 22.10 I received the following error:
` /home/mike/zephyrproject/zephyr/include/zephyr/device.h:83:41: error: '__device_dts_ord_DT_N_ALIAS_led0_P_gpios_IDX_0_PH_ORD' undeclared here (not in a function)
   83 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_,   ##dev_id) `

To reproduce:

1. Follow the instructions at:
https://docs.zephyrproject.org/3.2.0/develop/getting_started/index.html

2. Obtain all esp32 blobs:
west blobs fetch hal_espressif

3. Attempt a build: 
west build -p always -b esp32  samples/basic/blinky

I apologize, I have only just started to use Zephyr on this hardware and do not have a fix, which I presume would be a minor update to the device tree.  I have tried with or without the second step above, the results do not change.

Complete output was as follows:
`(.venv) mike@serafina:~/zephyrproject/zephyr$ west build -p always -b esp32  samples/basic/blinky
-- west build: making build dir /home/mike/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/mike/zephyrproject/zephyr/samples/basic/blinky
-- CMake version: 3.24.2
-- Found Python3: /home/mike/zephyrproject/.venv/bin/python3.10 (found suitable exact version "3.10.7") found components: Interpreter 
-- Cache files will be written to: /home/mike/.cache/zephyr
-- Zephyr version: 3.3.99 (/home/mike/zephyrproject/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: esp32
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (/home/mike/bin/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/home/mike/bin/zephyr-sdk-0.15.2)
-- Found Dtc: /home/mike/bin/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/mike/zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts
-- Generated zephyr.dts: /home/mike/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/mike/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/mike/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/mike/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/mike/zephyrproject/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration '/home/mike/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to '/home/mike/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/mike/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/mike/bin/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mike/zephyrproject/zephyr/build
-- west build: building application
[1/185] Preparing syscall dependency handling

[5/185] Generating include/generated/version.h
-- Zephyr version: 3.3.99 (/home/mike/zephyrproject/zephyr), build: zephyr-v3.3.0-1-g5cb53ee1e962
[154/185] Performing configure step for 'EspIdfBootloader'
-- Found Git: /usr/bin/git (found version "2.37.2") 
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/mike/bin/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/mike/bin/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
[... 161 lines cut ...]
 3912 |         a1 ## a2 ## a3 ## a4 ## a5 ## a6 ## a7
      |         ^~
/home/mike/zephyrproject/zephyr/include/zephyr/devicetree/gpio.h:164:9: note: in expansion of macro 'DT_PHA_BY_IDX'
  164 |         DT_PHA_BY_IDX(node_id, gpio_pha, idx, pin)
      |         ^~~~~~~~~~~~~
/home/mike/zephyrproject/zephyr/include/zephyr/drivers/gpio.h:317:24: note: in expansion of macro 'DT_GPIO_PIN_BY_IDX'
  317 |                 .pin = DT_GPIO_PIN_BY_IDX(node_id, prop, idx),                 \
      |                        ^~~~~~~~~~~~~~~~~~
/home/mike/zephyrproject/zephyr/include/zephyr/drivers/gpio.h:352:9: note: in expansion of macro 'GPIO_DT_SPEC_GET_BY_IDX'
  352 |         GPIO_DT_SPEC_GET_BY_IDX(node_id, prop, 0)
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/home/mike/zephyrproject/zephyr/samples/basic/blinky/src/main.c:20:40: note: in expansion of macro 'GPIO_DT_SPEC_GET'
   20 | static const struct gpio_dt_spec led = GPIO_DT_SPEC_GET(LED0_NODE, gpios);
      |                                        ^~~~~~~~~~~~~~~~
/home/mike/zephyrproject/zephyr/include/zephyr/devicetree.h:230:25: note: in expansion of macro 'DT_CAT'
  230 | #define DT_ALIAS(alias) DT_CAT(DT_N_ALIAS_, alias)
      |                         ^~~~~~
/home/mike/zephyrproject/zephyr/samples/basic/blinky/src/main.c:14:19: note: in expansion of macro 'DT_ALIAS'
   14 | #define LED0_NODE DT_ALIAS(led0)
      |                   ^~~~~~~~
/home/mike/zephyrproject/zephyr/samples/basic/blinky/src/main.c:20:57: note: in expansion of macro 'LED0_NODE'
   20 | static const struct gpio_dt_spec led = GPIO_DT_SPEC_GET(LED0_NODE, gpios);
      |                                                         ^~~~~~~~~
ninja: build stopped: subcommand failed.
`

rftafas, 2023-03-08:
This board has no LED. Try this instead: https://www.youtube.com/watch?v=Z_7y_4O7yTw&

Note this:

>Note
>Blinky is compatible with most, but not all, [Supported Boards](https://docs.zephyrproject.org/3.2.0/boards/index.html#boards). If your board does not meet Blinky’s [Requirements](https://docs.zephyrproject.org/3.2.0/samples/basic/blinky/README.html#blinky-sample-requirements), then [Hello World](https://docs.zephyrproject.org/3.2.0/samples/hello_world/README.html#hello-world) is a good alternative.

The actual problem is that it shouldn't be listed as compatible. Maybe this is the proper fix for this issue (removal from supported list).

sylvioalves, 2023-03-08:
You can check the sample README file to get more information if needed.    
For this scenario, the raw `esp32` board as pointed above has no LED, which needs an overlay file as below `esp32.overlay`:

```c
/ {
	aliases {
		led0 = &myled0;
	};

	leds {
		compatible = "gpio-leds";
		myled0: led_0 {
			gpios = <&gpio0 13 GPIO_ACTIVE_LOW>;
		};
	};
};
```
Set the LED pin accordingly:
`gpios = <&gpio0 13 GPIO_ACTIVE_LOW>;`

closed 2023-03-08 by sylvioalves

nbxmike, 2023-03-09:
Thank you, I apologize for this distraction.

isabyr, 2024-05-22:
I have the same problem, I've added esp32_devkitc_wroom.overlay to the blinky project as suggested @sylvioalves but the build error remains, looks build ignores the overlay file.
Then I've made changes to `boards/espressif/esp32_devkitc_wroom/esp32_devkitc_wroom_procpu.dts` and now it compiles without errors.
```
--- a/boards/espressif/esp32_devkitc_wroom/esp32_devkitc_wroom_procpu.dts
+++ b/boards/espressif/esp32_devkitc_wroom/esp32_devkitc_wroom_procpu.dts
@@ -19,6 +19,14 @@
                i2c-0 = &i2c0;
                sw0 = &button0;
                watchdog0 = &wdt0;
+               led0 = &onboard_led;
+       };
+
+       leds {
+               compatible = "gpio-leds";
+               onboard_led: led_0 {
+                       gpios = <&gpio0 2 GPIO_ACTIVE_LOW>;
+               };
        };
 
        buttons {
```
Is this normal behavior?

rftafas, 2024-05-22:
Yes, it is.
DTS changed after Hardware Model V2 and you need to specify the CPU.

sylvioalves, 2024-05-22:
> I have the same problem, I've added esp32_devkitc_wroom.overlay to the blinky project as suggested @sylvioalves but the build error remains, looks build ignores the overlay file. Then I've made [...]

Yes. And as mentioned by rftafas, the blinky sample code requires `esp32_devkitc_wroom_procpu.overlay` as overlay file.

isabyr, 2024-05-22:
@rftafas @sylvioalves Finally fixed the problem, thank you very much!