zephyrproject-rtos/zephyr #55623: Some Bluetooth host qualification test cases fail due to regression in LLCP

sjanc, 2023-03-09:
**Describe the bug**

Some PTS tests require iteration of  "connect-> some GATT operations -> disconnect" and those are currently failing due to issues with connection creation.

It looks like third connection attempt is never completed (no connection complete event on both sides), this is regardless of CONFIG_BT_MAX_CONN value.

I bisected this and first bad commit is 
commit 6dbc446342994c53e875016a0e03d35bdf9d31ac
Author: Erik Brockhoff <erbr@oticon.com>
Date:   Thu Dec 1 16:24:38 2022 +0100
    Bluetooth: controller: use RX node piggy-back for NTF when possible

Revering it on top of current main fix the issue.

**To Reproduce**
Execute SM/CEN/SCJW/BI-01-C PTS qualification test case.

**Expected behavior**
Connection attempt is successful.

**Impact**
This currently affect host qualification (but likely is a general problem affecting other applications)

erbr-ot, 2023-04-20:
The commit mentioned was reverted by https://github.com/zephyrproject-rtos/zephyr/pull/55633 
New/fixed solution is WIP

kruithofa, 2023-04-20:
@sjanc  is this bug still present with current main?

sjanc, 2023-04-20:
Reverting fix the regression so I think it can be closed

closed 2023-04-20 by sjanc