zephyrproject-rtos/zephyr #55631: RFC: Better support flashing multiple applications/cores using sysbuild

nordicjm, 2023-03-09:
## Introduction

This introduces an improvement to `west flash` when being used with system to support multiple cores in a single SoC or flashing when things like security features are enabled, which presently will fail. The idea is to enhance `west flash` to reset boards only when flashing has been complete and to allow having arguments specified that should not be ran more than once per set of boards.

### Problem description

Before sysbuild, all zephyr images needed to be build separately and then flashed, because of the order of building things, this was not usually an issue (e.g. someone would build a bootloader for their board before building an application which used mcuboot and flash them in that order) but with the introduction of sysbuild, images are now flashed as part of a single process and the same flash runner mechanism is used (connect to device, recover/erase, flash, reset) for each image. This presents 2 problems:

1. Image flashing order can be in any order, it should not be assumed that images will always be set so that they are in a particular order, this means that for processors that have security features e.g. code protection, if an image is flashed which is not bootable and the target device is rebooted at the end of the flashing process, will the next images be able to be flashed? In the case of the nRF5340, this is usually a "no"
2. Commands that were traditionally used when only a single image was flashed are supported when flashing multiple images using sysbuild - but these options are applied once per image. This presents a problem if e.g. `--recover` is used with the nrfjprog flash runner because the first image being flashed will recover the device then be flashed, the second image will recover the device, erasing the previously flashed image, then write itself, and so forth, leading to an incomplete or unbootable board.

At present, using `west flash --recover` on an nRF5340 with sysbuild and multiple images will result in an unbootable application. Also at present, flashing a nRF5340 with sysbuild with multiple images will result in the flashing failing after the first target is flashed if it enables (or, rather, does not properly disable) security provisions in the device.

### Proposed change

To address this issue, the follow changes are proposed:

1. To prevent restarting of devices until all images are flashed
2. To group commands for a particular SoC for flash runners into those that can be used for each flash operation (e.g. device ID) and those that should only be ran once (e.g. recover) and only run the once-only commands for the first image flash operation

### Proposed change (Detailed)

For both of these, the proposed change is to alter how the `west flash` script to address both these issues (note that at present it is not possible to add this functionality from a SoC perspective - doing that would require SoC v2: https://github.com/zephyrproject-rtos/zephyr/pull/50305):

1. Boards can have a .yml file which lists the "reset hold" order of board names, when this field is present, west flash will use it to determine when the last image is flashed for those list of boards and reboot them. For example in the nrf5340dk board, there would be 2 groups, 1 for `nrf5340dk_nrf5340_cpuapp` and `nrf5340dk_nrf5340_cpuapp_ns`, and the other for `nrf5340dk_nrf5340_cpunet` - this represents the 2 cores in the nRF5340 SoC.
2. Boards can have a .yml file which lists the flash runner arguments that should be ran only once including a set of a board names that it would apply to. When one of the commands is used for a listed board, it will prevent running that command again for any other boards in the same list during that `west flash` process. An example for the nrf5340dk board is to have `--recover` be linked for 2 groups, 1 as `nrf5340dk_nrf5340_cpuapp` and `nrf5340dk_nrf5340_cpuapp_ns`, and the other group as `nrf5340dk_nrf5340_cpunet`. This would allow the --recover command to work when using `west flash` on a single core with multiple images (e.g. mcuboot and application)

When SoC v2 is introduced, this can also be applied at a SoC level, e.g. for nrf5340-based boards, it does not need to be applied to each board individually because this is a SoC requirement, so all boards using an nRF5340 should be able to just "inherit" the behaviour.

### Dependencies

No dependencies

### Concerns and Unresolved Questions

Should the delayed reset apply to all boards by default with the option to override it?

## Alternatives

There are no alternatives

mbolivar-nordic, 2023-03-10:
@nordicjm thanks for the RFC! I'm curious why you assigned us instead of yourself, though?

mbolivar-nordic, 2023-03-10:
> (note that at present it is not possible to add this functionality from a SoC perspective - doing that would require SoC v2: #50305):

Can you please expand on why, for the record? Especially for SoCs where all the cores are on the same arch, like nRF5340.

As discussed offline, adding a new file format to the board specification feels like it's papering over the symptoms. I can understand that some boards might need some overrides due to specific hardware on them, but it intuitively feels like we ought to be able to solve this within a single SoC and hardware model v1, assuming that the SoC all fits in a single arch, without having to duplicate this logic all over every board directory.

nordicjm, 2023-03-10:
> @nordicjm thanks for the RFC! I'm curious why you assigned us instead of yourself, though?

Should have @'d in a comment and assigned to self, switched this.

> Can you please expand on why, for the record? Especially for SoCs where all the cores are on the same arch, like nRF5340.

The problem is linkage between the boards, if you are using `nrf5340dk_nrf5340_cpuapp` then how do you know there is another core, and how do you know the other core is named `nrf5340dk_nrf5340_cpunet`? Likewise with `bl5340_dvk_cpuapp` having `bl5340_dvk_cpuapp_ns` or `bl5340_dvk_cpunet`, there is no way (with SoC v1) to know at a SoC level the name of another board that is part of the same SoC. This excludes the complexity that would be associated with multiple chips being part of the same board e.g. `nrf9160dk_nrf52840` and `nrf9160dk_nrf9160`, despite those being on the same board, it is assumed at this stage that flashing each of these can be done regardless of the state of the other.

tejlmand, 2023-03-13:
> To group commands for a particular SoC for flash runners into those that can be used for each flash operation (e.g. device ID) and those that should only be ran once (e.g. recover) and only run the once-only commands for the first image flash operation

why do they need to be grouped and why is this needed per SoC ?
So far it seems more is related to runner handling, and less on the SoC.

Initially what we need would be the knowledge of:
- an argument can be provided for all images (for example `--erase`, `--snr`)
- an argument can be provided on / before first image (for example `--recover`, `--erase-all`)
- an argument can be provided on / after last  image (for example `--reset`)

this means that a user invoking:
```
$ west flash --recover --reset --snr 1234
```
then `--reset` is applied for final image, `--recover` is applied on first image, `--snr` is applied for all images.

That should be possible to inform in the dedicated binary runner.
Loose thought, a custom action indicating this (instead of a `store_true`) could then be used to specify when the option should be applied if given:
 https://github.com/zephyrproject-rtos/zephyr/blob/516b6730c6ec0b191ab50f5d6866a65cfbe5d1ca/scripts/west_commands/runners/nrf_common.py#L71-L72
For example, first, last, all.

The invocation of the dedicated runner can then pass information regarding current image / total images here:
https://github.com/zephyrproject-rtos/zephyr/blob/516b6730c6ec0b191ab50f5d6866a65cfbe5d1ca/scripts/west_commands/run_common.py#L173-L174

If the above `west flash` only contains a single `image`, that 1 of 1 is passed down, and `--recover`, `--reset`, and `--snr` are all applied on that single invocation.

Has such approach been considered, and if it has, what was the reason for considering such approach not meeting the requirement ?

nordicjm, 2023-03-13:
> why do they need to be grouped and why is this needed per SoC ?

Ideally because all nRF5340s are the same, at present it must be done at a board level.

> then --reset is applied for final image, --recover is applied on first image, --snr is applied for all images.
> 
> That should be possible to inform in the dedicated binary runner.

Not quite, for the nrf5340 `--recover` on the application core clears the application core, `--recover` on the network core clears both application and network cores, the nRF5340 is probably the odd soc out with this behaviour - this probably won't be a problem when https://github.com/zephyrproject-rtos/zephyr/issues/53650 is implemented. One can also not assume that a --reset can only be done on the last image, if for example the flash order is application core then network core - the final `--reset` will not reset the application core.

tejlmand, 2023-03-13:
> Ideally because all nRF5340s are the same, at present it must be done at a board level.

I was asking in the relation to the runner level, which is even above the board and SoC.

>  for the nrf5340 --recover on the application core clears the application core, --recover on the network core clears both application and network cores

Correct, but `--recover`  would still need to happen at the beginning.
Doing a recover after anything but the first image is bound to erase something which shouldn't have been erased.

Note, sysbuild is still responsible for the ordering of images when creating the `domains.yml`.

> the final --reset will not reset the application core.

Correct, but such ordering is the responsibility for sysbuild.
Just like user is responsible today when building for nRF5340dk, like a wireless sample:
- build an rpmsg for netcore
- build mcuboot for app core
- build app for app core
- recover netcore (if APPPROTECT is on)
- flash rpmsg on netcore
- flash mcuboot on appcore
- flash app on appcore
- reset appcore (which also resets netcore)

so if sysbuild ensures the order, instead of the user, then the simple approach I specified should be sufficient.
If appcore images must be flashed last, then just like that is the responsibility of the user today, then perhaps that should be the responsibility of sysbuild, and not the underlying flash tool (the west flash runner).
Ideally the runner should know as little as possible about image ordering and sysbuild as possible.

nordicjm, 2023-04-11:
> Has such approach been considered, and if it has, what was the reason for considering such approach not meeting the requirement ?

This then means that every runner would need its own (generally, fully duplicated) code and logic for handling all of this. Every runner would need code along the lines of
```
if (image == last_image) {
    reset-command;
}
```
Which seems very redundant and a maintenance burden