zephyrproject-rtos/zephyr #55641: Socket Timeout Ignored

clamattia, 2023-03-09:
**Describe the bug**
Socket timeouts are ignored. BSD-socket timeouts are supported via `setsockopt` but the send timeout is ignored. Thus the application of an unknowing developer might hammer the socket with `send`-calls in a busy loop.

From a discussion with @rlubos in the discord it came out, that the likely reason might be https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/sockets/sockets.c#L672 . `send_check_and_wait` is supposed to wait for the timeout but immediately returns:
```
if (!K_TIMEOUT_EQ(timeout, K_FOREVER)) {
	goto out;
}
```
Instead the author might have intended:
```
if (K_TIMEOUT_EQ(timeout, K_NO_WAIT)) {
	goto out;
}
```

Additionally it would be nice to have a regression test for this.

This is likely linked to other issues related to socket send. For example civetweb's need for polling on the sockets.

AndreyDodonov-EH, 2023-03-09:
@clamattia Very good catch.

Additionally, `timeout` in `send_check_and_wait` is ignored completely afterwards.
If it was `K_FOREVER`, however, then `buf_timeout` is used, which is a hard-coded to be 10 seconds.
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/sockets/sockets.c#L664

As for civetweb, it's not because of Zephyr, it's more generic.
Also it's dropped, unfortunatelly, even though after couple of fixes in our fork of Civetweb (mainly thread fixes) and after fixes of POSIX in Zephyr it works perfectly fine for us.
Pity it was dropped.

clamattia, 2023-03-13:
If I read posix correctly, for `K_FOREVER` it is required to actually wait forever:

https://pubs.opengroup.org/onlinepubs/9699919799/functions/send.html
"If space is not available at the sending socket to hold the message to be transmitted, and the socket file descriptor does not have O_NONBLOCK set, send() shall block until space is available. "

PR #55775 (2023-03-20): net: sockets: Fix SO_SNDTIMEO and SO_RCVBUF options
SO_SNDTIMEO socket option did not configure TX timeout on a socket properly. When testing the fix, it turned out that also SO_SNDBUF/SO_RCVBUF socket options were broken, not modifying the unerlying TCP window sizes properly.

This PR fixes the above issues. First two commits address the problem with SO_RCVBUF and adds new tests that verify that TCP windows are affected properly when changing SO_RCVBUF/SO_SNDBUF. 

The remaining commits, fix the TX timeouts on a socket, if configured, and add tests that verify this functinoality.

Fixes #5564 [... cut]