zephyrproject-rtos/zephyr #55653: Bluetooth examples not run on esp32c3 based boards.

trembel, 2023-03-09:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/54131

<div type='discussions-op-text'>

<sup>Originally posted by **icatorze** January 26, 2023</sup>
No one bluetooth examples run on esp32c3. Application hangs on line: register_chipv7_phy(init_data, cal_data, PHY_RF_CAL_FULL); of this phy_init.c file.

Other examples like blinky and applications that use I2C run normally. 

Are there limitations to run bluetooth apps over esp32c3?</div>

stephanosio, 2023-03-13:
Please do not create issues from a Q&A discussion.

closed 2023-03-13 by stephanosio

twilfredo, 2023-03-18:
> ### Discussed in #54131
> 
> Originally posted by **icatorze** January 26, 2023 No one bluetooth examples run on esp32c3. Application hangs on line: register_chipv7_phy(init_data, cal_data, PHY_RF_CAL_FULL); of this phy_init.c file.
> 
> Other examples like blinky and applications that use I2C run normally.
> 
> Are there limitations to run bluetooth apps over esp32c3?

I think there's is a regression, the bluetooth samples worked when I tested on v3.1.0

commit afe57dd3 (2023-03-27): west.yml: update hal_espressif to bring ESP32S3 BLE support

This update includes the following:
* Add ESP32-S3 support
* Modify BT heap to use system heap instead of custom heap,
which adds ~13kB of RAM to application.

Fixes #55653 #54131