zephyrproject-rtos/zephyr #55660: docs: Provide detail about debugging qemu

alevkoy, 2023-03-09:
**Describe the bug**
[The application development guide](https://docs.zephyrproject.org/latest/develop/application/index.html#application-debugging) describes how to start a debugging session for an application running on a QEMU host. It would be easier to follow the instructions if it provided additional details about running the debug server and accessing the console output.

In particular:
* `ninja debugserver` won't work unless there is a `qemu-fifo` file in the build directory, and you have to create one. (If I wasn't supposed to have to do that, that might be my error or a separate issue.)
* Having attached to the debug server, you can set breakpoints in the test and do various other things. Neither `gdb` nor the server will show the console output, so if you just `continue` after attaching, it will appear as though nothing happened.
* You can view the console output during the test with something like `tail -f qemu-fifo`.

**Impact**
I had to spend a couple hours messing around to debug a test on qemu usefully.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.2
 - Commit SHA or Version used: bc5fbc929bb805e3c562c7280dd02a6f9f1f9087

carlescufi, 2023-03-10:
Hey @alevkoy thanks for the issue. This looks like a very good first issue to fix, would you like to take a stab at a Pull Request?

alevkoy, 2023-03-10:
Yeah. I hope to get a PR up in the next few days.

PR #55839 (2023-04-04): Organize debugging documentation
Add some detail about debugging with QEMU. Streamline/clean up the text slightly.

Also create `qemu-fifo` during the build, so the documented `ninja debugserver` command works. (I realize this isn't really a documentation issue. I can make another issue for this if somebody wants.)

Fixes #55660
closed 2023-04-04 by carlescufi