zephyrproject-rtos/zephyr #55662: Bluetooth TX Power setting on the nucleo_wb55rg board.

2songh, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
When requesting pairing, priority is low.
When distance increases, the connection is lost.
I think Bluetooth TX power is weak.

**Describe the solution you'd like**
I want to increase TX power through setting.

**Question**
I think the default value is 0dBm.
I want to raise this value more than 6dBm.
Can I change and define this value? (ex. Kconfig)

carlescufi, 2023-03-10:
@2songh you know you can send arbitrary HCI commands from your application?
`bt_hci_cmd_send_sync()` is a public API

2songh, 2023-03-13:
I already know that.

[image]
First test, I only changed the param->cmd, param->value[0], param->value[1].
TX Power did not change.

Second test, I found information related to CONFIG.
And I added CONFIG.
CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL=y
CONFIG_BT_HCI_VS_EXT=y

So I want to know which value to change to increase the TX Power.

ex) param->value[0]
      0x18 -> 0x1F

closed 2023-03-15 by 2songh