zephyrproject-rtos/zephyr #55669: Bluetooth test: host/att/mtu_update fails to build w gcc12

aescolar, 2023-03-10:
**Describe the bug**
tests/bluetooth/bsim/host/att/mtu_update  fails to build with gcc 12 due to its new memcpy checks

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja  -DBOARD=nrf52_bsim -DCONF_FILE=prj.conf -DCMAKE_C_FLAGS="-Werror" ${ZEPHYR_BASE}/tests/bluetooth/bsim/host/att/mtu_update 
3. ninja 

**Expected behavior**
It builds without scary warnings/errors

**Impact**
Prevents regression for users using gcc-12

**Logs and console output**
```
FAILED: CMakeFiles/app.dir/../zephyr/samples/bluetooth/mtu_update/central/src/central_mtu_update.c.obj 
ccache /usr/bin/gcc -DKERNEL -DNRF52833_XXAA -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -I${ZEPHYR_BASE}/../zephyrproject/modules/bsim_hw_models/nrf_hw_models/src/nrfx/mdk_replacements -I${ZEPHYR_BASE}/include -I${ZEPHYR_BASE}/bsim_out/tests/bluetooth/bsim/host/att/mtu_update/prj.conf/zephyr/include/generated -I${ZEPHYR_BASE}/soc/posix/inf_clock -I${ZEPHYR_BASE}/boards/posix/nrf52_bsim -I${ZEPHYR_BASE}/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/nrfx_glue -I${ZEPHYR_BASE}/subsys/bluetooth -I${ZEPHYR_BASE}/../zephyrproject/modules/hal/nordic/nrfx -I${ZEPHYR_BASE}/../zephyrproject/modules/hal/nordic/nrfx/drivers/include -I${ZEPHYR_BASE}/../zephyrproject/modules/hal/nordic/nrfx/mdk -I${ZEPHYR_BASE}/modules/hal_nordic/nrfx/. -I${ZEPHYR_BASE}/../zephyrproject/modules/bsim_hw_models/nrf_hw_models/src/HW_models -I${ZEPHYR_BASE}/../zephyrproject/modules/bsim_hw_models/nrf_hw_models/src/nrfx/nrfx_replacements -I${ZEPHYR_BASE}/../zephyrproject/modules/crypto/tinycrypt/lib/include -I${ZEPHYR_BASE}/../bsim/components/libUtilv1/src -I${ZEPHYR_BASE}/../bsim/components/libPhyComv1/src "-Werror" -fno-strict-aliasing -O0 -imacros ${ZEPHYR_BASE}/bsim_out/tests/bluetooth/bsim/host/att/mtu_update/prj.conf/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=${ZEPHYR_BASE}/tests/bluetooth/bsim/host/att/mtu_update=CMAKE_SOURCE_DIR -fmacro-prefix-map=${ZEPHYR_BASE}=ZEPHYR_BASE -fmacro-prefix-map=${ZEPHYR_BASE}/../zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -fprofile-arcs -ftest-coverage -fno-inline -m32 -include ${ZEPHYR_BASE}/arch/posix/include/posix_cheats.h -fno-freestanding -fshort-enums -std=c11 -MD -MT CMakeFiles/app.dir${ZEPHYR_BASE}/samples/bluetooth/mtu_update/central/src/central_mtu_update.c.obj -MF CMakeFiles/app.dir${ZEPHYR_BASE}/samples/bluetooth/mtu_update/central/src/central_mtu_update.c.obj.d -o CMakeFiles/app.dir${ZEPHYR_BASE}/samples/bluetooth/mtu_update/central/src/central_mtu_update.c.obj -c ${ZEPHYR_BASE}/samples/bluetooth/mtu_update/central/src/central_mtu_update.c
${ZEPHYR_BASE}/samples/bluetooth/mtu_update/central/src/central_mtu_update.c: In function ‘discover_func’:
${ZEPHYR_BASE}/samples/bluetooth/mtu_update/central/src/central_mtu_update.c:85:17: error: ‘memcpy’ reading 17 bytes from a region of size 4 [-Werror=stringop-overread]
   85 |                 memcpy(&uuid, BT_UUID_GATT_CCC, sizeof(uuid));
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ${ZEPHYR_BASE}/samples/bluetooth/mtu_update/central/src/central_mtu_update.c:22:
${ZEPHYR_BASE}/include/zephyr/bluetooth/uuid.h:110:52: note: source object ‘<Ub090>’ of size 4
  110 |         ((struct bt_uuid *) ((struct bt_uuid_16[]) {BT_UUID_INIT_16(value)}))
      |                                                    ^
${ZEPHYR_BASE}/include/zephyr/bluetooth/uuid.h:901:9: note: in expansion of macro ‘BT_UUID_DECLARE_16’
  901 |         BT_UUID_DECLARE_16(BT_UUID_GATT_CCC_VAL)
      |         ^~~~~~~~~~~~~~~~~~
${ZEPHYR_BASE}/samples/bluetooth/mtu_update/central/src/central_mtu_update.c:85:31: note: in expansion of macro ‘BT_UUID_GATT_CCC’
   85 |                 memcpy(&uuid, BT_UUID_GATT_CCC, sizeof(uuid));
      |                               ^~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

**Environment**
 - OS: Linux
 - Toolchain: gcc 12.1.0 (Ubuntu's gcc-12 package)
 - Commit SHA: 68955ebdb6d8a9c6598810446fecc316209089e1 (latest HEAD)

aescolar, 2023-03-10:
CC @Thalley @jori-nordic

Thalley, 2023-03-10:
The problem would appear that `uuid` is a `bt_uuid_128`, and `BT_UUID_GATT_CCC` is a `bt_uuid_16`. 

Should be simply enough to declare a 
```c
static struct bt_uuid_16 ccc_uuid = BT_UUID_INIT_16(0);
```
And use that for the CCC

aescolar, 2023-03-13:
@Thalley would you be so kind as to send a PR with the fix?

Thalley, 2023-03-13:
> @Thalley would you be so kind as to send a PR with the fix?

I can add it to my todo list, but won't supply a PR for at least 2 weeks

PR #55724 (2023-03-13): tests: Bluetooth: fix att_mtu warnings on gcc-12
The warnings are correct, we were copying 17 bytes from a 4-byte source.

Fixes #55669
closed 2023-03-13 by carlescufi