zephyrproject-rtos/zephyr #55671: Network stack running out of RX buffers with small TCP packets

ssharks, 2023-03-10:
**Describe the bug**
If the receive queue is not read it should fill up to the maximum receive window and stop there. This should limit ingress traffic on this particular TCP connection, but not impact other parts of the network stack or other connections.
Although if the sender transmits small messages, significantly less than a NET_BUF_DATA_SIZE, the TCP receive cue is able contain much more then receive_window_size / NET_BUF_DATA_SIZE buffers. This has the potential to starve the network stack from receive buffers effectively braking all network functionality.

**To Reproduce**
Steps to reproduce the behavior:
Make a server that transmits small TCP packets and a client that does not read from the buffer.

**Expected behavior**
The communication on that particular TCP connection should be coming to a stop when the TCP receive window is full without impacting other connections.

**Impact**
Knowing the problem, periodically reading the TCP queue likely could solve the issue.

**Environment (please complete the following information):**
 - Platform independent, should be reproducible in a unit test

**Additional information**
Although it should be only running out of network buffers, the TCP receive queue looks to be claiming also net_pkt's, which I whould not expect.

rlubos, 2023-03-10:
Should this be considered as a bug though? It was a conscious design choice that L4 passes net_pkt's received to the upper layer, and the socket holds a net_pkt FIFO. TCP just passes the data packets as they were received in a zero-copy manner. Yes, it has a consequence of hogging net_pkt's until they're consumed at the app layer, however doing it otherwise (i. e. squashing new data into already queued buffers) would have a negative performance impact - I'm really not sure we want to go this way.

Plus, I think that dealing with incoming data at the net_buf level would not be that straightforward, given that net_buf API is not thread-safe. In the current design, net_buf is owned by the net_pkt, and once net_pkt is passed to the application, it takes full ownership over it. There's no concern of the thread safety here. 
If, however, we'd allow exchanging the data over net_buf, we'd need to introduce some extra resource protection mechanism between TCP/socket layer, adding yet another level of complexity.

In any case though, I don't think this issue should be marked as a bug, but rather as an enchancement. And I really think we'd need to discuss this thoroughly first, if we really want to go this way (perhaps next net forum in April?).

CC @jukkar @tbursztyka

ssharks, 2023-03-10:
Thanks for looking into this, it is indeed not a real bug, but it is an easy way to create a denial of service attack on a device this way. The only way I could think of to resolve this is to apply a kind of defragmenting on the rx queue as soon as it starts to consume more then receive_window_size / NET_BUF_DATA_SIZE buffers.
This would indeed need to let go of the packet coupling in the rx buffer. I was not aware of the thread safety issues there.
Discussing that on in the next forum sounds like a good plan.

rlubos, 2023-03-10:
I just wonder if this problem isn't a bit exaggerated. Yes, you can easily run of net_pkt/net_buf's, but only if the application ceases to consume them at the app level. Packets/buffers are typically allocated with 100-200 ms timeout, so IMO a lot of time is given for the app to release buffers.

And even if we somehow optimize it at the TCP level, by squashing the new data when received, we can only do this for TCP, as it's stream protocol. The very same problem would appear in case of UDP sockets (even easier to trigger, as you don't really need to establish a connection beforehand) - if an attacker floods the device with UDP datagrams, and the application would not keep up consuming them, it would also lead to packet/buffer starvation in the system (as we do have a common global pool of buffers/packets). I really don't see a way around this.

jukkar, 2023-03-13:
I think we pondered the idea of avoiding denial of service several years ago but at that time there was lot of other things to do. Anyway, as currently we always can receive more data that we are able to handle, IMHO only way to overcome this is to tell the network device driver to throttle the RX and start to drop packets. The packets would be dropped anyway if we do not have enough net_bufs but when that happens, it might be that something bad has already happened because of this out-of-mem situation. So it would be better if we are able to throttle the RX side before running out of net_bufs.
Not sure how we could implement this, either leave the throttling decision to driver itself (with suitable config), or provide a way to tell driver to throttle or not actively from net stack or from application.