zephyrproject-rtos/zephyr #55679: samples/subsys/zbus/runtime_obs_registration/ fails due to log output on stm32l562e_dk

FRASTM, 2023-03-10:
**Describe the bug**
When running the samples/subsys/zbus/runtime_obs_registration/  on the stm32l562e_dk
it can happen that the test fails

To pass the test, the sample.yaml describes the console output as follows:
```
    harness_config:
      type: multi_line
      ordered: false
      regex:
        - "I: System started"
        - "I: Activating filter"
        - "I: Deactivating filter"
        - "I: Bypass filter"
        - "I: Disable bypass filter"
        - "I:  >-- Raw data fetched"
        - "I:  -|- Filtering data"
        - "I:  --> Consuming data: Acc x=0, y=0, z=0"
        - "I:  --> Consuming data: Acc x=2, y=2, z=2"
        - "I:  --> Consuming data: Acc x=4, y=4, z=4"
        - "I:  --> Consuming data: Acc x=6, y=6, z=6"
        - "I:  --> Consuming data: Acc x=7, y=7, z=7"
        - "I:  --> Consuming data: Acc x=8, y=8, z=8"
        - "I:  --> Consuming data: Acc x=9, y=9, z=9"
        - "I:  --> Consuming data: Acc x=10, y=10, z=10"
```

When the test fails, the reason is found on the output console log : 
```
"System started"
"I: Activating filter"
```
instead of
```
"I: System started"
"I: Activating filter"
```

Even with the LOG_INF("System started"); at the beginning of the main.c,  the output console could be "System started" or ": System started" or "I: System started"

**To Reproduce**
Steps to reproduce the behavior:
1. twister --device-testing --device-serial /dev/ttyACM0 -p stm32l562e_dk -T samples/subsys/zbus/runtime_obs_registration/
2. or by west build -p auto -b stm32l562e_dk samples/subsys/zbus/runtime_obs_registration/
3. see error

**Expected behavior**
test PASSED

**Impact**
Ci failure

**Logs and console output**
```
I:  >-- Raw data fetched
System started
I: Activating filter
I:  >-- Raw data fetched
I:  -|- Filtering data
I:  --> Consuming data: Acc x=0, y=0, z=0
I:  >-- Raw data fetched
I:  -|- Filtering data
I:  --> Consuming data: Acc x=2, y=2, z=2
I:  >-- Raw data fetched
I:  -|- Filtering data
I:  --> Consuming data: Acc x=0, y=0, z=0
I:  >-- Raw data fetched
I:  -|- Filtering data
I:  --> Consuming data: Acc x=4, y=4, z=4
I:  >-- Raw data fetched
I:  -|- Filtering data
I:  --> Consuming data: Acc x=0, y=0, z=0
I: Deactivating filter
I: Bypass filter
I:  >-- Raw data fetched
I:  --> Consuming data: Acc x=6, y=6, z=6
I:  >-- Raw data fetched
I:  --> Consuming data: Acc x=7, y=7, z=7
I:  >-- Raw data fetched
I:  --> Consuming data: Acc x=8, y=8, z=8
I:  >-- Raw data fetched
I:  --> Consuming data: Acc x=9, y=9, z=9
I:  >-- Raw data fetched
I:  --> Consuming data: Acc x=10, y=10, z=10
I: Disable bypass filter
```

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain Zephyr SDK, 0.15.2
 - Commit SHA 02eac5cbc2252675d08ccf54f3eee2b132269953

**Additional context**
When

rodrigopex, 2023-03-10:
@FRASTM Is it necessary to have the "System Started"? I guess, no other board has that. This is not standard, right? If I add that, as a required message, all the other boards' tests will fail.

rodrigopex, 2023-05-15:
@FRASTM, sorry for the delay. It got lost on the other GitHub notifications.

It is an issue unrelated to zbus. I can solve that by replacing the regex, but I think it's worth looking into why this happens on this board. "All" other boards perform that ok.

rodrigopex, 2023-05-16:
> I found a fix #57931 when configuring the usart1-console clock source on HSI oscillator instead of PCLK2 (default)

Great news! Can we close this issue? @FRASTM

closed 2023-05-18 by rodrigopex