zephyrproject-rtos/zephyr #55698: Integrate bsim thru the west manifest

aescolar, 2023-03-10:
**NOTE 1**: This is not a source code integration request, but a **tool**
**NOTE 2**: This tool is already used today extensively by Zephyr, and part of CI. It is part of the docker image. The change here is just to fetch it using west for users convenience.
**NOTE 3**: The ask here for the TSC is a green light to create a new repository in the zephyr-rtos organization which would contain a clone of: https://github.com/BabbleSim/bsim_west

## Origin

The original project is https://github.com/BabbleSim/bsim_west 
This organization contains several repositories, but the one in question would be: https://github.com/BabbleSim/bsim_west 

## Purpose

BabbleSim is a tool used to simulate the BLE environment. It has been used extensively in Zephyr for 5+ years, and is an integral part of the BT subsystem regression testing.

## Mode of integration

The ask is to clone the https://github.com/BabbleSim/bsim_west repository into a new
https://github.com/zephyr-rtos/bsim_west 
This repository contains a west manifest, with a set of projects under the BabbleSim group, which would be disabled by default in Zephyr.
See https://github.com/zephyrproject-rtos/zephyr/pull/55696 
Users who use BabbleSim for their daily work can then chose to enable this group with
`west config manifest.group-filter +babblesim`
and utilize this version or keep using a separate one.

## Maintainership

@aescolar 

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/55696 

## Description

The ask is to create a new repository: https://github.com/zephyr-rtos/bsim_west 
and start it by cloning https://github.com/BabbleSim/bsim_west

Once this is done, the next step will be to modify (slightly) the build and test scripts
for the nrf52_bsim, so that if the environment variables used today to find
BabbleSim are not present, instead of erroring out, it will ask west for the presence 
and path to this project, and if present, it would continue with that.

## License

Apache 2.0  (note that anyhow, this is a tool used for workstation development and testing, and not something that ends up built with Zephyr in the embedded devices.)

aescolar, 2023-03-29:
To clarify the motivation and to hopefully help with the concern of having a manifest which points to a separate organization:
* Here we are adding an optional way of fetching an optional simulation tool.
* There is a commitment to keep support for fetching the simulator like today (i.e. not thru the zephyr manifest, and not necessarily thru west). This will remain a very typical way of getting it, and a necessary way for some more elaborate usecases.
* If the github organization were to disappear, anybody can clone from their local copies into another organization/git host, and keep using it from there (with west or without). And if nobody cares to do that, it means nobody actually needs to use it, and you can continue without it. 

The drive for doing this is just popular demand (i.e. users who want a solution with less steps to both fetch what they need and keep it updated).
That is what the current PR (#55696) targets: From that PR being merged on, which babblesim you need to run Zephyr with, can be fetched with a west update together with the rest of Zephyr modules (if you enable the group).
Users not interested on BT or networking simulation with babblesim, would just not enable that group. Users who fetch babblesim by other means wouldn't either.

nashif, 2023-04-13:
approved by the TSC, closing as done.

closed 2023-04-13 by nashif