zephyrproject-rtos/zephyr #55701: SAM_GMAC appears to drop packets larger than ~500 bytes

TomzBench, 2023-03-10:
**Describe the bug**
When using the atmel,sam-gmac driver, the driver appears to silently discard packets larger than about ~500 bytes

**To Reproduce**
```
# build the echo server example
west build -b atsame54_xpro -s samples/net/sockets/echo_server -p
# connect to shell
picocom -b 115200 /dev/ttyACMX
# in another terminal send some bytes
# 20 bytes (OK)
head -c 20 /dev/zero | sed 's/\x00/a/g` | nc x.x.x.x 4242
aaaaaaaaaaaaaaaaaaaa
# 500 bytes (FAIL)
head -c 500 /dev/zero | sed 's/\x00/a/g` | nc x.x.x.x 4242
```

**Expected behavior**
I would expect the Zephyr OS to receive 500 bytes just as well as 20 bytes. And if I exceed some kind of byte limit then I would expect a log to acknowledge the dropped packet, (and perhaps a comment explaining why)

**Impact**
In order to use the network on this hardware we have to keep our bytes under 500 length which is not really usable outside of small prototypes. ("show stopper")

**Logs and console output**
I've attached logs with ETHERNET_LOG_LEVEL_DBG=y. Happy to provide more

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr 3.2 and Zephyr 3.3
 - 4538-g8da815f94a2a

**Additional context**
We first investigated this in the Zephyr network layer over discord chat but the consensus there was that there was either a network issue on our local testing network filtering the packet, or the driver is not forwarding the packets up the stack. We rule out the local area network by testing the echo server on the native system and it works ok.  For reference, here is link to discord chat: https://discord.com/channels/720317445772017664/733037635194585148/1067800347684376618

And the attacked logs (demonstrating 20 byts (OK) and silence when 500 bytes are received)
[gmac.txt](https://github.com/zephyrproject-rtos/zephyr/files/10943357/gmac.txt)

delbel, 2023-04-24:
I started looking into this last week. Adding GMAC_DCFGR_RXBMS(3) to GMAC_DCFGR fixes the issue for 500 bytes, but it still doesn't work for 1200 bytes. GMAC_DCFGR_RXBMS(3) combined with CONFIG_NET_BUF_DATA_SIZE=256 effectively removes the cap.

TomzBench, 2023-04-24:
@delbel thanks for looking into this.

So it looks like the default is 4KB / 8 which is 512. So when accounting for packet headers adds up to why my packets were dropped.

Out of curiosity, what is expected when an Ethernet driver has no place to put the incoming bytes?  Can the driver not signal to kernel too block or is there a queue mechanism?  For example, if I were to transfer 4KB payloads instead of 512, i expect the same phenamena to happen and that large payloads will still be dropped. Perhaps when this happens we can also add a `LOG_WRN` message to others who run into packet size limitation?

Anyway, 4KB packets are usable for us and is no longer a blocking issue for our team, so we are happy for the fix!

closed 2023-04-25 by carlescufi

delbel, 2023-04-25:
@TomzBench When I increase CONFIG_ETH_SAM_GMAC_BUF_RX_COUNT from 12 to 13 after my RXBMS patch, incoming data is no longer dropped silently. For example `head -c 65535 /dev/zero | sed 's/\x00/a/g' | nc x.x.x.x 4242` results in the following output on my atsame54_xpro:

```
[00:00:15.062,000] <inf> net_echo_server_sample: TCP (IPv4): Accepted connection
[00:00:15.063,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv4)...
--- 8 messages dropped ---
[00:00:15.068,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
--- 23 messages dropped ---
[00:00:15.068,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.069,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 360 bytes
[00:00:15.071,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.071,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 180 bytes
[00:00:15.072,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.073,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.074,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 360 bytes
[00:00:15.075,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.076,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.076,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 360 bytes
[00:00:15.078,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.086,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.109,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.110,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.110,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 360 bytes
[00:00:15.111,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.112,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.113,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 360 bytes
[00:00:15.114,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.115,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.115,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 360 bytes
[00:00:15.117,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.118,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.118,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 360 bytes
[00:00:15.120,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.121,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.121,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 360 bytes
[00:00:15.123,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.124,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.124,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 360 bytes
[00:00:15.126,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.127,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.127,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 360 bytes
[00:00:15.128,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.129,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 1280 bytes
[00:00:15.130,000] <dbg> net_echo_server_sample: handle_data: TCP (IPv4): Received and replied with 195 bytes
[00:01:04.839,000] <inf> net_echo_server_sample: IPv4 TCP: Received 1 KiB/sec
[00:02:05.038,000] <inf> net_echo_server_sample: TCP (IPv4): Connection closed
```

It seems as though CONFIG_ETH_SAM_GMAC_BUF_RX_COUNT=12 should work with CONFIG_NET_BUF_DATA_SIZE=128 because 12 * 128 = 1536 is the max frame size, so I'm not sure why it looks like the last buffer is being ignored. Frames larger than 11 * 128 = 1408 are silently dropped, which makes me think either the GMAC DMA is not copying the last buffer or the driver is not reading the last buffer.

TomzBench, 2023-04-26:
@delbel thanks for digging deeper, good catch.

It looks like the code that checks for enough buffer space is adding a +1 during the assertion check
[image]

But when assignment happens (which allocates for the rx queue's) does not add the same +1
[image]

So perhaps the +1 should be added to the `MAIN_QUEUE_RX_DESC_COUNT` to match the assertion expectation?  Perhaps this could be patches in a separate PR since your PR was merged already

delbel, 2023-04-26:
@TomzBench I had the same thought about adding a +1 to MAIN_QUEUE_RX_DESC_COUNT. I've already confirmed this works and will open a new PR.

PR #57197 (2023-04-25): drivers: ethernet: sam0: Fix receive buffer initialization
The SAM GMAC driver is not utilizing the entire receive buffer, causing large packets to be dropped. Fix it by setting Receiver Packet Buffer Memory Size Select to Full (RXBMS = 3) in the GMAC_DCFGR register.

Fixes #55701
commit 5beb8686 (2023-04-25): drivers: ethernet: sam0: Fix receive buffer initialization

The SAM GMAC driver is not utilizing the entire receive buffer, causing
large packets to be dropped. Fix it by setting Receiver Packet Buffer
Memory Size Select to Full (RXBMS = 3) in the GMAC_DCFGR register.

Fixes #55701
PR #57307 (2023-04-27): drivers: ethernet: sam0: Increase RX buffer count by one
The SAM GMAC driver is not able to use all of the receive buffers concurrently. Frames larger than (buffer size) * (buffer count - 1) are silently dropped. Fix it by adding one to MAIN_QUEUE_RX_DESC_COUNT.

Fixes #55701
commit 83fc7320 (2023-04-27): drivers: ethernet: sam0: Increase RX buffer count by one

The SAM GMAC driver is not able to use all of the receive buffers
concurrently. Frames larger than (buffer size) * (buffer count - 1) are
silently dropped. Fix it by adding one to MAIN_QUEUE_RX_DESC_COUNT.

Fixes #55701