zephyrproject-rtos/zephyr #55718: Samples: TF-M on LPC55S69 broken

mimok, 2023-03-13:
**Describe the bug**
TF-M hangs during platform provisioning on lpcxpresso55s69 (e.g. tfm_ipc sample). Indeed, for an unkown reason, the update of `swap_count` value enters an infinite loop when `FLASH_VerifyProgram()` function is called in the Flash driver. To be more specific problem comes from commit https://github.com/zephyrproject-rtos/trusted-firmware-m/commit/039dbd4f7b53521a479a735d42f9993094a35fc2

Removing this commit solves the problem. Actually, this problem is related to TF-M and not Zephyr OS but at the end Zephyr OS is also affected.

**To Reproduce**
Steps to reproduce the behavior:
1. compile, flash and run `tfm_ipc` sample (or any other TF-M sample) on lpcxpresso55s69

**Expected behavior**
Platform provisioning should be done without any problem.

**Impact**
TF-M is unusable on lpc55s69

**Environment (please complete the following information):**
- Zephyr OS 3.3.0 and TF-M 1.7.0

mimok, 2023-03-13:
In my opinion, the problem comes from `write_otp_nv_counters_flash()` function where `swap_count` is read from flash, incremented and written back onto the flash without erasing it. While this can work on MPS2_AN521 or others ARM boards (because Flash is emulated using RAM) this should not work on MCU implementing real flash.

microbuilder, 2023-03-13:
It's a known issue that the LPC55S69 only works with profile small with TF-M 1.7.0 due to this issue, even if the release notes are a bit vague: https://tf-m-user-guide.trustedfirmware.org/releases/1.7.0.html#known-issues ... I'll try to get a fix upstream for this that we can cherry-pick. I hadn't realized something hadn't already been upstreamed.

microbuilder, 2023-03-13:
@erwango FYI

RcColes, 2023-03-13:
Had a quick look at this - Zephyr seems to be missing `6a38b2b552`, which is a fix to this file but not, I think, to this particular bug. I've had a go at reproducing the issue but I've only got a NUCLEO-L552ZE-Q around - as far as I can tell this has the same flash setup as the STM32l562E_DK, but I don't see any problems (at least on upstream/master).

I'm surprised that Profile Small would affect this, since IIRC it doesn't really change how flash is laid out/programmed.

I'll keep having a look at the file and see if I can spot anything, though without a proper debugging setup it'll be tricky. @mimok If you can provide a dump of the flash state when the bug occurs that'd be really useful

butok, 2023-03-13:
Hello,

The issue has been **located in the TFM common OTP code**, in the flash_otp_nv_counters_backend.c file, the write_otp_nv_counters_flash() function.
The issue is reproduced only during OTP Provisioning, so the flash have to be mass-erased before the provisioning process. 
If the board was provisioned by a no-issue code, without erasing, it is not reproduced.

Quick and rude workaround (flash_otp_nv_counters_backend.c):
The disabled problematic code is marked by the “//DM NXP” comment.
So, LPC55S69 regression test+demo+medium_profile are functional after applying it.

```
enum tfm_plat_err_t write_otp_nv_counters_flash(uint32_t offset, const void *data, uint32_t cnt)
{
    enum tfm_plat_err_t err = TFM_PLAT_ERR_SUCCESS;
    size_t copy_size;
    size_t idx;
    size_t start;
    size_t end;
    size_t input_idx = 0;
    size_t input_copy_size;
    uint32_t *swap_count;
    uint8_t swap_count_buf[TFM_HAL_ITS_PROGRAM_UNIT > 4 ? TFM_HAL_ITS_PROGRAM_UNIT : 4];

    start = round_down(offset, TFM_OTP_NV_COUNTERS_SECTOR_SIZE);
    end = round_up(offset + cnt, TFM_OTP_NV_COUNTERS_SECTOR_SIZE);
    ARM_FLASH_CAPABILITIES DriverCapabilities;
    uint8_t data_width;

    DriverCapabilities = OTP_NV_COUNTERS_FLASH_DEV.GetCapabilities();
    data_width = data_width_byte[DriverCapabilities.data_width];

    if (end > TFM_OTP_NV_COUNTERS_AREA_SIZE) {
        /* Erase is beyond the TFM_OTP_NV_COUNTERS_AREA */
        return TFM_PLAT_ERR_SYSTEM_ERR;
    }

    /* If it's not part of the sectors that are being erased, first erase the
     * sector with the swap_count flag.
     */
    if (end < offsetof(struct flash_otp_nv_counters_region_t, swap_count)) {
        err = erase_flash_region(round_down(TFM_OTP_NV_COUNTERS_AREA_ADDR +
                                 offsetof(struct flash_otp_nv_counters_region_t, swap_count),
                                 TFM_OTP_NV_COUNTERS_SECTOR_SIZE),
                                 TFM_OTP_NV_COUNTERS_SECTOR_SIZE);
        if (err != TFM_PLAT_ERR_SUCCESS) {
            return err;
        }
    }

    err = erase_flash_region(round_down(TFM_OTP_NV_COUNTERS_AREA_ADDR + offset,
                                        TFM_OTP_NV_COUNTERS_SECTOR_SIZE),
                             round_up(cnt, TFM_OTP_NV_COUNTERS_SECTOR_SIZE));
    if (err != TFM_PLAT_ERR_SUCCESS) {
        return err;
    }

#if 0 //DM NXP
    /* Don't write the swap count, as this is done at the end */
    if (end > offsetof(struct flash_otp_nv_counters_region_t, swap_count)) {
        end = round_down(offsetof(struct flash_otp_nv_counters_region_t, swap_count),
                         TFM_HAL_ITS_PROGRAM_UNIT);
    }
#endif //DM NXP   

    for (idx = start; idx < end; idx += copy_size) {
        copy_size = (idx + sizeof(block)) <= end ? sizeof(block) : end - idx;

        err = (enum tfm_plat_err_t)OTP_NV_COUNTERS_FLASH_DEV.ReadData(TFM_OTP_NV_COUNTERS_BACKUP_AREA_ADDR + idx,
                                                 block,
                                                 copy_size / data_width);
[... 84 lines cut ...]
    if (err < 0) {
        return TFM_PLAT_ERR_SYSTEM_ERR;
    }
#else //DM NXP
    /* Restore the is_valid flag */
    if (end < offsetof(struct flash_otp_nv_counters_region_t, swap_count)) {
        err = copy_flash_region(round_down(TFM_OTP_NV_COUNTERS_BACKUP_AREA_ADDR +
                                           offsetof(struct flash_otp_nv_counters_region_t, swap_count),
                                           TFM_OTP_NV_COUNTERS_SECTOR_SIZE),
                                round_down(TFM_OTP_NV_COUNTERS_AREA_ADDR +
                                           offsetof(struct flash_otp_nv_counters_region_t, swap_count),
                                           TFM_OTP_NV_COUNTERS_SECTOR_SIZE),
                                TFM_OTP_NV_COUNTERS_SECTOR_SIZE);
        if (err != TFM_PLAT_ERR_SUCCESS) {
            return err;
        }
    }
#endif //DM NXP
    err = make_backup();
    if (err != TFM_PLAT_ERR_SUCCESS) {
        return err;
    }
    return err;
}
```

mimok, 2023-03-13:
Problem occurs during call to `tfm_plat_otp_init()` so i would think that it is not related with TF-M profile as this call is handled in `bl2_main.c`. In addition, problem also occurs on stm32l562e which don't have the same restriction as lpc55s69, isn't it ?

Hereafter the stack trace i get with lpc55s69, i will try to make a flash dump this afternoon.

```
(gdb) where
#0  write_otp_nv_counters_flash (offset=offset@entry=764, data=data@entry=0x30006cfc, cnt=cnt@entry=4)
    at /home/mgrand/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/common/template/flash_otp_nv_counters_backend.c:321
#1  0x10003218 in create_or_restore_layout ()
    at /home/mgrand/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/common/template/flash_otp_nv_counters_backend.c:536
#2  init_otp_nv_counters_flash ()
    at /home/mgrand/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/common/template/flash_otp_nv_counters_backend.c:197
#3  0x10002710 in tfm_plat_otp_init () at /home/mgrand/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/common/template/otp_flash.c:23
#4  0x10000762 in main () at /home/mgrand/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/bl2_main.c:113
```

microbuilder, 2023-03-13:
> Problem occurs during call to `tfm_plat_otp_init()` so i would think that it is not related with TF-M profile as this call is handled in `bl2_main.c`.

They may not be related, it just came to our attention just before the 1.7.0 release that TF-M was only passing test suites on the LPC55s69 with profile small since most of the testing in CI is done on the various MPSx-ANxxx targets via FVP or QEMU. The profile small requirement is possibly masking other issues. The note was added to the release notes since there wasn't time to resolve and upstream changes before the release. We've mostly been testing on the ST u585 on our part at at Linaro for actual HW die to the larger flash memory, so didn't notice the issue(s) either.

Once the OTP issue is addressed upstream, it's easy enough to test other profiles with the LPC and see what issues remain on that device, though it's a bit constrained for flash size for more complex samples and you can't run certain test suites on it out of the box.

butok, 2023-03-13:
> `tfm_plat_otp_init()`
The issue reproduced & workaround in the write_otp_nv_counters_flash(), as mentioned in the previous reply.

butok, 2023-03-13:
BTW: The issue is valid for all profiles. 
In small one is disabled PS so it does not cause a panic error, but Attestation tests are failed.

mimok, 2023-03-13:
@butok, your workaround seems to bypass the `swap_count` which is used (i believe) to preserve the memory state in case of power failure a better workaround in my opinion is to comme back to the file version before the problematic commit.

@RcColes I don't find the commit `6a38b2b552`. Hereafter a screencapture of the OTP memory content of the STM32l562. Until address 0x080132F0, OTP memory contents are the same whether your commit is applied or not. However 32-bit values at address 0x132F0, 0x132F4 and 0x132FC are incrmented by one after each reboot. Board reboot each time `PC` reaches call to `tfm_arch_free_msp_and_exc_ret()` in `tfm_core_svc_handler()` (indeed `exec_return` is set to 0xfffffffd)

[image]

butok, 2023-03-13:
> in my opinion, is to come back to the file version before the problematic commit.

Agree, but it requires more changes. Better to wait for a normal TFM fix.

mimok, 2023-03-13:
Here after an example of memory dump before calling `OTP_NV_COUNTERS_FLASH_DEV.ProgramData()` at line 434 in `write_otp_nv_counters_flash()` and after. I think there is something wrong with content of addresses 0x80132f0-4. These values are incremented after each reboot.

Hope it will help but i'm suprised stm32l552 does not exhibit the same behavior as stm32l562 does.

```
(gdb) x/32 0x080132F0
0x80132f0:      0x00000003      0x00000002      0xffffffff      0xffffffff
0x8013300:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
0x8013310:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
0x8013320:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
0x8013330:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
0x8013340:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
0x8013350:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
0x8013360:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
(gdb) n
443         err = make_backup();
(gdb) x/32 0x080132F0
0x80132f0:      0x00000003      0x00000002      0x00000000      0x0000001f
0x8013300:      0x000002f0      0x000002f0      0x00000003      0x00000000
0x8013310:      0x300233f8      0x00000001      0xffffffff      0xffffffff
0x8013320:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
0x8013330:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
0x8013340:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
0x8013350:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
0x8013360:      0xffffffff      0xffffffff      0xffffffff      0xffffffff
```

mimok, 2023-03-14:
I have found a first problem. At line 434 of `flash_otp_nv_counters_backend.c`, the size of the buffer passed to `OTP_NV_COUNTERS_FLASH_DEV.ProgramData()` should be `sizeof(swap_count_buf) / data_width` instead of `sizeof(swap_count_buf)`

Now LPC55S69 enters an infinite boot loop just like stm32l562 does (however reason behind this boot loop seems to not be the same for the two boards).

RcColes, 2023-03-14:
@mimok Good spot, and thanks for the info. I'm doing some rewriting now to better catch edge-cases, will let you know when it's available and then we can test it / @microbuilder can push the change to Zephyr

microbuilder, 2023-03-14:
> @mimok Good spot, and thanks for the info. I'm doing some rewriting now to better catch edge-cases, will let you know when it's available and then we can test it / @microbuilder can push the change to Zephyr

I can cherry pick the upstream change, sure.

mimok, 2023-03-14:
I have found a second problem. At line 421/422, `TFM_OTP_NV_COUNTERS_AREA_SIZE` should be replaced by `TFM_OTP_NV_COUNTERS_BACKUP_AREA_ADDR`. Once this is down, stm32l565 tfm_ipc sample works correctly.

In addition, at line 409/4010, I would put `TFM_OTP_NV_COUNTERS_AREA_ADDR` inside call to `round_down()` to improve readability.

A patch file with my modifications is attached to this post. While STM32l562 is now OK, LPC55S69 stills reboot in loop.

[flatsh_otp_nv_counters_backend.patch](https://github.com/zephyrproject-rtos/zephyr/files/10969276/flatsh_otp_nv_counters_backend.patch)

microbuilder, 2023-03-14:
Upstream patch for review: https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/20030

mimok, 2023-03-14:
with this patch stm32l562 works great but lpc55s69 hangs during `provision_assembly_and_test()`. There is something bad somewhere in this code. @butok any idea ?

mimok, 2023-03-14:
I found the problem `<` operator at line 390 of the patched file shall be replaced by '<=':
```
    if (erase_end_offset <= swap_count_erase_start_offset) {
        err = erase_flash_region(TFM_OTP_NV_COUNTERS_AREA_ADDR +
                                 swap_count_erase_start_offset,
                                 TFM_OTP_NV_COUNTERS_SECTOR_SIZE);
        if (err != TFM_PLAT_ERR_SUCCESS) {
            return err;
        }
    }
```
Now, with this modification, code works great on LPC55S69 and STM32l562

mimok, 2023-03-15:
@butok On TF-M review website I see that your test is passing with the vanilla patch. On my side, the boards hangs during provisioning. What kind of test did you do ? My test procedure consists in running zephyr `TFM_IPC` sample on zephyr 3.3.0 with the patch applied.

AndreyButokNXP, 2023-03-15:
Hi @mimok, 
I run the vanila TFM Regression Tests (without Zephyr) inside MCUx SDK.

mimok, 2023-03-15:
Do you use main branch of TF-M or branch 1.7.0 ?

AndreyButokNXP, 2023-03-15:
Main one.

RcColes, 2023-03-15:
It's very interesting that we see such variation between the two branches. @mimok You're correct, I'll put in another patch.

microbuilder, 2023-03-15:
> It's very interesting that we see such variation between the two branches. @mimok You're correct, I'll put in another patch.

Thanks @RcColes https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/20047

microbuilder, 2023-03-16:
> Thanks @RcColes https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/20047

I see this was merged this morning. I'll cherry-pick the two changes in the morning to test them with Zephyr CI, and link to the PRs here.

mimok, 2023-03-16:
Thank you for the feedback.

microbuilder, 2023-03-17:
@mimok Waiting for CI to finish, but manifest update with cherry-picked fixes available here if you want to have a look at it: https://github.com/zephyrproject-rtos/zephyr/pull/55914

mimok, 2023-03-17:
Looks like ok, i will give it a try this evening.

mimok, 2023-03-17:
@microbuilder Everything is working out of box, great! Only strange thing is that the version diplayed in the banner is not the right one:
`*** Booting Zephyr OS build zephyr-v2.5.0-27753-g4e1a458907e3 ***`

microbuilder, 2023-03-23:
@mimok Are you on the Zephyr discord? Wanted to reach out about this PR and TF-M, but don't see your username?

mimok, 2023-03-23:
No but i can join the channel.

PR #55914 (2023-03-24): manifest: trusted-firmware-m: Add OTP flash fixes
Updates TF-M to include various fixes for OTP flash memory handling.

Fixes #55718
closed 2023-03-24 by fabiobaltieri