zephyrproject-rtos/zephyr #55728: wrong internal temperature measured on the nucleo_wl55jc

FRASTM, 2023-03-13:
**Describe the bug**
With the samples/sensor/stm32_temp_sensor/ on the nucleo_wl55jc, the temperature is too low 

Please also mention any information which could help others to understand
the problem you're facing
Reverting the https://github.com/zephyrproject-rtos/zephyr/pull/55507/ and the temperature is correct
```
*** Booting Zephyr OS build zephyr-v3.3.0-1008-g02eac5cbc225 ***                
STM32 Die temperature sensor test                                               
Current temperature: 18.9 �°C 
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_wl55jc samples/sensor/stm32_temp_sensor/
2. west flash 
3. See error

**Expected behavior**
temperature measured around 19°C

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.3.0-1008-g02eac5cbc225 ***                
STM32 Die temperature sensor test                                               
Current temperature: 5.7 �°C      
```                                              

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain  0.15.0
 - Commit SHA  02eac5cbc2252675d08ccf54f3eee2b132269953

FRASTM, 2023-03-13:
data captured (data->raw) from ADC1 channel 12  is  ~890.0 when it should be ~932.0

FRASTM, 2023-03-15:
Moreover,  when the `CONFIG_TIMESLICING=n` , the temp measured by the stm32wl55jc nucleo board is correct 19°C

FRASTM, 2023-03-15:
surprisingly, the same PR gives error during

andyross, 2023-03-15:
Sorry, can you clarify?  Are you saying that this driver regressed after #55507 merged, and that you can recover it by disabling timeslicing?  That sounds... very weird to me.  Obviously timesliced threads shouldn't be involved in an ADC computation, right?

FRASTM, 2023-03-17:
> 
Yes, very strange indeed, I did not see a good reason.

closed 2023-03-28 by carlescufi