zephyrproject-rtos/zephyr #55745: CAN_INIT_PRIORITY default value is too high for spi based driver

JiapengLi, 2023-03-14:
**Is your enhancement proposal related to a problem? Please describe.**

I am stucking at a #34518 related error for a while. 

When testing mcp2515 driver, i found Official example works, but mine is not. Below error keeps showing up, no matter how I change the dts file. 
```
 <err> can_mcp2515: SPI bus spi@40023000 not ready
```

After quite some debugging work (not expierenced), I realize it is caused by initial sequence. 
Add below line to prj.conf could fix the issue.

```
CONFIG_CAN_INIT_PRIORITY=80
```

**Describe the solution you'd like**
make default configuration works for driver like mcp2515 etc.

henrikbrixandersen, 2023-04-28:
I am considering just changing the default from `CONFIG_KERNEL_INIT_PRIORITY_DEVICE` to `CONFIG_APPLICATION_INIT_PRIORITY`. From the help string of `CONFIG_APPLICATION_INIT_PRIORITY`:
> This priority level is for end-user drivers such as sensors and display which have no inward dependencies.

Since CAN controller drivers should have no inward dependencies, this should be fine. Opinions? CC: @martinjaeger, @alexanderwachter, @str4t0m

alexanderwachter, 2023-04-28:
Gut feeling is default to app if MCP2517 else device. I don't see any problems with ISO-TP, because the user has to actively use it.
What about CANopen? Is there some heart beat that starts automatically?

henrikbrixandersen, 2023-04-28:
> What about CANopen? Is there some heart beat that starts automatically?

CANopen must be initialised by the application as well.

henrikbrixandersen, 2023-04-28:
> Initializing all CAN drivers with APPLICATION level somehow feels wrong.

I am not proposing to use `APPLICATION` _level_ for CAN controller driver initialization. The level should be kept as `POST_KERNEL`. I am proposing changing the default _priority_ to `CONFIG_APPLICATION_INIT_PRIORITY`.

PR #58763 (2023-06-01): drivers: can: increase default initialization priority from 50 to 80
Increase the default CAN controller driver initialization priority from KERNEL_INIT_PRIORITY_DEVICE (50) to 80 to allow CAN controllers on SPI busses have their bus initialized prior to initializing the CAN controller driver.

The only in-bound dependency on CAN controller drivers - apart from application level code - is the drivers/net/canbus.c driver, which already defaults to an initialization level of 81.

Fixes: #55745
commit b12eb78f (2023-06-01): drivers: can: increase default initialization priority from 50 to 80

Increase the default CAN controller driver initialization priority from
KERNEL_INIT_PRIORITY_DEVICE (50) to 80 to allow CAN controllers on SPI
busses have their bus initialized prior to initializing the CAN controller
driver.

The only in-bound dependency on CAN controller drivers - apart from
application level code - is the drivers/net/canbus.c driver, which already
defaults to an initialization level of 81.

Fixes: #55745
closed 2023-06-01 by nashif