zephyrproject-rtos/zephyr #55754: QUAD SPI nor flash not working with DMA2 on STM32WB55

pebeoo, 2023-03-14:
**Describe the bug**
We are using "stm32-qspi-nor" driver on DMA1 in our project and is working correclty. But due to some project constraints we want to move this driver to DMA2 but unfortunatelly it is not possible. Driver works correctly on all DMA1 channels but is not working on any DMA2 channel. 

**To Reproduce**
Set dmas for quadspi device to any channel on DMA2 and try to perfrom some communication with flash chip.

**Expected behavior**
QUAD SPI nor flash working with DMA2 channels.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: 27ec69eca183887c6b3948b3e92b8cc4f935532e

FRASTM, 2023-03-14:
@pebeoo could you please detail your device Tree, 
when using the DMA2  : DMAMUX channels 7 to 13 are connected to DMA2 channels 1 to 7
(DMAMUX Request  is 20 for quadspi)

pebeoo, 2023-03-14:
@FRASTM I tried several options for example for quadspi on DMA2 channel 0, using dmamux:

`&quadspi {
    pinctrl-0 = <&quadspi_clk_pa3 &quadspi_bk1_ncs_pd3
                 &quadspi_bk1_io0_pd4 &quadspi_bk1_io1_pd5
                 &quadspi_bk1_io2_pd6 &quadspi_bk1_io3_pd7>;
    pinctrl-names = "default";
    dmas = <&dmamux1 7 20 0x0000>;
    dma-names = "tx_rx";
    status = "okay";
    qspi_nor_flash@0 {
        compatible = "st,stm32-qspi-nor";
    ...`

And using dma2 description:

`&quadspi {
    pinctrl-0 = <&quadspi_clk_pa3 &quadspi_bk1_ncs_pd3
                 &quadspi_bk1_io0_pd4 &quadspi_bk1_io1_pd5
                 &quadspi_bk1_io2_pd6 &quadspi_bk1_io3_pd7>;
    pinctrl-names = "default";
    dmas = <&dma2 1 20 0x0000>;
    dma-names = "tx_rx";
    status = "okay";
    qspi_nor_flash@0 {
        compatible = "st,stm32-qspi-nor";
        ...`

FRASTM, 2023-03-14:
Pin mapping depends on the board hw,
Do you mean that You should have 
```
    dmas = <&dmamux1 1 20 0x480>;
    dma-names = "tx_rx";

```
makes the transfer correct on the dmamux channel 1 (dma1)
and that 
```
    dmas = <&dmamux1 720 0x480>;
    dma-names = "tx_rx";

```
is not correct when dmamux channel 7 (dma2) is selected 

I guess you also enabled dma nodes: 
```
&dma1 {
	status = "okay";
};

&dma2 {
	status = "okay";
};

&dmamux1 {
	status = "okay";
};
```

pebeoo, 2023-03-14:
Yes exacly, both dmas and dmamux status is set to "okay" and everything is working for:

`dmas = <&dmamux1 1 20 0x480>;
    dma-names = "tx_rx";`

And with following settings is not working:

`dmas = <&dmamux1 7 20 0x480>;
    dma-names = "tx_rx";`

Additionally we are using now DMA2 for other devices on other channels.

FRASTM, 2023-03-14:
in that case, I suggest to test like the _octospi_ does with channel 0 (from 0 to 6) of the DMA2 :

```
 &quadspi1 {
            dmas = <&dma2 0 20 0x480>; /* request 20 for QUADSPI1 */
            dma-names = "tx_rx";
};
```

pebeoo, 2023-03-14:
It was tested also that way on all DMA2 channels, it did not work.

FRASTM, 2023-03-14:
> It was tested also that way on all DMA2 channels, it did not work.

I am trying to build a sample for a quadspi on another board (no stm32wb55 with quadspi available)

pebeoo, 2023-03-15:
During debugging we found one small think and it could be the case. In file flash_stm32_qspi.c DMA1 is hardcoded as DMA instance in line 1125:

`#ifdef CONFIG_DMAMUX_STM32
	/* HAL expects a valid DMA channel (not DAMMUX) */
	/* TODO: Get DMA instance from DT */
	hdma.Instance = __LL_DMA_GET_CHANNEL_INSTANCE(DMA1,
						      dev_data->dma.channel+1);
#else`

FRASTM, 2023-03-15:
> During debugging we found code which could be problematic. In file flash_stm32_qspi.c DMA1 is hardcoded as DMA instance in line 1125:
> 
> > #ifdef CONFIG_DMAMUX_STM32
> > /* HAL expects a valid DMA channel (not DAMMUX) _/
> > /_ TODO: Get DMA instance from DT */
> > hdma.Instance = __LL_DMA_GET_CHANNEL_INSTANCE(DMA1,
> > dev_data->dma.channel+1);
> > #else

Oh, I see that. Does a simple change
```
hdma.Instance = __LL_DMA_GET_CHANNEL_INSTANCE(dev_data->dma.reg,
						      dev_data->dma.channel);
```

makes the instance pointing to the right channel of the right DMA 2

```
&quadspi {
            dmas = <&dma2 0 20 0x480>; /* request 20 for QUADSPI1 */
            dma-names = "tx_rx";
};
```

pebeoo, 2023-03-15:
This change is working for me.

closed 2023-04-04 by MaureenHelm