zephyrproject-rtos/zephyr #55758: JSON parser returning when unexpected sub-object is encountered

BeckmaR, 2023-03-14:
**Describe the bug**
The JSON parser returns when an unexpected nested object is closed with `}`, instead of continuing to parse the rest of the JSON.

**To Reproduce**
I have created a PR with tests that show the behavior:
https://github.com/zephyrproject-rtos/zephyr/pull/55757

This works:
```C
ZTEST(lib_json_test, test_json_key_not_in_descr_partial)
{
	struct test_struct ts;
	char encoded[] = "{\"key_not_in_descr\":123456, \"some_int\": 123}";
	int ret;

	ret = json_obj_parse(encoded, sizeof(encoded) - 1, test_descr,
			     ARRAY_SIZE(test_descr), &ts);
	zassert_equal(ret, 0b10, "some_int should be decoded");
}
```

This does not:
```C
ZTEST(lib_json_test, test_json_key_obj_not_in_descr_partial)
{
	struct test_struct ts;
	char encoded[] = "{\"key_not_in_descr\":{}, \"some_int\": 123}";
	int ret;

	ret = json_obj_parse(encoded, sizeof(encoded) - 1, test_descr,
			     ARRAY_SIZE(test_descr), &ts);
	zassert_equal(ret, 0b10, "some_int should be decoded");
}
```

**Expected behavior**
Since the JSON parser does not barf on unrecognized keys that are strings or numbers, I would expect it to continue parsing also in this case. I personally expect the parser to ignore all keys that it doesn't know, but I would like to clarify if that is the library's design goal.

**Impact**
I can design around this in my application but any JSON-accepting application could be stopped from working by inserting a stray object at the start of the JSON data.

**Logs and console output**
See GH Action for my PR:
```
START - test_json_key_not_in_descr_partial
 PASS - test_json_key_not_in_descr_partial in 0.001 seconds
===================================================================
START - test_json_key_obj_not_in_descr_partial
    Assertion failed at WEST_TOPDIR/zephyr/tests/lib/json/src/main.c:579: lib_json_test_test_json_key_obj_not_in_descr_partial: (ret not equal to 0b10)
some_int should be decoded
 FAIL - test_json_key_obj_not_in_descr_partial in 0.002 seconds
```

**Environment (please complete the following information):**
See PR linked above.

BeckmaR, 2023-03-30:
@mrfuchs No, I wasn‘t aware of that issue.

mrfuchs, 2023-06-06:
> @mrfuchs Can you please look at this bug please and see if it still exists?

This issue is a duplicate of #47988 and #49588. It still exists. #47989 fixes it, but that PR makes use of recursion (as the JSON library itself) and (therefore?) was not merged. I'll try to update that PR with an iterative approach.

mrfuchs, 2023-07-27:
Fixed by #47989 which was merged in c0d7218.

closed 2023-07-27 by mrfuchs