zephyrproject-rtos/zephyr #55767: Networking config option causes zephyr to lock up

rcolatobe, 2023-03-14:
Using an STM32_H743VIT board zephyr hangs during start up.The project worked with zephyr 3.2 and this behavior started after upgrading to zephyr 3.3. Running git bisect the behavior occurs at and after commit b9da848c. The project has CONFIG_NETWORK enabled, which enables CONFIG_ENTROPY_GENERATOR, and CONFIG_RNG

When building the project, using the aforementioned the following warning output is:
```
[2/19] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (/home/rcolato/projects/nuclear_sondebase/sondebase/zephyr), build: zephyr-v3.2.0-3632-g5cbff34e35a6
[9/19] Building C object zephyr/drivers/entropy/CMakeFiles/drivers__entropy.dir/entropy_stm32.c.obj
/home/rcolato/projects/nuclear_sondebase/sondebase/zephyr/drivers/entropy/entropy_stm32.c: In function 'entropy_stm32_rng_init':
/home/rcolato/projects/nuclear_sondebase/sondebase/zephyr/drivers/entropy/entropy_stm32.c:609:2: warning: #warning RNG requires HSI48 clock to be enabled using device tree [-Wcpp]  609 | #warning RNG requires HSI48 clock to be enabled using device tree      |  ^~~~~~~
[11/19] Linking CXX executable zephyr/zephyr_pre0.elf
```

However the warning does not appear on the release version of zephyr 3.3 without code changes.
While writing this issue I found this issue https://github.com/zephyrproject-rtos/zephyr/issues/55013 which shows a fix, but the build should fail if the clock isn't configured correctly. Could there be a check added to prevent this hanging behavior on STM builds that omit the device tree entry for the hsi48 node?

erwango, 2023-03-15:
> However the warning does not appear on the release version of zephyr 3.3 without code changes.

Can you clarify this statement ?

> Could there be a check added to prevent this hanging behavior on STM builds that omit the device tree entry for the hsi48 node?

No we can't do this. HSI48 used as RNG source is a config choice. Other choices are available and valid.
We're just warning that your configuration has issues.

rcolatobe, 2023-03-27:
```
-- west build: making build dir pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: app
-- CMake version: 3.24.1
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.7") found components: Interpreter
-- Cache files will be written to: /.cache/zephyr
-- Zephyr version: 3.3.0 (zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: board, Revision: X
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (zephyr-sdk-0.15.2)
-- Found Dtc:  (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: 
-- Found devicetree overlay: 
-- Found devicetree overlay: 
-- Generated zephyr.dts: build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file:/build/zephyr/dts.cmake
/build/zephyr/zephyr.dts:594.7-623.5: Warning (simple_bus_reg): /soc/can: missing or empty reg/ranges property
Parsing Kconfig
Loaded configuration
Merged configuration 
Merged configuration
Merged configuration build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to /build/zephyr/.config'
Kconfig header saved to '/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring and generating APP
-- Configuring done
-- Generating done
-- Build files have been written to: /build
-- west build: building application
[1/393] Preparing syscall dependency handling

[2/393] Generating include/generated/version.h
-- Zephyr version: 3.3.0 (/zephyr), build: zephyr-v3.3.0
[383/393] Linking CXX executable zephyr/zephyr_pre0.elf

[387/393] Linking CXX executable zephyr/zephyr_pre1.elf

[393/393] Linking CXX executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      487148 B         2 MB     23.23%
             RAM:      382272 B       512 KB     72.91%
     BACKUP_SRAM:          0 GB         4 KB      0.00%
           SRAM1:          0 GB       128 KB      0.00%
           SRAM2:          0 GB       128 KB      0.00%
           SRAM3:         16 KB        32 KB     50.00%
           SRAM4:          0 GB        64 KB      0.00%
            DTCM:          0 GB       128 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
```
```
-- west build: making build dir /build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: APP
-- CMake version: 3.24.1
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.7") found components: Interpreter
-- Cache files will be written to: /.cache/zephyr
-- Zephyr version: 3.2.99 (zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: board, Revision: 0.0.1
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/zephyr-sdk-0.15.2)
-- Found Dtc:(found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: 
-- Found devicetree overlay: 
-- Found devicetree overlay:
-- Generated zephyr.dts:/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: build/zephyr/dts.cmake
/build/zephyr/zephyr.dts:590.7-619.5: Warning (simple_bus_reg): /soc/can: missing or empty reg/ranges property
Parsing Kconfig
Loaded configuration 
Merged configuration 
Merged configuration
Merged configuration /build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to 'build/zephyr/.config'
Kconfig header saved to '/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring and generating APP
-- Configuring done
-- Generating done
-- Build files have been written to: build
-- west build: building application
[1/389] Preparing syscall dependency handling

[2/389] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (zephyr), build: zephyr-v3.2.0-3633-gb9da848c0bc0
[222/389] Building C object zephyr/drivers/entropy/CMakeFiles/drivers__entropy.dir/entropy_stm32.c.obj
/zephyr/drivers/entropy/entropy_stm32.c: In function 'entropy_stm32_rng_init':
zephyr/drivers/entropy/entropy_stm32.c:597:2: warning: #warning HSI48 clock should be enabled or other domain clock selected [-Wcpp]
  597 | #warning HSI48 clock should be enabled or other domain clock selected
      |  ^~~~~~~
[379/389] Linking CXX executable zephyr/zephyr_pre0.elf

[383/389] Linking CXX executable zephyr/zephyr_pre1.elf

[389/389] Linking CXX executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      486284 B         2 MB     23.19%
             RAM:      382208 B       512 KB     72.90%
     BACKUP_SRAM:          0 GB         4 KB      0.00%
           SRAM1:          0 GB       128 KB      0.00%
           SRAM2:          0 GB       128 KB      0.00%
           SRAM3:         16 KB        32 KB     50.00%
           SRAM4:          0 GB        64 KB      0.00%
            DTCM:          0 GB       128 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
```
The first build output is based on zephyr 3.3 release and excludes the build warning.
The second build output is based on the commit referenced before and includes the build warning.
No changes to application code were made between builds.

erwango, 2023-03-28:
> The first build output is based on zephyr 3.3 release and excludes the build warning.
The second build output is based on the commit referenced before and includes the build warning.
No changes to application code were made between builds.

Note that, in the same PR, a [commit](a30ea45ef4721001dcb64f85e98ae65a0dfd9eb8) was introduced that removes the warning and provide a Kconfig option `ENTROPY_STM32_CLK_CHECK` which allows to derogate on the clock check.

erwango, 2023-05-24:
@rcolatobe Is this issue still valid ?

closed 2023-08-08 by github-actions[bot]