zephyrproject-rtos/zephyr #55769: samples/subsys/fs/fat_fs: Infinite loop if SD card is not inserted

koffes, 2023-03-14:
**Describe the bug**
If you run the above sample without an SD card inserted, the code will not return an appropriate error, but hang in an infinite loop.
Output: 
```
*** Booting Zephyr OS build zephyr-v3.2.0-5640-g33220ef086cc ***
[00:00:00.258,697] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.259,429] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.260,162] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.260,864] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.261,596] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.262,329] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.263,031] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.263,763] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.264,495] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.265,197] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.265,930] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.265,930] <inf> sd: Card does not support CMD8, assuming legacy card
```

This was tested using latest main (33220ef086cc13e9f0e2334fbe4a853876792414) and the target `nrf5340_audio_dk_nrf5340_cpuapp `.
This worked in November, so I will start a bisect now to find what changes have happened.

**Expected behavior**
A proper error return code from `disk_access_init(disk_pdrv)` if the SD card is not present.

**Impact**
Unsure yet, but may be severe if this hangs for all DKs with no SD card inserted.

danieldegrasse, 2023-03-15:
@koffes Appreciate you bisecting this- I don't have a Nordic board to reproduce on, but I cannot reproduce this using a FRDM-K64F, which uses the same SPI based SDHC driver that the `nrf5340_audio_dk_nrf5340_cpuapp` does.

I see the following output when running `samples/subsys/fs/fatfs` without an SD card inserted:

```
*** Booting Zephyr OS build zephyr-v3.3.0-1109-g33220ef086cc ***
[00:00:00.004,000] <err> sd: Card error on CMD0
[00:00:00.004,000] <err> main: Storage init ERROR!
[00:00:00.009,000] <err> sd: Card error on CMD0
[00:00:00.009,000] <err> fs: fs mount error (-5)
Error mounting disk.
```

Can you see where the code hangs on your board? I would expect CMD0 to return an error when no card is inserted.

koffes, 2023-03-17:
Thanks for the reply, We will have a look and get back to you.

alexsven, 2023-03-17:
I've done some digging/debugging and can see that we get:
`<inf> sd: Legacy card detected, no CMD8 support, resp: 0x0000`
11 times, since the number of retries is set to 10, then we get:
`<inf> sd: Card does not support CMD8, assuming legacy card`
after which it calls 'sd_idle(card)'. This returns 0 to sd_command_init() which then continues with 
```C
if (!sdmmc_card_init(card)) {
	return 0;
}
```

Is this right? I would assume that if the response in sd_send_interface_condition is 0x0000 then it means there is no card present? Or is the problem that 
```C
sdhc_request(card->sdhc, &cmd, NULL);
```
should have returned an error?

jfischer-no, 2023-03-17:
The SPI bus (spi4 node) is shared between sdhc0 and cs47l63 (IMO design issue), I suggest to look there at line states/configuration.

alexsven, 2023-03-17:
Forgot to mention that this is when there is no SD card inserted, of course. When the SD card is inserted, everything works as normal.

danieldegrasse, 2023-03-17:
> Is this right? I would assume that if the response in sd_send_interface_condition is 0x0000 then it means there is no card present? Or is the problem that
>
> ```
> sdhc_request(card->sdhc, &cmd, NULL);
> ```
>
> should have returned an error?

`sdhc_request` should indeed return an error on CMD0 when no card is inserted. 

> The SPI bus (spi4 node) is shared between sdhc0 and cs47l63 (IMO design issue), I suggest to look there at line states/configuration.

This is where I would look as well. I wonder if the other device on the bus is causing the data read during `sdhc_request` to match what would be expected from an SD card as a response to CMD0, resulting in `sdhc_request` succeeding when it should not

alexsven, 2023-03-20:
I've looked at the SPI lines and here you can see how it looked like before the changes in the upmerge:
[image]
and this is after the changes in the upmerge:
[image]
As you can see, there is nothing on the MISO, and the CS for CS47L63 stays high. I also see that there are more calls to the SPI. I've added the capture here as well:
[After upmerge with SD.zip](https://github.com/zephyrproject-rtos/zephyr/files/11015600/After.upmerge.with.SD.zip)

alexsven, 2023-03-20:
We have done some more debugging and see that sdhc_request expects the MISO to be high if no SD card present. Since we have a level shifter in our design and the pin is floating when there is no SD card, it seems the level shifter is pulling the MISO low when CS is low. If the SD card spec says MISO has to be high when no card is present that is something we have to fix for the next spin of our board. 

This is however not going to fix the problem for all of our current boards. Is there a place where the spec expects something else than 0 as a response that could be checked? [Here](https://github.com/zephyrproject-rtos/zephyr/blob/837091c05627950adfbd8bc47fe57560ec39607d/subsys/sd/sdmmc.c#L556) for instance?

danieldegrasse, 2023-05-17:
@koffes Are you ok if I close this issue? It seems like the problem has been resolved using the additional check in the SD stack.

koffes, 2023-05-17:
Yes, this should have been linked.

closed 2023-05-17 by koffes