zephyrproject-rtos/zephyr #55786: Attribute handle in GATT read callback is all 0 from Zephyr 3.2 stack

fbzhomi, 2023-03-14:
Hi,
I found an issue that attribute handle in customized read callback of GATT in application are all 0 using Zephyr 3.2 BT host stack. My environment is Linux OS and zeohyr build is host only build.
I have a customized characteristic uuid and read callback defined as below. 
static struct bt_uuid_16 service_uuid = BT_UUID_INIT_16(0x1234);

static struct bt_uuid_16 char_uuid = BT_UUID_INIT_16(0x5678);

static ssize_t read_char_cb(struct bt_conn *conn,
                                          const struct bt_gatt_attr *attr,
			                                    void *buf,
                                          uint16_t len,
                                          uint16_t offset) {
	const char *value = attr->user_data;
      LOG_INF("read_char_cb on handle: %d", attr->handle);   -> all 0
	return bt_gatt_attr_read(conn, attr, buf, len, offset, value,
				 sizeof(char_value));
}
When I define a service and corresponding characteristic, I associate the read callback into the definition.

/* Custom GATT DB Declaration */
BT_GATT_SERVICE_DEFINE(svc,
	BT_GATT_PRIMARY_SERVICE(&service_uuid),
	BT_GATT_CHARACTERISTIC(&char_uuid.uuid,
	                       BT_GATT_CHRC_READ,
                         BT_GATT_PERM_READ,
			                   read_char_cb,
                         NULL,
                         char_value),
);
Then when I test the build, I test dut against with nRF connect app in Iphone12 pro, and I found when I read the characteristic, read_char_cb is triggered but attribute handle is 0. I define multiple other characteristics like above format using other UUID and callbacks, I encounter the same issue that all attribute handle in customized read callback in application are all 0.   This seems like bug in zephyr host stack since attribute handle comes from the stack when defining the GATT characteristic.

fbzhomi, 2023-03-14:
Also the attribute handle gets from bt_gatt_find_by_uuid(svc.attrs, svc.attr_count, &char_uuid.uuid) is also a 0. I suspect these two could be similar issues.

henrikbrixandersen, 2023-03-14:
Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

alwa-nordic, 2023-03-17:
You can use `bt_gatt_attr_get_handle` to get the local server handle of an attribute.

I think it would be best if we removed `struct bt_gatt_attr::handle`. I suspect that field was always just a hack for passing the handle to `bt_gatt_discover_func_t` from an ATT-read-by-type. We would ideally have a new type for the ATT read result.

alwa-nordic, 2023-03-17:
I see that the handle field is used for dynamically registered services (`bt_gatt_service_register`). So we can't just remove it. But we can document it as "for-internal-use, except in `bt_gatt_discover_func_t` \*sigh\*".

jori-nordic, 2023-03-20:
> I see that the handle field is used for dynamically registered services (`bt_gatt_service_register`). So we can't just remove it. But we can document it as "for-internal-use, except in `bt_gatt_discover_func_t` *sigh*".

I don't like this. Either we fix it for both, or mark it for internal use for both.

alwa-nordic, 2023-03-20:
> I don't like this. Either we fix it for both, or mark it for internal use for both.

As the implementation is now, using `bt_gatt_attr_get_handle` in `bt_gatt_discover_func_t` will work. But I believe that's unintentional.

How about: We document `struct bt_gatt_attr::handle` as "For-internal-use. See `bt_gatt_attr_get_handle`.". And, we leave `bt_gatt_discover_func_t` as-is for now. The documentation on `bt_gatt_discover_func_t` has fairly clear documentation on `attr`, establishing an alternative meaning of `struct bt_gatt_attr::handle` in that context.

alwa-nordic, 2023-03-21:
> However how to retrieve attribute handle based on UUID value, bt_gatt_find_by_uuid doesn't work to me if application defines a static service like below?

You can compose the functions, like so:

```c
uint16_t handle = bt_gatt_attr_get_handle(bt_gatt_find_by_uuid(NULL, 0, uuid));
```

Is this not working?

closed 2023-04-13 by carlescufi