zephyrproject-rtos/zephyr #55794: Bluetooth: ascs: Invalid Target_Latency[i] and/or Target_PHY[i] should not reject the Codec Config procedure

MariuszSkamra, 2023-03-15:
**Describe the bug**
List of Config Codec operation parameters contains Target_Latency[i] and Target_PHY[i] parameters which are not reflected in the list of available Reason codes. I've created an [issue](https://bluetooth.atlassian.net/browse/ES-22930?atlOrigin=eyJpIjoiMTUxOGE2OTUxODNhNDViZTljMjYxNzlhNWU1YzAyZTciLCJwIjoiaiJ9) about missing Reason codes. 
The answer I got:
"The target_[thing] parameters are merely targets for the server and not mandatory to hit (we don’t even have a meaningful description of what a target latency actually IS, there are no numbers involved) thus we left those out,."

Meaning that the operation should not be rejected. The expected behavior would be rather to fallback to some preferred values.

**Expected behavior**
The Codec Config operation won't be rejected

**Impact**
The operation is rejected with invalid (unrelated) error codes

MariuszSkamra, 2023-03-17:
Neither `BT_ATT_ERR_VALUE_NOT_ALLOWED` nor `BT_ATT_ERR_OUT_OF_RANGE` are ASCS Response Codes. Just to remind you, the response comes in notification. The client does not have to perform ATT Write Request, can use ATT Write Command, so ATT Error PDU is not an option.

MariuszSkamra, 2023-03-20:
Could be an option, I agree. I do like the specification is not strict about what shall be done. I asked to reopen this issue.

Thalley, 2023-06-05:
Re-pinged the erratum to get more clarification

Thalley, 2023-10-16:
This issue is still awaiting an update from the linked erratum: https://bluetooth.atlassian.net/browse/ES-22930?atlOrigin=eyJpIjoiMTUxOGE2OTUxODNhNDViZTljMjYxNzlhNWU1YzAyZTciLCJwIjoiaiJ9

closed 2024-03-06 by Thalley