zephyrproject-rtos/zephyr #55807: modem_quectel_bg9x does not initialize properly

n3rd4i, 2023-03-15:
**Describe the bug**
Zephyr BG9x driver does not work with Quectel BG95 modem's very well.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? Custom board based on nRF52840 + BG95 modem
 - What have you tried to diagnose or workaround this issue? Yes, no solution
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? Not sure.

**To Reproduce**
Steps to reproduce the behavior:
1. build project with config:
```cfg
CONFIG_NETWORKING=y
CONFIG_NET_SOCKETS=y
CONFIG_MODEM=y
CONFIG_MODEM_QUECTEL_BG9X=y
CONFIG_MODEM_QUECTEL_BG9X_APN="hologram"
```
2. west build
3. west flash
4. check terminal (check full logs below): 
```log
modem_cmd_handler: command ATH ret:-5
```

**Expected behavior**
Modem should power up and initialize properly.

**Impact**
Unclear how the modem can be used considering this issue.

**Logs and console output**
```log
# SEGGER J-Link RTT Viewer V7.66a Terminal Log File
# Compiled: 15:16:36 on May 19 2022
# Logging started @ 15 Mar 2023 11:15:03
00> [00:00:00.250,183] <inf> modem_quectel_bg9x: Setting Modem Pins
00> [00:00:01.004,821] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
00>                                             4e 4f 52 4d 41 4c 20 50  4f 57 45 52 20 44 4f 57 |NORMAL P OWER DOW
00>                                             4e                                               |N                
00> [00:00:03.000,335] <inf> modem_quectel_bg9x: ... Done!
00> [00:00:03.000,366] <inf> modem_quectel_bg9x: Waiting for modem to respond
00> [00:00:53.000,427] <err> modem_quectel_bg9x: Timeout waiting for RDY
00> [00:00:53.000,640] <inf> main: Board: custom_nrf52840
00> [00:00:53.000,671] <inf> main: end program
```
* Reset button pressed here
```log
00> [00:00:00.250,183] <inf> modem_quectel_bg9x: Setting Modem Pins
00> [00:00:02.752,258] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
00>                                             52 44 59                                         |RDY              
00> [00:00:02.752,319] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [RDY] (len:3)
00> [00:00:03.000,305] <inf> modem_quectel_bg9x: ... Done!
00> [00:00:03.000,305] <inf> modem_quectel_bg9x: Waiting for modem to respond
00> [00:00:03.000,335] <dbg> modem_cmd_handler: modem_cmd_send_ext: SENT DATA
00>                                             41 54 45 30                                      |ATE0             
00> [00:00:03.000,701] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
00>                                             41 54 45 30                                      |ATE0             
00> [00:00:03.000,854] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
00>                                             4f 4b                                            |h cmd [OK] (len:2)
00> [00:00:03.051,025] <dbg> modem_cmd_handler: modem_cmd_send_ext: SENT DATA
00>                                             41 54 48                                         |ATH              
00> [00:00:03.051,422] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
00>                                             45 52 52 4f 52                                   |ERROR            
00> [00:00:03.051,452] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [ERROR] (len:5)
00> [00:00:03.101,623] <err> modem_cmd_handler: command ATH ret:-5
00> [00:00:03.101,867] <inf> main: Board: custom_nrf52840
00> [00:00:03.101,867] <inf> main: end program
00> [00:00:05.446,899] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
00>                                             41 50 50 20 52 44 59                             |APP RDY          
(Connection lost)
```

**Environment (please complete the following information):**
 - OS: `Windows`
 - Toolchain: `gnuarmemb`
 - Commit SHA or Version used: `Zephyr version: 3.2.0`

**Additional context**
BG95 modem internally has `mdm-power-gpios` connected to the same pin as `mdm-reset-gpios`. Because of this reason the custom board does not have control over `mdm-power-gpios`. Since this pin is `mandatory` from DTS file perspective, `mdm-power-gpios` is mapped to the same pin as `mdm-reset-gpios`.

nordicjm, 2023-03-15:
>  BG95 modem internally has mdm-power-gpios connected to the same pin as mdm-reset-gpios. Because of this reason the custom board does not have control over mdm-power-gpios. Since this pin is mandatory from DTS file perspective, mdm-power-gpios is mapped to the same pin as mdm-reset-gpios.

If the driver has been designed around having a power pin and a reset pin, and you are combining those 2 into a single pin, then you're misusing the driver.

n3rd4i, 2023-03-15:
this is the information that is provided by Quectel (from BG95 datasheet). Because of this, there is no need for controlling the `mdm-power-gpios` pin since its the same as `mdm-reset-gpios` ?
[image]

rerickson1, 2023-03-15:
One option may be to make `mdm-power-gpios` optional instead of required.

n3rd4i, 2023-03-15:
That is a good suggestion, however I've noticed that actually `mdm-reset-gpios` is not [actually] used, so I've made a patch to remove this dependency. However, this did not fix the issue, the same issue is present.

```patch
From 7db204ce8a37e8025009fb1b6e15cdf460000906 Mon Sep 17 00:00:00 2001
Date: Wed, 15 Mar 2023 17:12:38 +0100
Subject: [PATCH] hotfix: removed mdm_reset_gpios pin dependency

---
 drivers/modem/quectel-bg9x.c         | 4 ++++
 dts/bindings/modem/quectel,bg9x.yaml | 2 +-
 2 files changed, 5 insertions(+), 1 deletion(-)

diff --git a/drivers/modem/quectel-bg9x.c b/drivers/modem/quectel-bg9x.c
index 75d49c4caf..f06f751c3f 100644
--- a/drivers/modem/quectel-bg9x.c
+++ b/drivers/modem/quectel-bg9x.c
@@ -22,7 +22,9 @@ static K_KERNEL_STACK_DEFINE(modem_workq_stack, CONFIG_MODEM_QUECTEL_BG9X_RX_WOR
 NET_BUF_POOL_DEFINE(mdm_recv_pool, MDM_RECV_MAX_BUF, MDM_RECV_BUF_SIZE, 0, NULL);
 
 static const struct gpio_dt_spec power_gpio = GPIO_DT_SPEC_INST_GET(0, mdm_power_gpios);
+#if DT_INST_NODE_HAS_PROP(0, mdm_reset_gpios)
 static const struct gpio_dt_spec reset_gpio = GPIO_DT_SPEC_INST_GET(0, mdm_reset_gpios);
+#endif
 #if DT_INST_NODE_HAS_PROP(0, mdm_dtr_gpios)
 static const struct gpio_dt_spec dtr_gpio = GPIO_DT_SPEC_INST_GET(0, mdm_dtr_gpios);
 #endif
@@ -1195,11 +1197,13 @@ static int modem_init(const struct device *dev)
 		goto error;
 	}
 
+#if DT_INST_NODE_HAS_PROP(0, mdm_reset_gpios)
 	ret = gpio_pin_configure_dt(&reset_gpio, GPIO_OUTPUT_LOW);
 	if (ret < 0) {
 		LOG_ERR("Failed to configure %s pin", "reset");
 		goto error;
 	}
+#endif
 
 #if DT_INST_NODE_HAS_PROP(0, mdm_dtr_gpios)
 	ret = gpio_pin_configure_dt(&dtr_gpio, GPIO_OUTPUT_LOW);
diff --git a/dts/bindings/modem/quectel,bg9x.yaml b/dts/bindings/modem/quectel,bg9x.yaml
index 3cc1eb01c9..795650ec91 100644
--- a/dts/bindings/modem/quectel,bg9x.yaml
+++ b/dts/bindings/modem/quectel,bg9x.yaml
@@ -14,7 +14,7 @@ properties:
 
     mdm-reset-gpios:
       type: phandle-array
-      required: true
+      required: false
 
     mdm-dtr-gpios:
       type: phandle-array
-- 
2.38.1.windows.1

```

n3rd4i, 2023-03-20:
I think the problem is that maybe later BG95 documentation show this difference:
[image]

Hence, I think that by using "APP RDY" instead of "RDY" fixes the issue for BG95 (specifically).

n3rd4i, 2023-04-21:
> That is a good suggestion, however I've noticed that actually `mdm-reset-gpios` is not used, so I've made a patch to remove this dependency. However, this did not fix the issue (it behaves the same).
> 
> ```diff
> From 7db204ce8a37e8025009fb1b6e15cdf460000906 Mon Sep 17 00:00:00 2001
> Date: Wed, 15 Mar 2023 17:12:38 +0100
> Subject: [PATCH] hotfix: removed mdm_reset_gpios pin dependency
> 
> ---
>  drivers/modem/quectel-bg9x.c         | 4 ++++
>  dts/bindings/modem/quectel,bg9x.yaml | 2 +-
>  2 files changed, 5 insertions(+), 1 deletion(-)
> 
> diff --git a/drivers/modem/quectel-bg9x.c b/drivers/modem/quectel-bg9x.c
> index 75d49c4caf..f06f751c3f 100644
> --- a/drivers/modem/quectel-bg9x.c
> +++ b/drivers/modem/quectel-bg9x.c
> @@ -22,7 +22,9 @@ static K_KERNEL_STACK_DEFINE(modem_workq_stack, CONFIG_MODEM_QUECTEL_BG9X_RX_WOR
>  NET_BUF_POOL_DEFINE(mdm_recv_pool, MDM_RECV_MAX_BUF, MDM_RECV_BUF_SIZE, 0, NULL);
>  
>  static const struct gpio_dt_spec power_gpio = GPIO_DT_SPEC_INST_GET(0, mdm_power_gpios);
> +#if DT_INST_NODE_HAS_PROP(0, mdm_reset_gpios)
>  static const struct gpio_dt_spec reset_gpio = GPIO_DT_SPEC_INST_GET(0, mdm_reset_gpios);
> +#endif
>  #if DT_INST_NODE_HAS_PROP(0, mdm_dtr_gpios)
>  static const struct gpio_dt_spec dtr_gpio = GPIO_DT_SPEC_INST_GET(0, mdm_dtr_gpios);
>  #endif
> @@ -1195,11 +1197,13 @@ static int modem_init(const struct device *dev)
>  		goto error;
>  	}
>  
> +#if DT_INST_NODE_HAS_PROP(0, mdm_reset_gpios)
>  	ret = gpio_pin_configure_dt(&reset_gpio, GPIO_OUTPUT_LOW);
>  	if (ret < 0) {
>  		LOG_ERR("Failed to configure %s pin", "reset");
>  		goto error;
>  	}
> +#endif
>  
>  #if DT_INST_NODE_HAS_PROP(0, mdm_dtr_gpios)
>  	ret = gpio_pin_configure_dt(&dtr_gpio, GPIO_OUTPUT_LOW);
> diff --git a/dts/bindings/modem/quectel,bg9x.yaml b/dts/bindings/modem/quectel,bg9x.yaml
> index 3cc1eb01c9..795650ec91 100644
> --- a/dts/bindings/modem/quectel,bg9x.yaml
> +++ b/dts/bindings/modem/quectel,bg9x.yaml
> @@ -14,7 +14,7 @@ properties:
>  
>      mdm-reset-gpios:
>        type: phandle-array
> -      required: true
> +      required: false
>  
>      mdm-dtr-gpios:
>        type: phandle-array
> -- 
> 2.38.1.windows.1
> ```

for this particular issue (the DTS part) I've created a PR to address it: https://github.com/zephyrproject-rtos/zephyr/pull/57110

n3rd4i, 2023-04-21:
> I think the problem is that maybe later BG95 documentation show this difference: [image]
> 
> Hence, I think that by using "APP RDY" instead of "RDY" fixes the issue for BG95 (specifically).

For this particular issue, the expected unsolicited mdm response has to change. This means there is need for another BG95 define so that the correct string is picked up on the initialization of the modem? e.g.:
```prj.conf
CONFIG_MODEM_QUECTEL_BG9x_UNSOL_RDY="APP RDY" // BG95 --> where default case (BG96) should be "RDY"
```
?

rerickson1, 2023-04-21:
> > I think the problem is that maybe later BG95 documentation show this difference: [image]
> > Hence, I think that by using "APP RDY" instead of "RDY" fixes the issue for BG95 (specifically).
> 
> For this particular issue, the expected unsolicited mdm response has to change. This means there is need for another BG95 define so that the correct string is picked up on the initialization of the modem? e.g.:
> 
> ```
> CONFIG_MODEM_QUECTEL_BG9x_UNSOL_RDY="APP RDY" // BG95 --> where default case (BG96) should be "RDY"
> ```
> 
> ?

Give it a try!

n3rd4i, 2023-06-03:
This fix is included now so this issue can be closed.

closed 2023-06-03 by n3rd4i