zephyrproject-rtos/zephyr #55842: Zephyr Configuration App/Project Board overlay does not have support for board revisions

optical-o, 2023-03-16:
Project/App configuration should look for files according the same scheme as the .overlay inside the board directory.

https://github.com/zephyrproject-rtos/zephyr/blob/0eb7391a63ba04e77276ee3e38264b6a3e25b259/cmake/modules/configuration_files.cmake#L100

https://github.com/zephyrproject-rtos/zephyr/blob/0eb7391a63ba04e77276ee3e38264b6a3e25b259/cmake/modules/configuration_files.cmake#L68

Example nrf9160dk_nrf9160:
Inside and app/sample/project you cannot differentiate between nrf9160dk_nrf52840_0_14_0 and nrf9160dk_nrf52840. Whenever you specify board with revision or without nrf9160dk_nrf52840 it always uses same .overlay file.

The same thing goes for prj files.

nordicjm, 2023-03-16:
One would expect prj files to target a specific board, not a specific board revision, but the overlay point is valid, that should support board revisions

optical-o, 2023-03-16:
For example on the nRF9160dk there is flash on revision 1.4.0 so one might one to use 
zephyr settings with different storage backends for each revision.
FS for 1.4.0 and associated config options with it. 
In Memory for <1.4.0. 
Or am i mistaken ?
The settings would cause warning on board without the Flash.

nordicjm, 2023-03-16:
You can do that using overlays (i.e. assuming that support for board revision overlays is added) e.g. `board/nrf9160dk_nrf9160_0.14.0.overlay` and `board/nrf9160dk_nrf9160_0.14.0.conf`

tejlmand, 2023-03-16:
> Project/App configuration should look for files according the same scheme as the .overlay inside the board directory.

they do support the same scheme.
Config files:
> Otherwise, if boards/<BOARD>.conf exists in the application configuration directory, the result of merging it with prj.conf is used.
>
>Otherwise, if board revisions are used and boards/<BOARD>_<revision>.conf exists in the application configuration directory, the result of merging it with prj.conf and boards/<BOARD>.conf is used.

Overlay files:
> If the current board has [multiple revisions](https://docs.zephyrproject.org/latest/hardware/porting/board_porting.html#porting-board-revisions) and boards/<BOARD>_<revision>.overlay exists, it will be used. This file will be used in addition to boards/<BOARD>.overlay if both exist.

only difference is that overlay files don't merge with a app "global" overlay file because overlay files usually are more board specific that config files.

The `prj_<board>.conf` style is pre-dates the `boards/<board>.conf` style and is kept mostly for historic reason.
I advice to use the `prj.conf` with an additional `boards/<board>.conf` for board adjustments.
Which is also the reason that only `boards/<board>_<revision>.conf` format is supported.

Ref: https://docs.zephyrproject.org/latest/build/kconfig/setting.html#the-initial-configuration
Ref: https://docs.zephyrproject.org/latest/build/dts/howtos.html#set-devicetree-overlays

tejlmand, 2023-03-16:
Closing, requested feature already exists through the use  of `boards/<board>_<revision>.[overlay,conf]` principle.

PR #55851 (2023-03-17): cmake: Throw error if no prj.conf file is in the app config dir
This now throws an error if there is no prj.conf file located in a user-specified APPLICATION_CONFIG_DIR, which otherwise would have used an empty configuration and not included board-specific files.

Fixes #55842
closed 2023-03-17 by carlescufi