zephyrproject-rtos/zephyr #55901: Multitude of warnings around lc3 and bap_lc3 when building docs

aescolar, 2023-03-17:
**Describe the bug**
When building  Zephyr's docs there is plenty of doxygen warnings  around the lc3 and bap_lc3 headers

**To Reproduce**
Steps to reproduce the behavior:
1. cd ${ZEPHYR_BASE}/doc
2. make html DT_TURBO_MODE=1 

**Expected behavior**
No warnings

**Impact**
Annoyance + they shadow other warnings

**Logs and console output**
```
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:382: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_16_1_1(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:382: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_16_1_1(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:383: found </em> tag without matching <em>
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:385: The following parameter of BT_BAP_LC3_BROADCAST_PRESET_16_1_1(_loc, _stream_context) is not documented:
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:549: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_16_1_2(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:549: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_16_1_2(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:550: found </em> tag without matching <em>
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:552: The following parameter of BT_BAP_LC3_BROADCAST_PRESET_16_1_2(_loc, _stream_context) is not documented:
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:392: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_16_2_1(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:392: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_16_2_1(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:395: found </em> tag without matching <em>
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:397: The following parameter of BT_BAP_LC3_BROADCAST_PRESET_16_2_1(_loc, _stream_context) is not documented:
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:559: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_16_2_2(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:559: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_16_2_2(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:562: found </em> tag without matching <em>
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:564: The following parameter of BT_BAP_LC3_BROADCAST_PRESET_16_2_2(_loc, _stream_context) is not documented:
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:404: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_24_1_1(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:404: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_24_1_1(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:405: found </em> tag without matching <em>
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:407: The following parameter of BT_BAP_LC3_BROADCAST_PRESET_24_1_1(_loc, _stream_context) is not documented:
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:571: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_24_1_2(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:571: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_BROADCAST_PRESET_24_1_2(_loc, _stream_context)
WARNING: zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:572: found </em> tag without matching <em>
[....]
```

**Environment**
 - OS: Ubuntu 22.04
 - Commit SHA or Version used: 58f35ead65074891bfeb4592199c534df6f3b7b3 (main as of yesterday)

aescolar, 2023-03-17:
CC @Thalley @asbjornsabo

Thalley, 2023-03-17:
This has been seen before: https://discord.com/channels/720317445772017664/884782246010175528/963075575226961970

This is a Sphinx/doxygen bug; can you provide the version of those that you are using to build?

Thalley, 2023-03-17:
@aescolar is this only failing locally for you, or in CI? 

Have you tried doing something like `pip3 install --user --upgrade -r requirements-doc.txt`?
I just built locally, and I don't see any of these warnings

aescolar, 2023-03-17:
@Thalley versions added

Thalley, 2023-03-17:
Thanks. Not sure what to do about this issue: The documentation is correct and it is seemingly a Sphinx/Doxygen issue, which I cannot reproduce either with the build commands supplied by this issue. 

We are using the same version of Sphinx (5.3.0), but my Doxygen version is lightly newer (1.9.6), but I don't see any reference of this issue in the Doxygen release notes. The issue with underscore was fixed back in release 1.8.14 (https://github.com/doxygen/doxygen/issues/6128)

aescolar, 2023-03-17:
> Thanks. Not sure what to do about this issue: The documentation is correct and it is seemingly a Sphinx/Doxygen issue, which I cannot reproduce either with the build commands supplied by this issue.

Feel free to close the issue then. It is not a too bad thing.

aescolar, 2023-03-17:
> It doesn't seem like the Documentation Build job shows any logs from building the docs, so not sure how to verify that.

@stephanosio can you comment on this?

>If this is something that happens to other than you

I'm using default Ubuntu 22.04 LTS packages, and default Zephyr instructions. So I'm guessing it is a relatively normal situation.

stephanosio, 2023-03-17:
Documentation build job logs, like any other workflow jobs, can be viewed through the GitHub UI.

stephanosio, 2023-03-17:
FYI @gmarull @kartben

gmarull, 2023-03-17:
Can you try Doxygen 1.9.6?

aescolar, 2023-03-23:
> @aescolar Have you tried with a newer version of doxygen?

Not yet. But I presume it will not produce the warnings

aescolar, 2023-03-23:
> > > @aescolar Have you tried with a newer version of doxygen?
> > 
> > 
> > Not yet. But I presume it will not produce the warnings
> 
> Some 1.9.x Doxygen versions have bugs, not sure if 1.9.1 is one of them. I know that 1.9.5 is unusable for us due to a similar issue.

We can close the issue if you want.

aescolar, 2023-03-23:
I confirm that with doxygen 1.9.6 the warnings are not present