zephyrproject-rtos/zephyr #55959: modem: gsm_ppp: Unable to restart modem after calling gsm_ppp_stop()

liamclarkza, 2023-03-19:
**Describe the bug**
After calling gsm_ppp_start() and successfully connecting to the internet, I stop the connection by calling gsm_ppp_stop(). Subsequent calls to gsm_ppp_start() do not start the modem and, when logging is enabled via CONFIG_MODEM_LOG_LEVEL_DBG=y, I can see that this is because there is a timeout occurring in gsm_configure().

The modem I am using is a Simcom SIM800. Additionally, power cycling the modem does not help which suggests that this may be a bug with the GSM code in Zephyr. I have also tried waiting for network down events and adding delays before calling gsm_ppp_start() but the issue stays the same.

I see that a similar issue was mentioned here: #52885

**To Reproduce**
With a GSM modem configured with the KConfig options below:
- call `gsm_ppp_start()` after boot
- after connecting successfully by waiting for NET_EVENT_L4_CONNECTED event, call `gsm_ppp_stop()`
- after disconnecting, call `gsm_ppp_start()`

**Expected behavior**
Subsequent calls to gsm_ppp_start() should work after calling gsm_ppp_stop().

**Logs and console output**
```
###### At boot, gsm_ppp_start() is called ######

[00:00:00.000,000] <dbg> modem_gsm: gsm_init: Generic GSM modem (0x20002c40)
[00:00:00.000,000] <dbg> modem_gsm: gsm_init: iface->read 0x8041b57 iface->write 0x8041b2b
[00:00:00.000,000] <dbg> modem_gsm: gsm_rx: starting
[00:00:02.001,000] <dbg> modem_gsm: gsm_configure: Starting modem 0x20002c40 configuration
[00:00:04.001,000] <dbg> modem_gsm: gsm_configure: modem not ready -116
[00:00:06.002,000] <dbg> modem_gsm: gsm_configure: modem not ready -116
[00:00:06.011,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:06.012,000] <dbg> modem_gsm: gsm_cmd_ok: ok
...
[00:00:07.242,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:07.242,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:07.296,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [] (len:10)
[00:00:07.296,000] <inf> modem_gsm: Manufacturer: SIMCOM_Lt
[00:00:07.297,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:07.297,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:07.351,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [] (len:14)
[00:00:07.351,000] <inf> modem_gsm: Model: SIMCOM_SIM800C
[00:00:07.351,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:07.351,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:07.407,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [] (len:28)
[00:00:07.408,000] <inf> modem_gsm: Revision: Revision:1418B09SIM800C24_BT
[00:00:07.408,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:07.408,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:07.463,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [] (len:15)
[00:00:07.463,000] <inf> modem_gsm: IMEI: <XXXXXXXXXXXXXXXX>
[00:00:07.464,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:07.464,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:07.518,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [] (len:15)
[00:00:07.518,000] <inf> modem_gsm: IMSI: <XXXXXXXXXXXXXXXX>
[00:00:07.518,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:07.518,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:07.573,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [] (len:20)
[00:00:07.573,000] <inf> modem_gsm: ICCID: <XXXXXXXXXXXXXXXX>
[00:00:07.574,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:07.574,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:07.628,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [+CREG: ] (len:10)
[00:00:07.628,000] <dbg> modem_gsm: on_cmd_net_reg_sts: Searching for network...
[00:00:07.628,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:07.628,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:08.632,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [+CREG: ] (len:10)
[00:00:08.632,000] <dbg> modem_gsm: on_cmd_net_reg_sts: Searching for network...
[00:00:08.633,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:08.633,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:09.636,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [+CREG: ] (len:10)
[00:00:09.636,000] <dbg> modem_gsm: on_cmd_net_reg_sts: Searching for network...
[00:00:09.637,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:09.637,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:10.641,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [+CREG: ] (len:10)
[00:00:10.641,000] <dbg> modem_gsm: on_cmd_net_reg_sts: Searching for network...
[00:00:10.641,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:10.641,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:11.646,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [+CREG: ] (len:10)
[00:00:11.646,000] <dbg> modem_gsm: on_cmd_net_reg_sts: Searching for network...
[00:00:11.646,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:11.646,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:12.650,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [+CREG: ] (len:10)
[00:00:12.650,000] <dbg> modem_gsm: on_cmd_net_reg_sts: Network registered, home network.
[00:00:12.651,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:12.651,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:12.655,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [+CGATT:] (len:9)
[00:00:12.655,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:12.655,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:12.655,000] <dbg> modem_gsm: gsm_finalize_connection: modem attach returned 0, read RSSI
[00:00:12.659,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [+CSQ:] (len:10)
[00:00:12.659,000] <dbg> modem_gsm: on_cmd_atcmdinfo_rssi_csq: RSSI: -63
[00:00:12.659,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:12.659,000] <dbg> modem_gsm: gsm_cmd_ok: ok
[00:00:12.659,000] <dbg> modem_gsm: gsm_finalize_connection: modem RSSI: -63, enable PPP
[00:00:12.663,000] <dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [CONNECT] (len:7)
[00:00:12.663,000] <dbg> modem_gsm: gsm_cmd_ok: ok
EVENT: L2 [1] up 
[00:00:12.714,000] <inf> net_ppp: Initializing PPP to use serial@40004400
EVENT: L3 [1] IPv4 address add <XXX.XXX.XXX.XXX>
EVENT: L4 [1] connected 

###### Now gsm_ppp_stop() is called ######

EVENT: L3 [1] IPv4 address del <XXX.XXX.XXX.XXX>
EVENT: L4 [1] disconnected 
EVENT: L2 [1] down 
[00:00:32.112,000] <inf> modem_gsm: GSM network interface down

###### Now gsm_ppp_start() is called ######

[00:00:34.112,000] <dbg> modem_gsm: gsm_configure: Starting modem 0x20002c40 configuration
[00:00:36.112,000] <dbg> modem_gsm: gsm_configure: modem not ready -116
[00:00:38.113,000] <dbg> modem_gsm: gsm_configure: modem not ready -116
[00:00:40.113,000] <dbg> modem_gsm: gsm_configure: modem not ready -116
[00:00:42.113,000] <dbg> modem_gsm: gsm_configure: modem not ready -116
...
```

**Additional context**
KConfig options:
```
#General Config
CONFIG_NEWLIB_LIBC=y
CONFIG_SHELL=y

#Main Stack Size
CONFIG_MAIN_STACK_SIZE=4096
CONFIG_IMG_BLOCK_BUF_SIZE=2048

#Minimal Heap mem pool size for the Hawkbit working
CONFIG_HEAP_MEM_POOL_SIZE=4096
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
CONFIG_MODEM_GSM_WORKQ_STACK_SIZE=4096

# Network timeout tweaks for GSM
CONFIG_NET_TCP_ACK_TIMEOUT=1000
CONFIG_NET_TCP_INIT_RETRANSMISSION_TIMEOUT=1000
CONFIG_NET_TCP_RECV_QUEUE_TIMEOUT=5000
CONFIG_NET_SOCKETS_CONNECT_TIMEOUT=5000

# UART support
CONFIG_SERIAL=y
CONFIG_UART_ASYNC_API=y
CONFIG_NET_PPP_ASYNC_UART=y

# GSM modem configuration
CONFIG_MODEM=y
CONFIG_MODEM_GSM_PPP=y
CONFIG_MODEM_GSM_APN="internet"
CONFIG_MODEM_GSM_SIMCOM=y
CONFIG_MODEM_GSM_FACTORY_RESET_AT_BOOT=y
CONFIG_GSM_PPP_AUTOSTART=n
CONFIG_PPP_NET_IF_NO_AUTO_START=y
CONFIG_MODEM_SHELL=y
CONFIG_MODEM_LOG_LEVEL_DBG=y

# Networking
CONFIG_NETWORKING=y
CONFIG_NET_NATIVE=y
CONFIG_NET_DRIVERS=y
CONFIG_NET_PPP=y
CONFIG_NET_L2_PPP=y
CONFIG_NET_SHELL=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n

# Network management events
CONFIG_NET_CONNECTION_MANAGER=y
CONFIG_NET_MGMT_EVENT=y
CONFIG_NET_MGMT_EVENT_MONITOR=y
CONFIG_NET_MGMT_EVENT_MONITOR_AUTO_START=y
CONFIG_NET_MGMT_EVENT_QUEUE_SIZE=128

# Logging
CONFIG_LOG=y
CONFIG_NET_LOG=y
```

nashif, 2023-05-30:
@rerickson1  Can you please look at this bug please and see if it still exists?

rerickson1, 2023-05-30:
I have no hardware to be able to test this. @bjarki-trackunit, @ycsin, @jukkar any chance you can replicate this issue?

closed 2023-08-15 by github-actions[bot]