zephyrproject-rtos/zephyr #55962: Incorrect minimal libc sqrt

AndreyDodonov-EH, 2023-03-19:
**Describe the bug**
Minimal Libc SQRT has extremely low number of iterations (6), which leads to incorrect results.

https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/libc/minimal/source/math/sqrt.c#L18

`sqrtf` works fine because it targets precision, but float is sometimes not enough.

**To Reproduce**
Take `sqrt` of any sufficiently small value.
In my case, it was 0.0000016483
Obviously, `sqrt` from libc minimal should be used (`CONFIG_MINIMAL_LIBC`)

**Expected behavior**
Precision of `sqrt` results are higher than `sqrtf`.

**Impact**
Had to use my own function (locally and/or maintain own fork).

**Additional context**
I'm by no means an expert on sqrt implementations, but as a quick fix I'd suggest just taking the algorithm from sqrtf and replacing float with double.

lawrencek52, 2023-03-21:
I read through the implementations for both sqrt() and sqrtf() at https://github.com/zephyrproject-rtos/zephyr/tree/main/lib/libc/minimal/source/math The two implementations are swapped. Although the parameters passed to, and returned from, each function are correct the implementations of the algorithm are swapped. 

The sqrtf() function operates on floats (24-bit mantissa) which have smallest number of about 1.175494351E-38, but the loop repeats until the error is less than 2.225e-308. There is a small possibility that the loop may never exit if the error is toggling above and below the tolerance. The loop should exit when the error is less than a few counts in the mantissa, or when an exit counter reaches a maximum (about 6 times around the loop?). Also the mixing of floats and doubles in the function will force the compiler to promote to double, perform a division by 3.0 (should be 3.0f), then demote back to float, significantly slowing down the function on machines like M4F that only have float in hardware.

The sqrt() function operates on doubles (56-bit mantissa) which have a smallest number of about 2.225E-308. In this function the loop exit criteria is 6 times around the loop, I am pretty sure this is not enough times around the loop. The loop should have an exit counter to limit the max number of times around the loop, and an early exit criteria if the error of less than a few counts in the mantissa.

Both the sqrt() and the sqrtf() functions have issues and need to be looked at carefully. I'm not sure what happens if you pass a negative number to the functions.

A possible ugly cheat to determine when you are done looping  (which only works for IEEE style numbers) is to do a bitwise EXOR of the "root" and "last" variables and see if the upper 30 (or 62) bits are all zeros, (ie the sign, exponent, and most of the mantissa are the same) then you know you have converged. Although I haven't tested this to make sure it works in every case.

lawrencek52, 2023-04-07:
I had a chance to look carefully at the algorithm. It turns out that 6 iterations is plenty (5 is usually enough for doubles, and 4 for floats), this is provided that the initial guess if somewhat reasonable. The current algorithm for the first guess is x/3, this is a really good guess for numbers like 10, but a horrible guess for numbers like 10e10 and 10e-10. In the cases of 10e+/-10 it takes about 24 iterations to converge, and gets worse as the numbers get larger or smaller. The solution is to create a better first guess, this can be done by dividing the exponent part of x by 2. In the next few days I will submit a patch (as soon as I teach myself the correct way to submit a patch).

PR #58479 (2023-07-14): libc: minimal: math: sqrt: fix Incorrect minimal libc sqrt
I crashed my computer and I am having difficulty with git getting back to where I was, but I think this code is correct.

---

Continuation of https://github.com/zephyrproject-rtos/zephyr/pull/56881

Fixes #55962
commit 7dae27a9 (2023-07-14): libc: minimal: math sqrt: sqrtf: fix numeric accuracy of sqrt and sqrtf.

Changed initial guess from a simple x/3 to dividing the exponent by 2.
This makes large or small numbers like 10e10 and 01e-10 converge in a few
loops.

Added a loop counter to ensure that the algorithm breaks out of the loop in
the case that the algorithm doesn't converge (toggling between two
numbers).

Added test cases for sqrt and sqrtf in libc. Tested with a range of numbers
between 10e10 and 10e-10. Verify good accuracy in test case.

Closes: #55962
closed 2023-07-14 by cfriedt