zephyrproject-rtos/zephyr #55965: firmware update fails deterministically every other time

tatuylonen, 2023-03-19:
**Describe the bug**
I'm trying to upgrade device (stm32f407zgt6) firmware over ethernet using HTTP.  It works every other time.  Every other time mcuboot outputs "E: Image in the secondary slot is not valid!".  Upgrading to the exact same image works the next time.  If I repeat the exact same upgrade (same image), it succeeds the second time.  This appears very deterministic.

My upgrade process is roughly as follows:
  - I have partitions defined for flash0 (boot, storage, slot0, slot1, scratch) and chosen zephyr,flash=&flash0 and zephyr,code-partition=&slot0_partition
  - I use MCUBOOT as bootloader; zephyr is compiled with CONFIG_BOOTLOADER_MCUBOOT=y, CONFIG_IMG_MANAGER=y, CONFIG_MCUBOOT_IMG_MANAGER=y, CONFIG_FLASH=y, CONFIG_FLASH_MAP=y, CONFIG_MPU_ALLOW_FLASH_WRITE=y.
  - Upgrade is done by a task running on my own work queue, but flash & boot manager calls are done in http callback (packet reception context???)
  - Upgrade first resolves DNS (getaddrinfo), then iterates over IP addresses calling http_client_req (HTTP GET of the image) for each IP address until one succeeds.
  - I call flash_img_init in the first http callback with HTTP_DATA_MORE and then call flash_img_buffered_write with flush=0
  - subsequent calls to http callback with HTTP_DATA_MORE call flash_img_buffered_write with flush=0
  - in http callback with HTTP_DATA_FINAL, I call flash_img_buffered_write with flush=1, then call boot_request_upgrade(BOOT_UPGRADE_TEST), then reboot with sys_reboot(SYS_REBOOT_COLD)
  - now mcuboot starts, every other time successfully swaps the new image in test mode (every other time fails with the "Image in the secondary slot is not valid!" error)
  - When my application has successfully performed a HTTP request, I use "if (!boot_is_img_confirmed()) { ... = boot_write_img_confirmed(); ...}" to confirm the image.  (If the new image loads but cannot connect to the server, watchdog will eventually reset the board and mcuboot will switch back to the previous image)

This process seems to work just fine every other time, but fails every other time.  I've also tried expressly specifying slot1 using flash_img_init_id, but it did not change the behavior.  I've tried adding delays and reboots between various operations, with no effect on this behavior.  Since the exact same image will succeed on the next upgrade, it is not a signature problem.  Since it is so deterministic, I don't think it is a bug in HTTP transfer or flash writing. It always seems to write to the secondary slot as it should, but it seems as if mcuboot or the boot manager code in zephyr leaves something in a wrong state in the flash.

Is this behavior a bug or am I doing something wrong?  I cannot find answers in documentation and I haven't found a good sample for this.

**To Reproduce**
See above description.

**Expected behavior**
Firmware upgrade should succeed every time, not just every other time.  The image in the secondary slot should be detected as valid as it (presumably) is valid.

**Impact**
Several days were wasted trying to get firmware upgrade working reliably.  Workaround was to repeat the exact same upgrade process multiple times, and it always seems to succeed on the second attempt. However attempts need to be limited (max 4 in my implementation) or otherwise a bad image would quickly wear out flash on MCU and prevent normal operation.  These multiple attempts add significant extra complexity.

tatuylonen, 2023-03-20:
I didn't explicitly make erase calls.  I note that flash_img_buffered_write seems to erase automatically if CONFIG_IMG_ERASE_PROGRESSIVELY is set, and the documentation of that configuration option kind of lets me think that it just changes how it is erased, but that it would be erased anyway.  There is also CONFIG_STREAM_FLASH_ERASE, which also seems to cause it to erase automatically.  Nothing in the flash_img documentation makes any reference to these or erasing in general.

Adding CONFIG_STREAM_FLASH_ERASE=y seems to solve my problem and firmware upgrade now succeeds every time.  Thank you for the hint.  I'm guessing MCUBOOT erased the flash area when it found invalid data there, which allowed the earlier implementation to work every other time.

HOWEVER, nowhere does the Flash Image API documentation page mention erasing or either of these configuration options, which subtly but significantly change the operation of the API.  It would also be nice if there was an overview of how to perform the whole firmware upgrade process.  Neither Zephyr's Device Firmware Upgrade page nor the MCUBOOT page linked from there is very informative on the sequence of calls needed or how configuration options subtly change the behavior.

nordicjm, 2023-03-20:
> HOWEVER, nowhere does the Flash Image API documentation page mention erasing or either of these configuration options, which subtly but significantly change the operation of the API. It would also [...]

@de-nordic

de-nordic, 2023-03-21:
@tatuylonen Can you enable extra logging in MCUmgr and post logs from "image upload" and consecutive "image list" oprations?

Laczen, 2023-03-22:
> @tatuylonen Can you enable extra logging in MCUmgr and post logs from "image upload" and consecutive "image list" oprations?

@de-nordic, I think the solution was found, but documentation was not clear.

de-nordic, 2023-03-22:
> > @tatuylonen Can you enable extra logging in MCUmgr and post logs from "image upload" and consecutive "image list" oprations?
> 
> @de-nordic, I think the solution was found, but documentation was not clear.

Yes I have missed the:

> Adding CONFIG_STREAM_FLASH_ERASE=y seems to solve my problem and firmware upgrade now succeeds every time.

jgl-meta, 2023-03-22:
Closing this since the filer has solved the issue by adding a Kconfig option.

closed 2023-03-22 by jgl-meta