zephyrproject-rtos/zephyr #55970: Set default link address size to 8 bytes

Hiihtaja77, 2023-03-20:
Hi !

https://github.com/zephyrproject-rtos/zephyr/blob/580d789cd0e923124cf81425a93e567631ba32ce/include/zephyr/net/net_linkaddr.h#L37

I have own out-of-tree device drivers and can't define your own ones:

/** Maximum length of the link address */
#ifdef CONFIG_NET_L2_IEEE802154
#define NET_LINK_ADDR_MAX_LENGTH 8
#else
#ifdef CONFIG_NET_L2_PPP
#define NET_LINK_ADDR_MAX_LENGTH 8
#else
#define NET_LINK_ADDR_MAX_LENGTH 6
#endif
#endif

But I need NET_LINK_ADDR_MAX_LENGTH 8 bytes

Can default LINK address size to be set to 8 bytes or have independant configuration for avoid binding with limited set of L2 drivers.

jukkar, 2023-03-20:
I would not change the default one, but you could propose a patch that allows application to define one. For example something like
```
diff --git a/include/zephyr/net/net_linkaddr.h b/include/zephyr/net/net_linkaddr.h
index 9e21e7dc9f..857df8f148 100644
--- a/include/zephyr/net/net_linkaddr.h
+++ b/include/zephyr/net/net_linkaddr.h
@@ -34,9 +34,11 @@ extern "C" {
 #ifdef CONFIG_NET_L2_PPP
 #define NET_LINK_ADDR_MAX_LENGTH 8
 #else
+#ifndef NET_LINK_ADDR_MAX_LENGTH
 #define NET_LINK_ADDR_MAX_LENGTH 6
 #endif
 #endif
+#endif
 
 /**
  * Type of the link address. This indicates the network technology that this
```

Hiihtaja77, 2023-03-20:
Yes I have own patch as well. But I'm implementing an out-of-tree driver/l2 radio driver CONFIG_NET_L2_XXX and looks like
need to modify each zephyr release because the link address reserved max space is hard coded to the lowest value.
Have it sense to improve configurability at some moment?

jukkar, 2023-03-20:
If the patch I wrote few comments above is not enough, just propose a PR that adds the Kconfig option you want.

Hiihtaja77, 2023-03-21:
Overall idea is able to have support for several network interfaces with different length of link addresses.
But length is used in hardcoded way https://elixir.bootlin.com/zephyr/v3.3.0/source/include/zephyr/net/net_linkaddr.h#L96
And some common solution need if several interfaces ( at list 3) with different lengths need to be used.