zephyrproject-rtos/zephyr #55991: RFC: topic-v2.7-riscv branch

cfriedt, 2023-03-20:
## Introduction

Create a topic branch for RISC-V support on top of `v2.7-branch`.

### Problem description

Although the TSC voted some time ago (and it was very close) to not incorporate unicore and SMP bugfixes back into the `v2.7-branch` (LTSv2), it is nevertheless important to support Zephyr Community members who depend on LTS releases. 

### Proposed change

Create a `topic-v2.7-riscv` branch and enable CI on that branch.

## Detailed RFC

### Proposed change (Detailed)

Create a `topic-v2.7-riscv` branch and enable CI on that branch.
- [x] fork upstream `v2.7-branch` to `topic-v2.7-riscv`
- [x] rebase `topic-v2.7-riscv` onto `v2.7_riscv_smp` from https://github.com/npitre/zephyr/

Note: the `v2.7_riscv_smp` fixes originally appeared in https://github.com/zephyrproject-rtos/zephyr/pull/51727

### Dependencies

Only `v2.7-branch`

### Concerns and Unresolved Questions

Although the TSC voted some time ago (and it was very close) to not incorporate unicore and SMP bugfixes back into the `v2.7-branch` (LTSv2), it is nevertheless important to support Zephyr community members who depend on LTS releases. 

## Alternatives

Maintaining (many) external forks of `v2.7-branch` instead of a centralized branch.

henrikbrixandersen, 2023-03-27:
The TSC voted against back porting these commits to the `v2.7-branch` in upstream Zephyr. With the newly pushed `topic-v2.7-riscv` branch we now effectively have an upstream-hosted fork of the official Zephyr v2.7 release branch containing commits which the TSC explicitly voted against back porting.

A lot of testing goes into our LTS releases in general and this was certainly also the case for v2.7. With the amount of commits/changes added to the RISC-V fork users can not rely on the QA being performed on the `v2.7-branch` prior to release holding up for the fork, but this is not evident in any way.

Personally, I think upstream `topic`/development branches should only be allowed for short-term feature development as a last resort (e.g. when development work involves contributors from multiple downstream users/companies). We should not use `topic` branches for hosting unofficial forks like the `topic-v2.7-riscv` branch, especially when the TSC already voted no to making said commits part of the official `v2.7-branch`.

cfriedt, 2023-03-28:
> The TSC voted against back porting these commits to the `v2.7-branch` in upstream Zephyr. With the newly pushed `topic-v2.7-riscv` branch we now effectively have an upstream-hosted fork of the [...]

Correct. The votes were 11-10 against, IIRC (discounting abstentions). However, it is entirely feasible and permitted to re-table a vote with more evidence as to why it is being re-tabled.

It would be premature, undemocratic, and punitive to have the TSC vote on
a) deleting this topic branch
b) voting on the feasibility of the backport without seeing all of the available evidence.

> A lot of testing goes into our LTS releases in general and this was certainly also the case for v2.7. With the amount of commits/changes added to the RISC-V fork users can not rely on the QA being [...]

The only changes that have been added to `topic-v2.7-riscv` are those that exist already in `main`. I.e. they are all backports. You can easily verify that.

The entire point of the `topic-v2.7-riscv` branch and this RFC are to collect evidence to satisfy QA requirements.

> Personally, I think upstream `topic`/development branches should only be allowed for short-term feature development as a last resort (e.g. when development work involves contributors from multiple downstream users/companies).

And that is exactly what this topic branch is being used for.

> We should not use `topic` branches for hosting unofficial forks like the `topic-v2.7-riscv` branch, especially when the TSC already voted no to making said commits part of the official `v2.7-branch`.

I disagree on that take. Several members of the Zephyr board have also reassured me that it is entirely permitted to re-table a vote to the TSC. Moreover, the work that goes in to something like that, involving several companies, *should* be done in the open, and not behind closed doors. Additionally, it is incredibly important to have a central location for doing so, where the entire Zephyr Community *and* the TSC is able to see all of the evidence that *does* justify the backport.

Preventing TSC members from seeing that additional evidence would be odd to say the least, and I would question the motivation for doing so.

Generally speaking, more information is usually useful than less information, when making a technical decision.

nashif, 2023-04-13:
approved by the TSC. closing

closed 2023-04-13 by nashif