zephyrproject-rtos/zephyr #56011: SMP not working on ESP32

Rhvarrier, 2023-03-20:
**Describe the bug**
A simple Symmetric Multiprocessing example is not working on ESP32 even though SMP seems supported

**To Reproduce**
Simple snippet to reproduce the issue:
`main.c`
```
#include <zephyr/kernel.h>
#include <zephyr/device.h>

#define THREADS_NUM	2

#define STACK_SIZE	1024
static K_THREAD_STACK_ARRAY_DEFINE(tstack, THREADS_NUM, STACK_SIZE);
static struct k_thread tthread[THREADS_NUM];

static int curr_cpu(void)
{
	unsigned int k = arch_irq_lock();
	int ret = arch_curr_cpu()->id;
	arch_irq_unlock(k);
	return ret;
}

void test_thread(void *arg1, void *arg2, void *arg3)
{
    const uint32_t counter = *(uint32_t*)arg1;
    printk("Staring thread number %d\n", counter);
    while(1)
    {
        uint32_t proc_id = curr_cpu();
        printk("Thread number %d running on cpu %d\n", counter, proc_id);
        k_sleep(K_SECONDS(1));
    }
}

int main() {
    uint32_t proc_id = curr_cpu();
    printk("Main running on cpu %d\n", proc_id);
    for (uint32_t i = 0; i < THREADS_NUM; i++) {
        
	             k_thread_create(&tthread[i], tstack[i], STACK_SIZE,
			       (k_thread_entry_t)test_thread,
			       (void *)&i, NULL, NULL,
			       K_PRIO_COOP(11) , 0, K_NO_WAIT);
	}
    
    while(1)
    {
        k_yield();
    }
    return 0;
}
```
prj.conf
```
CONFIG_SMP=y
CONFIG_MP_MAX_NUM_CPUS=2
```

**Expected behavior**
One of the possible output on UART console:
```
Main running on cpu 0
Staring thread number 0
Thread number 0 running on cpu 1
Staring thread number 1
Thread number 1 running on cpu 0
Thread number 0 running on cpu 1
Thread number 1 running on cpu 0
```

**Impact**
SMP not working

**Logs and console output**
No output in UART. However if I disable SMP, I get the following output as expected:
```
Main running on cpu 0
Staring thread number 0
Thread number 0 running on cpu 0
Staring thread number 1
Thread number 1 running on cpu 0
Thread number 0 running on cpu 0
Thread number 1 running on cpu 0
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr
 - Commit SHA or Version used: latest main branch d4ed0d7a2d

Rhvarrier, 2023-03-21:
@jgl-meta I see that, you have tagged this bug as low priority. However it seems to a blocking issue. I have also tried running the unit test in `zephyr/tests/kernel/smp/` and it does not run at all on ESP32, therefore it may be a regression

jgl-meta, 2023-03-22:
If it is only effecting one platform it is typically classified as a low priority bug.

cfriedt, 2023-03-22:
@Rhvarrier - you have not really provided complete reproducibility information. That would, for example, include something like this, where a specific in-tree board is referenced and a command is provided.
```
west build -b esp32c3_devkitm -t flash tests/kernel/smp/
west espressif monitor
```

However, have you considered the possibility that there is only 1 applications core on your esp32 part? Some of them have a separately programmable baseband core (i.e. are AMP capable), but I was not aware that any of them are SMP capable. In that case, the output provided looks to be expected. 

I would keep this as priority low until additional information is provided.

ivrabie, 2023-03-22:
I have the same issue for the esp32 DevKit, I used the the example provided by @Rhvarrier .
The build command was 
```
west build -b esp32 
west flash 
west espressif monitor
``
[esp32-smp.zip](https://github.com/zephyrproject-rtos/zephyr/files/11044303/esp32-smp.zip)

If SMP is activated based on logs it doesn't boot application, the ESP IDF bootloader start but not the applications.
Here are the logs when SMP is activated:
```
ts Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6596
load:0x40078000,len:15216
ho 0 tail 12 room 4
load:0x40080400,len:3992
entry 0x40080634
I (29) boot: ESP-IDF d2564ca5c 2nd stage bootloader
I (29) boot: compile time 21:50:25
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28) 
I (111) esp_image: segment 1: paddr=00010044 vaddr=3ffb0000 size=0090ch (  2316) load
I (120) esp_image: segment 2: paddr=00010958 vaddr=3ffb090c size=0018ch (   396) load
I (128) esp_image: segment 3: paddr=00010aec vaddr=40080000 size=069ech ( 27116) load
I (148) esp_image: segment 4: paddr=000174e0 vaddr=00000000 size=08b58h ( 35672) 
I (161) esp_image: segment 5: paddr=00020040 vaddr=3f400040 size=02d40h ( 11584) map
I (165) esp_image: segment 6: paddr=00022d88 vaddr=00000000 size=0d290h ( 53904) 
I (186) esp_image: segment 7: paddr=00030020 vaddr=400d0020 size=05130h ( 20784) map
I (197) boot: Loaded app from partition at offset 0x10000
I (197) boot: Disabling RNG early entropy source...
```

cfriedt, 2023-03-22:
Just by inspection, `esp32c3_devkitm` is the only in-tree board named similarly to what was mentioned above. I'm going to proceed based on the assumption that is the actual board in question.

The following command fails, but the testsuite passes in CI, which means that this particular board / test combination is filtered-out.
```
west build -p auto -b esp32c3_devkitm -t flash tests/kernel/smp && west espressif monitor
...
/Users/cfriedt/zephyrproject/zephyr/arch/riscv/core/smp.c: In function 'arch_sched_ipi':
/Users/cfriedt/zephyrproject/zephyr/arch/riscv/core/smp.c:66:44: error: 'RISCV_MSIP_BASE' undeclared (first use in this function)
   66 | #define MSIP(hartid) ((volatile uint32_t *)RISCV_MSIP_BASE)[hartid]
      |                                            ^~~~~~~~~~~~~~~
/Users/cfriedt/zephyrproject/zephyr/arch/riscv/core/smp.c:81:25: note: in expansion of macro 'MSIP'
   81 |                         MSIP(_kernel.cpus[i].arch.hartid) = 1;
      |                         ^~~~
/Users/cfriedt/zephyrproject/zephyr/arch/riscv/core/smp.c:66:44: note: each undeclared identifier is reported only once for each function it appears in
   66 | #define MSIP(hartid) ((volatile uint32_t *)RISCV_MSIP_BASE)[hartid]
      |                                            ^~~~~~~~~~~~~~~
/Users/cfriedt/zephyrproject/zephyr/arch/riscv/core/smp.c:81:25: note: in expansion of macro 'MSIP'
   81 |                         MSIP(_kernel.cpus[i].arch.hartid) = 1;
      |                         ^~~~
/Users/cfriedt/zephyrproject/zephyr/arch/riscv/core/smp.c: In function 'ipi_handler':
/Users/cfriedt/zephyrproject/zephyr/arch/riscv/core/smp.c:66:44: error: 'RISCV_MSIP_BASE' undeclared (first use in this function)
   66 | #define MSIP(hartid) ((volatile uint32_t *)RISCV_MSIP_BASE)[hartid]
      |                                            ^~~~~~~~~~~~~~~
/Users/cfriedt/zephyrproject/zephyr/arch/riscv/core/smp.c:100:9: note: in expansion of macro 'MSIP'
  100 |         MSIP(csr_read(mhartid)) = 0;
      |         ^~~~
[114/189] Performing configure step for 'EspIdfBo
```

 Likely the reason that it is filtered-out is because this particular SoC only has 1 application processor. I.e. there is only 1 CPU, which I mentioned earlier.
```
filter: (CONFIG_MP_MAX_NUM_CPUS > 1)
```

This is not a bug. The provided output in the console output in the description is to be expected (there is only cpu0 to execute on).
```
Main running on cpu 0
Staring thread number 0
Thread number 0 running on cpu 0
Staring thread number 1
Thread number 1 running on cpu 0
Thread number 0 running on cpu 0
Thread number 1 running on cpu 0
```

rftafas, 2023-03-24:
For reference: there are some variants of ESP32 single core, which are not supported and are tagged NRND.

Rhvarrier, 2023-03-24:
That's true. For this issue, I have tested in on 2 ESP32 boards both of them have 2 cores as I'm able to run SMP using the official Freertos API

cfriedt, 2023-03-24:
> @cfriedt, indeed the SMP isn't working as reported. ESP32-C3 is single core, so no SMP. However, ESP32 should be working and it is not. So I am reopening this.

Ack - sorry @Rhvarrier, @sylvioalves - I completely forgot about `boards/xtensa/exp32*`. My mistake. At one point, I did have one of those, but I think I gave it away.

Rhvarrier, 2023-03-29:
Hello @sylvioalves
Has there been any advances for this issue? Please let me know if you need more information to reproduce this issue

xyzzy42, 2023-04-21:
I see the same thing on esp32.  A simple app with nothing but a shell won't start once SMP is enabled.  Did this SMP ever work?  If it did, then at least the problem can probably be bisected.

sylvioalves, 2023-04-21:
We have started checking this. Already fixed the sample code to work but there SMP is not working as it should yet.

Rhvarrier, 2023-06-21:
Any updates concerning this issue?

rftafas, 2023-06-21:
@Rhvarrier - We figured it is needed some regression checks to know when things started to go wrong. Any help would be appreciated, as this falls short to other tasks on our side.

MrMarteng, 2023-07-05:
Ran into the same issue. An update yet?

rftafas, 2023-07-06:
Same as above. The issue seems to be around for way longer than expected. Unfortunately, it is not our priority (not even on the horizon). FWIW, we are either implementing support for single core devices or focused on AMP. My advise would be trying to find when it worked and what happened from there. Some forensics:

This is where it started: https://github.com/zephyrproject-rtos/zephyr/releases/tag/zephyr-v1.11.0
This is when we started to spot problems: https://github.com/zephyrproject-rtos/zephyr/pull/39606

I can only suggest trying to figure what happened in between.

rftafas, 2023-08-15:
Hi, an update here: I was checking the logs, we haven't paid attention that the device in use is ESP32 Revision 1. We are not going to support it, only Revision 3 and later.

We will be adding more checks for that situation and check if we missed it somewhere.

I also suggest doing the test with any ESP32 Revision 3 device.

xyzzy42, 2023-08-16:
I tested on an revision 3 device, doesn't work.
```
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```

rftafas, 2023-08-16:
Thanks for the feedback. We could narrow it down further.

Judging by when it was implemented (see above), I'm almost sure that Zephyr's SMP was made on top of Revision 1. Consider that ESP-IDF had a lot of workarounds (software, cpu support, tools, libraries, etc) to make SMP working on Rev1 and none were ported to Zephyr and also consider that it didn't crash during initial tests (otherwise, it wouldn't be merged). Consequently, I suppose that SMP was enabled but non-functional and that was always the case, it seems. Disclaimer: one should do some detailed regression analysis to confirm that, as this is untested hypothesis.

I hope this 'forensics' help those willing to engage earlier than we do on this.

Finally, some important points:

1) Revision 1 should not have booted at all. We will increase our checks and fix it. PTAL: https://github.com/zephyrproject-rtos/zephyr/issues/51622
2) If SMP is to be made to work on a reasonable effort, one must use Revision 3.
3) SMP is not a priority on our side, even on Revision 3.

Rhvarrier, 2023-08-16:
Thanks for the update. However I don't understand why SMP is not a priority.

anuraj-rp, 2023-10-01:
Hi,

Ran into the same issue on ESP32-S3-DevKitM. 

Any ideas on how to hack it to make it work?

kartben, 2023-12-01:
Removing the stale label as, ironically, I had the same issue earlier today and stumbled upon this very issue.
If there is really no situation where SMP works, even with Rev. 3 devices, then shouldn't the Kconfig be set to n for the time being?
Currently, the `pi` sample compiles just fine but then hangs at runtime which is confusing.

xyzzy42, 2023-12-01:
Maybe the documentation should be updated to say that SMP isn't supported and is not planned?

rftafas, 2023-12-01:
I'm ok with taking this option out. On the other hand, SMP was a community driven effort prior to Espressif working on Zephyr. This also means that we are not comfortable on removing other people's work and we can't talk for everyone else. There may be more people willing to engage in supporting SMP.

FWIW, another unverified hypothesis is that initial implementation ran from RAM and thus, could be made to work on REV1. IMHO, this is useless beyond a POC.

rftafas, 2023-12-01:
> @rftafas - is this specific to RISCV? Or is this the xtensa esp platforms?

It's Xtensa only. ESP32-P4 will be the first Risc-V multicore that is capable of SMP.

legoabram, 2024-11-05:
@rhvarrier @kartben @anuraj-rp @xyzzy42 @sylvioalves @ivrabie I've gotten SMP to at least mostly functional on `main`. Both cores are executing code simultaneously, and not crashing. However, the scheduler and uptime counter behave unpredictably and erratically due to the handling of the sys clocks on the xtensa cores. The primary issue there, I think, is just due to the CCOUNT registers between the cores being completely different counters which is confusing sys_clock_announce.

[image]

https://github.com/legoabram/zephyr/tree/abram/esp32-smp

Primary issues I found were:

1. MISC0 instead of MISC1 being used for the CPU pointer in places that didn't use the ZSR macros
2. Incorrect stack initialization of secondary core. (The fix still needs review and optimization) It was causing a memory access error during a window overflow routine.
3. IPI's not correctly configured, and directed IPIs ignored the bitmap..
4. _current not initialized during an early call to global IRQ lock. (I wrote a BAD hack for this. Forgive me)
5. Xtensa sys clock driver shared a last_count between the cores which caused massive jumps in the system counter.

I don't have a board setup in-tree to support the changes yet as I was using a local project to run it. Will update that soon. I currently only have access to the esp32 ethernet kit, so I can't test on any other chips right now.

alexpaschoaletto, 2024-12-18:
Ok, so after opening issue #83168 I went through another search within existing issues and discussions and eventually found this one. The update I bring is that for the current 0.17.0 SDK and 4.0.99 version of the kernel, enabling `CONFIG_SMP` fails to build.

It certainly doesn't help that my available esp32 seems to be of a revision that is no longer supported, although if I understand correctly revision 1.0 should be the precise one which enabled SMP:

> Judging by when it was implemented (see above), I'm almost sure that Zephyr's SMP was made on top of Revision 1. [...]

If anyone knows of other targets that support SMP (`qemu_x86_64` apparently does but I'm having trouble running that aswell 😅), I'd be grateful. Thanks in advance.

rftafas, 2024-12-18:
@alexpaschoaletto did you check the content above? Seems it is working (I could not test it).

Also, there is an ongoing conversation on Discord, Xtensa.

alexpaschoaletto, 2024-12-18:
@rftafas I noticed that and will test it eventually.
I'm not yet part of the Discord community, but will check that out aswell, thanks!

williamtcdns, 2025-01-27:
@dcpleung , is SMP working for other Xtensa boards such as soc/intel/intel_adsp ?

dcpleung, 2025-01-27:
> [@dcpleung](https://github.com/dcpleung) , is SMP working for other Xtensa boards such as soc/intel/intel_adsp ?
> 
> Are those boards passing the SMP tests at `tests/kernel/smp` ?

Yes, they are passing in our internal CI on actual hardware.

closed 2025-05-14 by github-actions[bot]