zephyrproject-rtos/zephyr #56013: Net Socket offload scope clarification

Crzyrndm, 2023-03-21:
I'm unsure if this is a bug report, a feature request, a missing/overlooked piece of documentation, or ...? I'm using Zephyr through nRF Connect SDK (Zephyr version is v3.2.99, NCS version is v2.3.0)

I am building an application which uses an external cell modem for networking (connecting to AWS IoT Core via MQTT/TLS). I have an existing (C++) driver for the cell modem from a previous application I'm using that exposes the socket interface. I found the page on ["socket offloading"](https://docs.zephyrproject.org/3.2.0/connectivity/networking/api/sockets.html#socket-offloading) which appears to provide the exact gateway I need for my driver to interface with the Zephyr networking stack

Given that this is a prototype and I already have a driver lib written that was easily integrated at the application level, I was intending to do the socket integration at the application level also. *Most* of the required API surface is exposed (notably `NET_SOCKET_OFFLOAD_REGISTER` and `z_reserve_fd` / `z_finlinaze_fd`), however the API struct `socket_op_vtable` (defined in `<zephyr>\subsys\net\lib\sockets\sockets_internal.h`) is not and this appears to the only indication that this API is meant to be restricted to kernel modules I have found (the documentation doesn't even seem to imply it other than potentially the context it is referred to in)

I can easily create a matching struct and just cast the pointer but that then means at minimum that I need to check the layout on any change in SDK version and I have not yet checked to see what other issues are going to arise using what is apparently(?) an internal API.

So
* Is this meant to be a kernel only API?
  * If yes: some better documentation of this would be beneficial
  * If no: why is that one struct private?

My preference would be that this is not meant to be a private API given the above situation

nordicjm, 2023-03-22:
Drivers are kernel only, see https://docs.zephyrproject.org/latest/kernel/drivers/index.html for details

Crzyrndm, 2023-03-22:
Any options/recommendations for prototyping with a modem that doesn't have an existing kernel driver and won't in the near future (would require a from scratch rewrite)?

closed 2023-03-24 by Crzyrndm