zephyrproject-rtos/zephyr #56015: Processor architecture not supported in CI

yperess, 2023-03-21:
**Describe the bug**
I have a pending PR with a failing test that appears to be due to a misconfiguration or a filter. See https://github.com/zephyrproject-rtos/zephyr/actions/runs/4475987894/jobs/7865921441?pr=56014#step:13:4552

**To Reproduce**
Steps to reproduce the behavior:
1. `west twister -i -v -p ehl_crb -s tests/subsys/edac/ibecc/edac.ibecc.injection.user`

**Expected behavior**
Test should pass

**Impact**
PRs that trigger this test will fail.

**Environment (please complete the following information):**
 - CI

finikorg, 2023-03-21:
I just run 
`west twister -i -v -p ehl_crb -s tests/subsys/edac/ibecc/edac.ibecc.injection.user`
works fine

nashif, 2023-03-21:
@yperess  the issue is in your PR, so it is not a bug in Zephyr (yet).

nashif, 2023-03-21:
Ok, so this happens only with https://github.com/zephyrproject-rtos/zephyr/pull/56014, so there is an issue including the file `zephyr/sys/time_units.h`

stephanosio, 2023-03-21:
> Ok, so this happens only with #56014, so there is an issue including the file `zephyr/sys/time_units.h`

Yea, I was just testing on the `main` and was unable to reproduce this issue using both SDK 0.15.2 and 0.16.0.

stephanosio, 2023-03-21:
Closing this issue since this is not a bug in the Zephyr main branch, and is only reproducible on the #56014 PR branch when building with Zephyr SDK 0.16.0 (which CI currently uses).

If you think the build failure seen in #56014 is a toolchain bug, please file an issue in the [sdk-ng repository](https://github.com/zephyrproject-rtos/sdk-ng).

nashif, 2023-03-21:
@stephanosio  the issue is, why including this header works fine with 0.15.2 but not with 0.16.0?

jhedberg, 2023-03-21:
I have a feeling the problem is with the chain of headers that `zefi.c` includes. It's built independently from the Zephyr build system (using the `zefi.py` script), and I think the idea is that it should *not* rely on things like Zephyr Kconfig options, which is what you end up exposing it to in #56014 . I can't yet explain the difference between 0.15.2 and 0.16.0 SDK versions however.

stephanosio, 2023-03-21:
> @stephanosio the issue is, why including this header works fine with 0.15.2 but not with 0.16.0?

As per https://github.com/zephyrproject-rtos/zephyr/pull/56014#issuecomment-1478305524 and https://discord.com/channels/720317445772017664/1087789120119644283/1087789224041910314, this fails with both SDK 0.15.2 and 0.16.0  -- SDK (or GCC) version is not a factor here.

After that, it is starting to seem more like yet another missing include or incorrect include order issue.

yperess, 2023-03-21:
Can I get clarification? I'm understanding that this is maybe an include order issue, maybe a toolchain issue (unlikely), or maybe an issue specific to how we build the failing platforms. Why is the issue closed? I'm far from the right person to fix this and clearly my PR exposed an issue. Yes it's not in main, but it's blocking another fix from going into main and possibly others.

stephanosio, 2023-03-22:
As explained above, the above test does not fail in `main` and only fails with the change you have made in your PR.

closed 2023-03-22 by stephanosio

yperess, 2023-03-22:
> As explained above, the above test does not fail in `main` and only fails with the change you have made in your PR.

So by that logic can I close https://github.com/zephyrproject-rtos/zephyr/issues/56006? Since it doesn't happen in main?

stephanosio, 2023-03-22:
> So by that logic can I close #56006? Since it doesn't happen in main?

In #56006, user opened a bug

Here you are opening a bug

stephanosio, 2023-03-22:
> So by that logic can I close #56006? Since it doesn't happen in main?

In #56006, user opened a bug issue because of a broken macro implementation, which is reproducible on main; he made no changes to the main itself.

Here you are opening a bug issue for a problem that only shows up after modifying the main.

stephanosio, 2023-03-22:
#56014, which supposedly fixes #56006, should fix whatever issue comes up as part of the fix; if not, the fix is incomplete.

It does not make sense to open a bug for an incomplete bug fix.

finikorg, 2023-03-22:
Can this help? https://github.com/zephyrproject-rtos/zephyr/pull/56088

vindicatorr, 2023-03-27:
I just encountered this issue in the #matter channel in that discord, so I don't know if there's a relevance of me posting this here, or if it should go to the sdk-ng repository mentioned earlier (or even go to the connectedhomeip repository)...

I too am using v16 of the toolchain.

I was altering `connectedhomeip/examples/all-clusters-minimal-app/nrfconnect` to use `config/zephyr` instead of `config/nrfconnect`.
Here's a snippet of the output (looks like the actual zephyr part of the build went fine):
```
...
[689/717] Linking CXX static library zephyr/kernel/libkernel.a
[690/717] Creating directories for 'chip-gn'
[691/717] No download step for 'chip-gn'
[692/717] No update step for 'chip-gn'
[693/717] No patch step for 'chip-gn'
[693/717] No configure step for 'chip-gn'
[694/717] Performing build step for 'chip-gn'
Starting Matter library build in <pathTo>/connectedhomeip/examples/all-clusters-minimal-app/nrfconnect/out/build/nrf52840dongle_nrf52840/modules/connectedhomeip
Done. Made 194 targets from 178 files in 5543ms
[1/372] ACTION //src/lib/asn1:gen_asn1oid(//config/zephyr/chip-gn/toolchain:zephyr)
[2/372] stamp obj/src/lib/asn1/gen_asn1oid.stamp
[3/372] stamp obj/src/lib/asn1/asn1oid_header.stamp
[4/372] stamp obj/src/lib/dnssd/platform_header.stamp
[5/372] stamp obj/src/lib/support/attributes.stamp
[6/372] ACTION //src/crypto:gen_crypto_buildconfig(//config/zephyr/chip-gn/toolchain:zephyr)
[7/372] ACTION //src/app:gen_app_buildconfig(//config/zephyr/chip-gn/toolchain:zephyr)
[8/372] ACTION //src/lib/core:gen_chip_buildconfig(//config/zephyr/chip-gn/toolchain:zephyr)
[9/372] stamp obj/src/crypto/gen_crypto_buildconfig.stamp
[10/372] stamp obj/src/app/gen_app_buildconfig.stamp
[11/372] stamp obj/src/lib/core/gen_chip_buildconfig.stamp
[12/372] stamp obj/src/crypto/crypto_buildconfig.stamp
[13/372] stamp obj/src/app/app_buildconfig.stamp
[14/372] ACTION //src/ble:gen_ble_buildconfig(//config/zephyr/chip-gn/toolchain:zephyr)
[15/372] ACTION //src/inet:gen_inet_buildconfig(//config/zephyr/chip-gn/toolchain:zephyr)
[16/372] stamp obj/src/lib/core/chip_buildconfig.stamp
[17/372] stamp obj/src/lib/support/logging_constants.stamp
[18/372] stamp obj/src/app/app_config.stamp
[19/372] stamp obj/src/ble/gen_ble_buildconfig.stamp
[20/372] stamp obj/src/inet/gen_inet_buildconfig.stamp
[21/372] stamp obj/src/platform/logging/headers.stamp
[22/372] stamp obj/src/ble/ble_buildconfig.stamp
[23/372] stamp obj/src/inet/inet_buildconfig.stamp
[24/372] stamp obj/third_party/nlassert/nlassert.stamp
[25/372] stamp obj/third_party/nlio/nlio.stamp
[26/372] c++ obj/src/trace/trace.trace.cpp.o
[27/372] stamp obj/src/trace/trace.stamp
[28/372] ACTION //src/lib/support:gen_chip_version(//config/zephyr/chip-gn/toolchain:zephyr)
[29/372] ACTION //src/system:gen_system_buildconfig(//config/zephyr/chip-gn/toolchain:zephyr)
[30/372] stamp obj/src/lib/support/gen_chip_version.stamp
[31/372] stamp obj/src/system/gen_system_buildconfig.stamp
[32/372] ACTION //src/platform:gen_platform_buildconfig(//config/zephyr/chip-gn/toolchain:zephyr)
[33/372] stamp obj/src/lib/support/chip_version_header.stamp
[34/372] ACTION //src/setup_payload:gen_additional_data_payload_buildconfig(//config/zephyr/chip-gn/toolchain:zephyr)
[35/372] stamp obj/src/system/system_buildconfig.stamp
[36/372] stamp obj/src/platform/gen_platform_buildconfig.stamp
[37/372] stamp obj/src/setup_payload/gen_additional_data_payload_buildconfig.stamp
[38/372] stamp obj/src/app/common/cluster-objects.inputdeps.stamp
[39/372] stamp obj/src/ble/ble.inputdeps.stamp
[40/372] stamp obj/src/crypto/crypto.inputdeps.stamp
[41/372] stamp obj/src/inet/inet.inputdeps.stamp
[42/372] stamp obj/src/access/access.inputdeps.stamp
[43/372] stamp obj/src/app/app.inputdeps.stamp
[44/372] stamp obj/src/app/server/server.inputdeps.stamp
[45/372] c++ obj/src/access/examples/libaccess.PermissiveAccessControlDelegate.cpp.o
FAILED: obj/src/access/examples/libaccess.PermissiveAccessControlDelegate.cpp.o 
<pathTo>/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++ -MMD -MF obj/src/access/examples/libaccess.PermissiveAccessControlDelegate.cpp.o.d -Wconversion -Os -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-unused -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -D_SYS__PTHREADTYPES_H_ -isystem<pathTo>/zephyr/include/zephyr/posix -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -isystem<pathTo>/zephyr/include -isystem<pathTo>/connectedhomeip/examples/all-clusters-minimal-app/nrfconnect/out/build/nrf52840dongle_nrf52840/zephyr/include/generated -isystem<pathTo>/zephyr/soc/arm/nordic_nrf/nrf52 -isystem<pathTo>/zephyr/lib/libc/newlib/include -isystem<pathTo>/zephyr/include/zephyr/posix -isystem<pathTo>/zephyr/lib/posix/getopt/. -isystem<pathTo>/zephyr/soc/arm/nordic_nrf/common/. -isystem<pathTo>/zephyr/subsys/mgmt/mcumgr/util/include -isystem<pathTo>/zephyr/subsys/mgmt/mcumgr/transport/include -isystem<pathTo>/zephyr/subsys/usb/device -isystem<pathTo>/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/nrfx_glue -isystem<pathTo>/zephyr/subsys/bluetooth -isystem<pathTo>/zephyr/subsys/net/lib/sockets/. -isystem<pathTo>/zephyr/subsys/settings/include -isys [... cut ...]
<pathTo>/zephyr/include/zephyr/sys/time_units.h:64:16: error: 'CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC' was not declared in this scope
   64 |         return CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC;
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
...
```
However, what I'm noticing is that the `.config` values don't appear to be transferred from the zephyr part of the build to the `gn` part.

Again, this all may be completely unrelated to whatever OP has going on, but given the timing, I thought I'd bring it forth.