zephyrproject-rtos/zephyr #56018: `settings_save_one` function in `bt_set_name` function does not save null terminating string character

IvanVnucec, 2023-03-21:
https://github.com/zephyrproject-rtos/zephyr/blob/d4d0907ede61fceb1e2200ee2867b9a4371917f1/subsys/bluetooth/host/hci_core.c#L3877

I think that the `settings_save_one` function should save `len+1` items, not `len`.

IvanVnucec, 2023-03-21:
It seems that this change triggers hard fault:
```
[00:00:34.800,506] <err> os: ***** HARD FAULT *****
[00:00:34.800,537] <err> os:   Fault escalation (see below)
[00:00:34.800,567] <err> os: ***** MPU FAULT *****
[00:00:34.800,567] <err> os:   Data Access Violation
[00:00:34.800,567] <err> os:   MMFAR Address: 0x20003f5c
[00:00:34.800,598] <err> os: r0/a1:  0x00116677  r1/a2:  0x00000000  r2/a3:  0x00000003
[00:00:34.800,598] <err> os: r3/a4:  0x00000008 r12/ip:  0x00116680 r14/lr:  0x00000000
[00:00:34.800,628] <err> os:  xpsr:  0x61000000
[00:00:34.800,628] <err> os: Faulting instruction address (r15/pc): 0x000020b4
[00:00:34.800,659] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:34.800,689] <err> os: Current thread: 0x20001fe8 (unknown)
[00:00:34.865,203] <err> fatal_error: Resetting system
```

Can anyone reproduce?

hermabe, 2023-03-21:
Please use the bug template and describe your actual issue. The null-termination seems to be handled by `set_settings` in `subsys/bluetooth/host/settings.c`.