zephyrproject-rtos/zephyr #56023: k_uptime_get() is not accurate on NXP platform in zephyr 3.2.0

awojak, 2023-03-21:
I've observed time difference between `k_uptime_get()` and real time. After 24h there is -55min of difference, so it's not acceptable. `gpt_hw_timer` is used as tick source for the kernel and by default `TICKLESS KERNEL` is enabled. External 32.768kHz quartz accuracy is 6ppm and gpt timer use that quartz. Tested on custom hardware with `RT1062CVJ5B`.

Workaround is to not use `TICKLESS KERNEL`.
After that change time difference after 24h is almost 0s.

IMO there can be some BUG in nxp_gpt_hw_timer driver. 
 
**To Reproduce:**
Steps to reproduce the behavior:
1. Enable gpt_hw_timer:
```
&gpt_hw_timer {
	status = "okay";
};
```
2. Be sure that `CONFIG_TICKLESS_KERNEL=y`
3. Check time difference using `k_uptime_get()` after 24h

**Expected behavior:**
Time difference should be exactly accuracy of the 32k quartz.

**Impact:**
Kernel uptime is not precise and it has affect for the functions like k_sleep() etc.

**Environment:**
 - OS: Debian 11
 - Toolchain Zephyr SDK 0.15.0
 - Zephyr 3.2.0

awojak, 2023-03-22:
I was able to reproduce this BUG on MIMXRT1020-EVK but time difference is much smaller -156s/day. Probably system load can have impact to it. I've modify sample code `blinky` and add three task with continuously changing sleep time of each task. 
```c
#include <zephyr/kernel.h>
#include <zephyr/drivers/gpio.h>

static void task1_loop(void *par1, void *par2, void *par3);
static void task2_loop(void *par1, void *par2, void *par3);
static void task3_loop(void *par1, void *par2, void *par3);

K_THREAD_DEFINE(task1_thread, 256,  task1_loop, NULL, NULL, NULL, 7, 0, 0);
K_THREAD_DEFINE(task2_thread, 256,  task2_loop, NULL, NULL, NULL, 7, 0, 0);
K_THREAD_DEFINE(task3_thread, 256,  task3_loop, NULL, NULL, NULL, 7, 0, 0);

#define SLEEP_TIME_MAX_MS 100
#define LED0_NODE DT_ALIAS(led0)

static const struct gpio_dt_spec led = GPIO_DT_SPEC_GET(LED0_NODE, gpios);

static void task1_loop(void *par1, void *par2, void *par3)
{
	uint32_t delay = 25;

	printk("Started %s\n", __func__ );
	while (1) {
		gpio_pin_toggle_dt(&led);
		k_msleep((int32_t)(delay % SLEEP_TIME_MAX_MS));
		delay++;
	}
}

static void task2_loop(void *par1, void *par2, void *par3)
{
	uint32_t delay = 50;

	printk("Started %s\n", __func__ );
	while (1) {
		gpio_pin_toggle_dt(&led);
		k_msleep((int32_t)(delay % SLEEP_TIME_MAX_MS));
		delay++;
	}
}

static void task3_loop(void *par1, void *par2, void *par3)
{
	uint32_t delay = 75;

	printk("Started %s\n", __func__ );
	while (1) {
		gpio_pin_toggle_dt(&led);
		k_msleep((int32_t)(delay % SLEEP_TIME_MAX_MS));
		delay++;
	}
}

void main(void)
{
	int ret;

	if (!device_is_ready(led.port)) {
		return;
	}

	ret = gpio_pin_configure_dt(&led, GPIO_OUTPUT_ACTIVE);
	if (ret < 0) {
		return;
	}

	printk("Initialized %s\n", __func__ );
}
```
Device was started at 16:05:00. Next day at 8:46:00 I've checked elapsed ms:
```dark
uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0  ***
Initialized main
Started task1_loop
Started task2_loop
Started task3_loop

uart:~$ kernel uptime
Uptime: 59950882 ms
```
Time difference is -109s. Approximately -156s/day

awojak, 2023-03-22:
I've noticed that the time difference gets bigger with more system load. When I changed `#define SLEEP_TIME_MAX_MS 5` then the time difference is -1610 s/day. For `#define SLEEP_TIME_MAX_MS 2` the time difference is -3550 s/day.

danieldegrasse, 2023-03-22:
Hi @awojak, just an initial question- does your application require the GPT hardware timer? Since Zephyr 3.3 we have switched RT boards to use the Systick timer by default (due to issues like the one you have encountered here): https://github.com/zephyrproject-rtos/zephyr/pull/53028 If you do not need to utilize low power features in your application, using the systick timer may be a good short term solution.

awojak, 2023-03-22:
Hi @danieldegrasse, I utilize low power feature in my application, so systick is not the solution. Temporary gpt2 CNT register could be used for counting time, but finally, a correct solution should be found.

awojak, 2023-03-23:
@danieldegrasse is it possible to fix this ? It is quite important in our project.

danieldegrasse, 2023-03-23:
@awojak A fix should be possible, yes. It will take some time to verify it though, given the way that you have reproduced the problem

danieldegrasse, 2023-03-27:
@awojak I've created https://github.com/zephyrproject-rtos/zephyr/pull/56251. I would like to run additional regression tests against this PR before it is merged, but from my testing it seems that the time difference has improved. I ran your uptime test over the weekend, with the following results:

```
RT time difference checker started at 06:28:00 PM CDT 3/24/23

Time difference checker checked at 02:29:11 PM CDT 3/25/23 (346ms difference, 0 s/day)

Time difference checker stopped at 09:07:00 AM CDT 3/27/23 (Time difference is 1133 ms, 0 s/day)
```

This test was run with `CONFIG_TICKLESS_KERNEL=y`

dleach02, 2023-05-17:
@awojak, did @danieldegrasse PR #56251 address your problem? I would like to close this issue report if it did.

awojak, 2023-05-24:
@dleach02 Problem is resolved, issue can be closed.

closed 2023-05-24 by danieldegrasse