zephyrproject-rtos/zephyr #56024: Add erbr-ot as contributor

aescolar, 2023-03-21:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

@erbr-ot , to be added to the Zephyr GitHub collaborator list, so he can triage issues and PRs and his approvals count.

## GitHub User

Provide the following information about the GitHub user:

1. Erik Brockhoff
2. @erbr-ot 

## Supporting Documents

* https://github.com/zephyrproject-rtos/zephyr/pulls?q=is%3Apr+author%3Aerbr-ot+
* Has been a Bluetooth controller "collaborator" for many years. He is already listed in the MAINTAINERS file.

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

aescolar, 2023-03-21:
CC @erbr-ot @jhedberg @cvinayak @carlescufi @wopu-ot @thoh-ot @kruithofa @mtpr-ot

stephanosio, 2023-03-21:
I have added `erbr-ot` to the GitHub collaborator team since he is already listed as a collaborator in the `MAINTAINERS.yml`.