zephyrproject-rtos/zephyr #56067: Fix build issues with arm clang compiler for mec17xx platforms

galak, 2023-03-21:
There are a number of different failures we see when building the arm clang compiler when building the mec17xx platforms.

galak, 2023-03-21:
https://github.com/zephyrproject-rtos/zephyr/issues/56067

galak, 2023-04-10:
Backport #56577 to fix init order issue w/arm-clang.

galak, 2023-04-10:
Backport #50049 to fix build warnings/errors with arm-clang.  Really just need commit f5e892749b82f2b2e3dc9e343205a37a21fb2f88.

galak, 2023-04-10:
Backport 49899 as start of fixing some linker issues w/arm-clang and logging.

jgl-meta, 2023-06-20:
@galak what's the status of this one? It was marked stale and has many PRs that have been merged.

jhedberg, 2023-08-29:
Many/most issues already fixed. If something is still broken please open a new bug report.

PR #56690 (2023-04-29): [Backport v3.2-branch] Update to Clang 15
Fixes #56067

Backport 39af51372dd2020e6a2397d0b2c04797102f3775~9..39af51372dd2020e6a2397d0b2c04797102f3775 from #50049
PR #56688 (2023-05-03): [Backport v3.2-branch] drivers: intc: mchp_ecia_xec: Ensure correct device init order
Fixes #56067
Backport 161ca0376f85b0de2999b646b8fd4f56b8c51e75 from #56577
PR #56691 (2023-05-04): [Backport v3.2-branch] logging: Use STRUCT_SECTION macros for log backend
Fixes #56067

Backport e6bbc6941a338a3f81757e7dfb28bd9822e5810d from #49899
closed 2023-08-29 by jhedberg